luc-github/ESP3D #7: Cannot compile project if using latest https://github.com/esp8266/Arduino

luc-github, 2015-05-22:
this is due to string management different than in sandeep's module
if remove all macro F() and all PROGMEM it can compile and ESP is working properly

need to see how to handle this

TRoager, 2015-05-22:
Ill follow here because i get the same error something about hasarg..

luc-github, 2015-05-22:
does removing F() and PROGMEM worked for you also ?

what is hasarg?

luc-github, 2015-05-23:
F() was not doing anything before 
should be same for PROGMEM but I did not checked -  looking a http://www.arduino.cc/en/Reference/PROGMEM 
it can be read directly

TRoager, 2015-05-23:
[image]

TRoager, 2015-05-23:
Dunno if that was the same problem you guys have. But heres my compile error, without editing anything :/

luc-github, 2015-05-24:
nope you is looks like typo issue should be ```hasArg``` not ```hasarg```

for the ```onNotFound``` it may due to version of ESP8266 you use - do you use the download from board manager ? did you use sandeep version before ?

TRoager, 2015-05-24:
i used https://github.com/sandeepmistry/esp8266-Arduino , and changed hasarg to hasArg and everything compiles fine now. Now ill try and upload it :)

luc-github, 2015-05-25:
the  0.0.5 worked fine for me but I wanted to use the board manager version because all update will come by this one,  I think sandeep won t maintain his repo anymore as no need

luc-github, 2015-05-27:
fixed with https://github.com/luc-github/ESP8266/commit/f48e419e0ff81413954debcb219540a543689ee1

Using properly PROGMEM allows to save 4k of memory currently almost 22K after first page displayed

closed 2015-05-27 by luc-github