luc-github/ESP3D #13: Web Front End for handling printer

luc-github, 2015-06-07:
need to define features:
current test code is able to get temperatures from M105 command, every second, but no UI is defined, plan to use some svg images to enhance UI
front end should allow to:
check printer status
edit eeprom?
pause/stop print
restart printer
list SD content
start a print using SD file ?
control temperature ?
change speed ？
jog axis ?

advanced:
send email is issue detected? print is done?

any comment suggestion is welcome

luc-github, 2015-06-08:
Basic Front end is now available : https://github.com/luc-github/ESP8266/commit/b981ffa4d6126afdb693fcc8d24ef420aaf636fa

only reporting for now : temparatures, positions, speed and flow, no action allowed yet

luc-github, 2015-06-10:
[image]

luc-github, 2015-06-11:
Command Send implemented -  now just need to define what commands need to be used, currently M112 = Emmergency Stop is implemented

[image]

luc-github, 2015-08-19:
movement icons: home buttom for axis, up/down and left/right at step of 1mm and 10mm (i think easy to add this options)
Suggestion from @c128
real time temperature control
stop/pause/restart button
sd_memory: upload, delete and select file to print
feed speed direct control
advanced option:

use low price I2C webcam as Ov7670 for direct printing remote monitoring
professional option:

3d model preview :)

luc-github, 2015-09-25:
I have done some simple svg images for the front end to make ui clear and simple

[image]

Pause / Stop / Print or Continue / Emergency Stop / SD card to see file and select file to print

```
<svg width="40" height="40" viewBox="0 0 40 40">
  <rect x="5" y="5" width="10" height="30" rx="2" ry="2" style="fill:rgb(0,0,0);stroke-width:1;stroke:rgb(0,0,0)" />
  <rect x="25" y="5" width="10" height="30"  rx="2" ry="2"  style="fill:rgb(0,0,0);stroke-width:1;stroke:rgb(0,0,0)" />
</svg>

<svg width="40" height="40" viewBox="0 0 40 40">
  <rect x="10" y="10" width="20" height="20"  rx="2" ry="2"  style="fill:rgb(0,0,0);stroke-width:1;stroke:rgb(0,0,0)" />
</svg>

<svg width="40" height="40" viewBox="0 0 40 40">
<polygon points="5,5 35,20 5,35" style="fill:fill:rgb(0,0,0);stroke:fill:rgb(0,0,0);stroke-width:1" />
</svg>

<svg width="40" height="40" viewBox="0 0 40 40">
<circle cx="20" cy="20" r="18" stroke="black" stroke-width="2" fill="red" />
<circle cx="20" cy="20" r="10" stroke="black" stroke-width="4" fill="red" />
<rect x="15" y="8" width="10" height="10" style="fill:red;stroke-width:1;stroke:red" />
<rect x="18" y="6"  rx="1" ry="1" width="4" height="14" style="fill:black;stroke-width:1;stroke:black" />
</svg>
 
<svg width="40" height="40" viewBox="0 0 40 40">
  <rect x="5" y="10" width="30" height="20"  rx="2" ry="2"  style="fill:rgb(0,0,0);stroke-width:1;stroke:rgb(0,0,0)" />
  <rect x="20" y="5" width="15" height="15"  rx="2" ry="2"  style="fill:rgb(0,0,0);stroke-width:1;stroke:rgb(0,0,0)" />
  <text x="10" y="25" font-family="Verdana" font-size="14"  fill="white">SD</text>
</svg>

```

luc-github, 2015-09-25:
For temperature / fan / flow / speed , I will use a combination of range/number imput :
[image]

luc-github, 2015-09-25:
And here possible jog UI but I want to simplify it - also it miss step selection (0.1/1//10)
[image]

I used same SVG and played with CSS to rotate to save time / memory, current SVG for home is almost 1K from internet - I think I will redo it to decrease size

luc-github, 2015-09-25:
here new set - not sure which one is better :
[image]

luc-github, 2015-09-25:
home image reduced by 50% and allowing text inside
here x y z and all compare to original one:
[image]

```
<svg width="40" height="40" viewBox="0 0 40 40">
<polygon points="7,40 7,25 4,28 0,24 20,4 26,10 26,6 32,6 32,16 40,24 36,28 33,25 33,40"  fill="black"   stroke-width:"1" stroke:"black" />
<line x1="25" y1="8" x2="33" y2="16" style="stroke:white;stroke-width:1" />
<polyline points="4,28 20,12 36,28" style="fill:none;stroke:white;stroke-width:1" />
<text x="15" y="35" font-family="Verdana" font-size="14"  fill="white">X</text>
</svg>
```

luc-github, 2015-09-25:
ok I have added color and text in image - now look little fancy but at least it is simple and clear, Extruder 1 and 2 will have same UI as Z

[image]

luc-github, 2015-09-25:
now need to integrate UI and code...

luc-github, 2015-10-15:
[image]
UI is functional in test branch - still missing function to select file and launch print
will work on it when current code is valided on ESP01 and not only nodemcu 1.0

luc-github, 2015-10-16:
seems 64K SPIFFS on ESP 01 bring some trouble
I worked around but now I use 98% of storage - I think implementing mecanism to grab SD file list and EEPROM content should be ok but UI won't follow due to current bug of 64K which are actually 48K

luc-github, 2015-10-16:
For SPIFFS > 64K and flash size > 512K should be no issue like for nodemcu 1.0
https://github.com/luc-github/ESP8266/commit/7379ac580d00718e216075818b040ed1df6c1650

c128, 2015-10-17:
good job

luc-github, 2015-10-17:
thank you but I need to rewrite printer.tpl, currently it use 26k on 52k available on ESP 01, no issue so far for module with bigger SPIFFS area but no more space to add UI for SD Card/EEPROM management

still some work to do

c128, 2015-10-17:
i suggest for extruder move 0.1 to 1mm, 1 to 10mm and 10 to 30mm

luc-github, 2015-10-17:
yes make sense, I just copied and past Z ui currently, now any change is in tpl so more easy no need to go to fw

c128, 2015-10-17:
i make this shield for esp with optical insulation 
 http://forums.reprap.org/file.php?359,file=59452,filename=foto1.png
esp01 + this shield is more expensive like esp12

i thick best solution for web server with esp is to use more powerful version not very expensive like esp-12 with external sd memory and lcd status for commercial use

luc-github, 2015-10-17:
nice work, yes I was thinking to use Adafruit HUZZAH not expensive and no need sd card

c128, 2015-10-17:
when you decide i can make easy shield pcb for ramps board

luc-github, 2015-10-17:
Ha thanks a lot :smile:

luc-github, 2015-10-19:
steps changes and file manager are done https://github.com/luc-github/ESP8266/commit/5c4e11ea3f1597761f68d05ec782549c07efc7b8

luc-github, 2015-10-19:
64K SPIFFS has now a light pinter.tpl with no fancy but all commands
[image]

luc-github, 2015-10-20:
fix missing home buttons for 64K
[image]

luc-github, 2015-10-20:
File selection and print are now implemented 
no directory/no file deletion no upload as ESP01 won't be able to handle it and this also can mess up with wifi SD Card which do not like to have to way to modify the SD Content
on NodeMCU 1.0 with 1M SPIFFS (not fully tested as only EP01 is connected to my printer):
[image]   
      
ESP01 with 64k SPIFFS    (tested and currently used on my DAVINCI)
[image]

luc-github, 2015-10-20:
I think I will drop the EEPROM management as it need around 4K ~5K space of data and this is tricky and I do not see the need for the moment
may be when core will be more optimized ( need to recover 4k due to SSL which we do not use for now) or using SPIFFS as swap but will work only on 4M module which I do not have, Node MCU  sending too much garbage at start is not a good candidate for printer usage.

So I close topic for now but any enhancement are welcome

commit 3dae051e (2016-11-11): Fix #13 "wait" issue for pipe going to dead loop

Fix % not showing during FW update on lcd status
closed 2015-10-20 by luc-github