luc-github/ESP3D #18: Randomly module reset

luc-github, 2015-06-11:
this can be seen when using a long time (memory leak ?) but memory says 24k when going to Home

result is connection is lost need refresh page manually, all history of status/error/info is lost

luc-github, 2015-06-12:
adding a system_get_free_heap_size() after each refresh show memory is always over 20K
no error is show on serial part 

may be watchdog issue

luc-github, 2015-06-13:
This definitively help, web page has been working during 8h without resetting the module

so keep waching

luc-github, 2015-06-13:
ok so still happen - less than before - so means have several root causes -  need to dig in

luc-github, 2015-06-15:
if module is connected to printer without web server actively used =>no issue
if module is connected to printer so no serial exchange / no command interpretation but web server actively used => issue
so currently it narrow to PROGMEM usage / Web Server usage

I will do the page without any call to PROGMEM and related access function to see if I can reproduce the reset

luc-github, 2015-06-15:
was able to reproduce without PROGMEM but using const char
so now testing modified HelloServer.ino which do refresh

[image]

luc-github, 2015-06-15:
Ok module reset again
I remove mDNS part and redo a test - NB  I open 2 web windows to stress the server

When crash again (normally) I will have a sketch to submit the issue to ESP8266 repo

luc-github, 2015-06-15:
crash again but I forget to disable Watchdog - so redo test again

luc-github, 2015-06-16:
Seems related to watchdog which actually cannot be disabled
Need to wait for a fix

TRoager, 2015-06-18:
i saw this: http://www.esp8266.com/viewtopic.php?f=23&t=588&start=140
But its not related to arduino. Maybe theres a fix somewhere for the watchdog anyway in the comments ?

luc-github, 2015-06-18:
well actually there are several issues :
1 - Watchdog functions are not really working because not implemented, but actually watch dog it is just symptom of issue not root cause
2 - the key problem is : there is heap fragmentation problem that may generate the reset

I have reported github.com/esp8266/Arduino/issues/428 and also saw some others have same kind of issue, I saw on gitter.im they will work on this soon, actually someone is already on this, as several people have issue with reset when it is supposed to work

luc-github, 2015-06-26:
you can follow here : github.com/esp8266/Arduino/issues/428
developer was able to reproduce it but still no clue of the root cause :
___I was able to reproduce this — although i'm not sure what is the root cause of this issue.
The exception happens because heap gets corrupted. Why this happens is not yet clear — this might be a simple buffer overrun or some concurrency/race condition kind of issue. Wasn't able to solve this on the first try but will definitely look more into this.___

I guess this is good news as before he was not able to see it so need to wait as issue in in core

TRoager, 2015-07-07:
Any status? :)

luc-github, 2015-07-07:
nope new staging version does not fix issue yet and several people have have random reset also so I guess they need some time to isolate the problem

luc-github, 2015-07-07:
do you reproduce issue also ?

TRoager, 2015-07-07:
I see, i just couldnt figure out where to follow up on it, other than the link you gave.

I havent had it running for a while so im not sure, after you said there were this issue i havent been running it since :/

luc-github, 2015-07-07:
reset are random, I even did more than 13K loops before getting it happen
I am currently using it as status viewer and listing all issues I see and will fix them soon,

luc-github, 2015-07-12:
I have added some resistors as suggested in several forum and diagrams, seems improve the module stability 
<IMG SRC=https://raw.githubusercontent.com/luc-github/ESP8266/master/Wires.png>

luc-github, 2015-09-28:
Ok fixes looks promising - on short tests not even one reset/crash
Now doing a long run test - finger crossed

luc-github, 2015-09-29:
OK issue looks solved - but need some additionnal workaround like reserve some space for some variable String

I will close it when staging module is released

luc-github, 2015-10-02:
in staging now so I close

closed 2015-10-02 by luc-github