luc-github/ESP3D #20: Add support for 3rd party application like GCodePrintr / octodroid

luc-github, 2015-06-12:
but need to define first for what app - seems octodroid has a repo but it is not very active https://github.com/mariogrip/octodroid

or I code client myself

TRoager, 2015-06-12:
theres also: 
https://gitlab.com/damfle/SmoothieControl

luc-github, 2015-06-12:
Thanks for sharing I will have look ^_^

TRoager, 2015-06-12:
Your welcome, i check this hub daily :)

luc-github, 2016-04-08:
Seems websocket is the way to go to properly support any app

luc-github, 2016-04-08:
But first need to identify an App to target it - so far I did not found any stable one

luc-github, 2017-05-19:
As UI is now monolitic I do not see a real use of working on standalone client

closed 2017-05-19 by luc-github

luc-github, 2018-02-15:
Just a note :    
Seems 3D Fox - 3D Printer / CNC Controller  https://play.google.com/store/apps/details?id=com.fox3d.controller
works pretty well with data port and ESP3D