luc-github/ESP3D #29: Error Wifi AP! M117 192.168.0.1

c128, 2015-08-03:
compiled without change anything in source give error
after try with change ip, name etc nothing to do and print "Error Wifi AP! M117 192.168.0.1" 

i think not initialize esp to ap mode
my version of esp8266 is 01
http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family#esp-01
arduino version 1.6.5

esp compiled with other source working 
for example this is ok and esp go to ap mode
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/esp8266com/esp8266/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino

luc-github, 2015-08-03:
do you use latest staging module ?
I usually use station mode after AP mode
I try try to reproduce issue

c128, 2015-08-03:
arduino ide is reinstalled with latest staging module: http://arduino.esp8266.com/package_esp8266com_index.json
http://arduino.esp8266.com/staging/package_esp8266com_index.json

no errors after compiling

luc-github, 2015-08-03:
can you also share what settings you are using?
it may help to reproduce issue because AP mode is working reading default configuration

c128, 2015-08-03:
try with ubuntu and arduino ide same config 
when compiling there is only problem to find lib (i think problem only for linux):

in webinterface.cpp

#include <PgmSpace.h>
replace with
#include <pgmspace.h>

but esp not start with AP 

i make video with my procedure

luc-github, 2015-08-03:
yes my bad I did not compiled on mac/linux - thanks for pointing this
fixed with https://github.com/luc-github/ESP8266/commit/8dd8ec207d602a81f378cc9272083ad0861f68e4

still checking the AP error your get as cannot reproduce

luc-github, 2015-08-03:
one additional information - you need to reset EEPROM as when flashing it does not erase EEPROM area - so if the EEPROM is corrupted in some way - AP won't start
you must plug module then connect GPIO2 to GND 2s after plug and before 10s - this will recreate a clean EEPROM

luc-github, 2015-08-04:
can you share what settings were used on ESP to settup AP ? as I am not able to reproduce issue
I changed IP, Channel, etc and no issue

c128, 2015-08-04:
nothing to do!
same photo:
[image]
[image]

c128, 2015-08-04:
i can program this without problem
http://ez-gui.com/manual/multiwii-clearflight-wifi-to-ezi-gui-how-to/
or this
http://www.arduinesp.com/wifiwebserver

luc-github, 2015-08-04:
this is flash settings -  I am talking about AP settings did you used to reproduce the issue as SID ESP8266 and default settings are working as you mentionned you were able to change them

luc-github, 2015-08-04:
from this page <img SRC="https://raw.githubusercontent.com/luc-github/ESP8266/master/Page3.png">

luc-github, 2015-08-06:
any update ? I still cannot reproduce issue

c128, 2015-08-06:
i think power problem with my USB
wait for testing with other pc

luc-github, 2015-08-06:
Ok

c128, 2015-08-12:
problem is impedance  of push button switch for program and reset 
you need good electronics for program this device

i make shield for esp8266 pin compatible with bluetooth hc-05 and optical isolated
for now prototype is working very well 
[image]

luc-github, 2015-08-12:
ok great,so issue can be closed ?

c128, 2015-08-12:
yes is ok
now testing 3d printer

closed 2015-08-12 by c128