luc-github/ESP3D #31: 'class ESP8266WebServer' has no member named 'send_P'

CaptBlaubaer, 2015-08-08:
webinterface.cpp: In function 'void handle_css()':
webinterface.cpp:1738: error: 'class ESP8266WebServer' has no member named 'send_P'
  web_interface->WebServer.send_P(200, TEXT_HTML, CSS);
                                             ^
'class ESP8266WebServer' has no member named 'send_P'

Arduino 1.6.5-r2 win7 using
http://arduino.esp8266.com/stable/package_esp8266com_index.json

luc-github, 2015-08-08:
you need to use staging if it is not yet in stable

luc-github, 2015-08-08:
yes seems in staging and not yet in stable version :
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/changes.md

CaptBlaubaer, 2015-08-08:
Hey luc,

perfect response time!
Compiling worked, but can`t connect now.

Best regards,
CaptBlaubaer

closed 2015-08-08 by CaptBlaubaer

luc-github, 2015-08-08:
Well I found some devices cannot and others can connect to AP ESP, they see AP but after password sent => unable to connect
Android Phone / tablet = no issue never
Windows laptop = some can / some cannot, I did not triggered what was the difference all in Win7