luc-github/ESP3D #32: reset after connecting to webserver

CaptBlaubaer, 2015-08-08:
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

is thrown by device via serial interface.
any idea what i`m making wrong?

Best regards,
CaptBlaubaer

luc-github, 2015-08-08:
using default settings or you already changed some ?

luc-github, 2015-08-08:
not sure it is same as mine : https://github.com/luc-github/ESP8266/issues/18
It happen randomly, I reported issue as I have it also with hellowwebserver sample code : github.com/esp8266/Arduino/issues/428

CaptBlaubaer, 2015-08-08:
am using default settings

luc-github, 2015-08-08:
check you power supply - be sure you added some pull up resistors - also do not forget after flash to remove GPIO 0 from GND

CaptBlaubaer, 2015-08-08:
[image]

Using this design, Prog jumper is removed. Powersupply 2a wallwart

luc-github, 2015-08-08:
I only have ESP01 so cannot comment about 12

CaptBlaubaer, 2015-08-09:
same behaviour with a brand new nodemcu.
reset after the first load of a web page

ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

luc-github, 2015-08-09:
Strange - I do not reproduce - I connect from W7 to AP and can navigate to each page without problem

luc-github, 2015-08-09:
can you reproduce when using webserver sample App ?

CaptBlaubaer, 2015-08-09:
AdvancedWebServer example works without a reset. device is station mode and connected to my router

luc-github, 2015-08-09:
may be try using AP mode for AdvancedWebServer 
or change code to use Station Mode by default in my code.

luc-github, 2015-08-13:
Are you using internet explorer browser ? 

I am using Chrome and having no issue -  but usint IE I can reproduce the issue - the only page it does not happen on IE is unknown page.
Can you confirm ?

luc-github, 2015-08-13:
Ok when can reproduce it is more easy 
The root cause is the css, I may introduce some bug by the deep clean up
If remove in webinterface.cpp:
``` "<link rel=\"stylesheet\" href=\"/mincss.css\" type=\"text/css\">\n</head>\n<body>"\```
it do not crash anymore under IE

additionally it solve a new issue that appear under chrome/IE about security setting block the refresh of the page on printer part

So css is definitively buggy - need to redo the UI without bootstrap CSS funtionnalities but keeping fancy

luc-github, 2015-08-19:
New css embedded seems not crashing under Internet Explorer anymore
https://github.com/luc-github/ESP8266/commit/1cb93ef7873f352ac9d21772eeed8d43aad7be52

closed 2015-08-19 by luc-github