luc-github/ESP3D #33: AP Configuration shows Station SSID and Password

CaptBlaubaer, 2015-08-08:
Feature request: Separate SSID and password for AP  and station.
I guess if you apply on on the AP Configuration page it`ll take over the SSID and will try to start an own AP with the values of my existing router.

luc-github, 2015-08-08:
Not sure to get it sorry. :blush: 

if you go to AP Web page configuration and Apply : next restart will start ESP on AP mode only with saved settings.
if you go to Station Wep page configuration and Apply : next restart will start ESP on Station mode only with saved settings, 

I only use temporary AP+STA  when you are in AP mode and you are configuring Station mode to scan available SSID. 

So actually you are never connected in AP and STA in same time with same SSID/Password - this would use ESP as a bridge which has no real value for ESP I think

Or did I missed something ?

CaptBlaubaer, 2015-08-08:
sorry, i only could load max one page before device resets. that's not enough for me to follow your idea.
will try again if my nodemcus will arrive.
i hope they don't show that behaviour.

luc-github, 2015-08-08:
I only have ESP01 - they all start properly I AP mode with default  and after I switch to Station mode without issue to let it reach my network.
But for what I can see on ES8266 github tracker - ESP have different behavior according the model

Good luck

luc-github, 2015-08-27:
Any update on this ?

luc-github, 2015-09-08:
If no update - I cose it - please reopen if needed

closed 2015-09-08 by luc-github