luc-github/ESP3D #34: test, errors and urgent options

c128, 2015-08-13:
program works at 250.000 baud, no problem to connect
stability is ok with good power supply
first problem detected: printer status not update and remain stopped if you use browser update
For restart connection to printer you need to select  "station configuration" and reconnect to "printer status"
i suggest to add this options for productive use:
- movement icons: home buttom for axis, up/down and left/right at step of 1mm and 10mm (i think easy to add this options)
- real time temperature control
- stop/pause/restart button
- sd_memory: upload, delete and select file to print
- feed speed direct control

advanced option:
- use low price I2C webcam as Ov7670 for direct printing remote monitoring 

professional option:
- 3d model preview  :)

luc-github, 2015-08-13:
Hi thanks for your feedback suggestion:
the new issue about refresh seems to buggy CSS and recent update in Chrome:|
you can workaround removing css call in webinterface.cpp line 50:
```"<link rel=\"stylesheet\" href=\"/mincss.css\" type=\"text/css\">\n</head>\n<body>"\```
until I redo all UI without CSS, currently it is working - but just ugly

about :
For restart connection to printer you need to select "station configuration" and reconnect to "printer status" - removing css call solve this - the refresh is automatic 

about :
commands yes I agree - but I was also thinking to add security check (https://github.com/luc-github/ESP8266/issues/19) before adding more controls in UI as if anyone has access to your printer from internet it is not good -  what do you think ?

about : 
adding camera I only have ESP01 so I am not sure will work on this - but I have a ESP07 somewhere should have enough pin on this one

about:
3D preview, are you confident an ESP01 has capability for this ? or are you using another module ?

c128, 2015-08-13:
security: i think wpa password is sufficient for now. Internet security? add firewall option for esp IP

camera: you can use SDA  to GPIO2 and SCK to GPIO0, i try to find library compatibility with esp

3D preview: i think esp-12 with 4mbyte can make this

luc-github, 2015-08-13:
WPA is wifi security - I mean to  secirity to access to your ESP web UI
Currently if someone access to ESP webUI from internet ( only if you allow outside connection of course) then , because web UI has no password, so anyone can reset your printer

about camera - yes would be great, thanks!

3D preview: no idea - I do not have one

c128, 2015-08-13:
remote connection to esp? suggest to use vpn connection with firewall protection
if you add password your printer remain subject to a potential DOS attack. 
Only solution is protect ip from internet.

luc-github, 2015-08-13:
Well this is also a solution - and the eaiesty one because it does not need to modify FW ^_^

Anyway I will solve this UI issue then apply new commands then

c128, 2015-08-13:
camera: i2c for esp is "software mode"  then is impossible to connect camera. I suggest to use wifi low price camera.

like this

http://www.aliexpress.com/item/IP-security-camera-wifi-module-for-720P-960P-1080P/32349867952.html

you need to add only streaming to webinterface code, easy and good solution

luc-github, 2015-08-13:
what you show me is only the wifi module for the camera - it is like ESP01 actually -  no ?

In all case handling external MJPEG flow it is just -HTML code - so it is easy

luc-github, 2015-08-19:
issue fixed by https://github.com/luc-github/ESP8266/commit/ac084da5b660d7a76c39ac9f6aff481319395496

and features are : https://github.com/luc-github/ESP8266/issues/13

So close as duplicate

closed 2015-08-19 by luc-github