luc-github/ESP3D #38: Error while esp8266 is running the code

AkhilReddyMandadi, 2015-09-19:
I connected HX711 to my arduino uno and trying to upload the data from it through the digital pins (7,8) to an website . The code is uploading perfectly but when I open the serial monitor im getting this.... 

ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

is thrown by device via serial interface.
any idea what i`m making wrong?

luc-github, 2015-09-21:
what version are you using of ESP8266 module : staging or stable  ?

luc-github, 2015-09-21:
also I read you use HX711 and  arduino uno, I do not see the link with  ESP8266 and this github source code.
Can you elaborate ?

luc-github, 2015-09-21:
wrong tracker should be arduino

closed 2015-09-21 by luc-github