luc-github/ESP3D #41: Random Exception (28) - pointer/memory issue

luc-github, 2015-10-01:
randomly this exception apear but I was not able to reproduce it on simple sketch so it is not module issue but code issue.
I do not know where the issue come from - I know urldecode function may bring this issue as it is not secured yet but this happen just on page navigation not on queries so it is not urldecode.

As I will rewrite UI engine with a template engine, I will review all page and enhance memory checking, I hope this will fix this issue

luc-github, 2015-10-02:
Using a Node MCU 1.0 I even cannot display page once - so decreasing PROGMEM usage by using SPIFFS should help

luc-github, 2015-10-02:
Reported on ESP8266 tracker using a test sketch

luc-github, 2015-10-05:
seems over usage of memory - because page use 8K in one string to be sent and more and more feature available memory is now in critical situation - so using the template from SPIFFS and build then send page part by part will avoid this issue in theory.
The template engine is almost ready, just miss definition for list box and check box and default selected item. 
I limit the creation part to same size as 1200 so it let enough memory for string manipulations.

luc-github, 2015-10-08:
seems some root cause is Captive portal as I can reproduce crash doing nothing but wait with captiveportal.ino sample sketch

luc-github, 2015-10-21:
I did a PR to fix issue github.com/esp8266/Arduino/pull/912 -  should be ok in next staging/stable

closed 2015-10-21 by luc-github

denmorru, 2017-03-26:
I got those 
`Exception (28) + Exception (0) + wdt reset`
when started newly uploaded ESP3D.ino to empty ESP-12F, without uploading esp3d.spiffs.bin

luc-github, 2017-03-26:
well if using `SPIFFS.begin();` without any SPIFFS partition - the behaviour may be random
I recently flashed a new nodeMCU without SPIFFSand restart was clean after usual garbage bootloader :
```
dÜbûŒœß„Û üÃ'œ‡;ƒÛ“n#›gdœ|Äûg|plŸ<gà�‚ìã’ƒbÛ'`s$
M117 ESP EEPROM reset
M117 SSID ESP8266
M117 192.168.0.1
```
I do not mean issue cannot happen - but hard to predict