luc-github/ESP3D #42: SOFTAP is broken due to memory overload

luc-github, 2015-10-02:
I cannot use AP mode anymore - need to disable CSS to get display when it was working before - no idea where this come from - I tried http://arduino.esp8266.com/versions/1.6.5-1084-ga39ce29/package_esp8266com_index.json and it is working

As I am rewriting UI using template engine and SPIFFS this will decrease usage of PROGMEM and string manipulation
Page will also be created and send part by parts to avoid usage of whole memory for the page - instead of creating a 8K page then cut it to be send by 2k parts - I will create 2k then send - then continue to redo 2K then send, etc... so page not be complete in memory and so won't have any size limitation.

luc-github, 2015-10-02:
Seems only disable captive portal solve the issue

luc-github, 2015-10-05:
Template engine is almost done I will create a branch to test it

luc-github, 2015-10-15:
No more memory issue using template engine and SPIFFS on nodemcu 1.0
need to do test on ESP01 for final validation

luc-github, 2015-10-15:
will close it when test code is in master

luc-github, 2015-10-16:
OK implemented in https://github.com/luc-github/ESP8266/commit/7379ac580d00718e216075818b040ed1df6c1650

closed 2015-10-16 by luc-github

denmorru, 2017-03-29:
Tested ESP-01 with 1M flash. Used 50% memory for SPIFFS. It is working,
but ony without `#define CAPTIVE_PORTAL_FEATURE`. Explain, why this feature makes device unbootable with ESP3D firmware version 0.9.78?