luc-github/ESP3D #43: Reset

lehar, 2015-10-09:
Thanks for a very good project :+1:  and thanks for sharing it.

When I applay a page, and there is a EPS8266.restart(), it will not boot up again. There is no debug messages (which at first start up) and there will be no life in the wireless network (ESP8266).

Do I something wrong, I use a NodeMCU 1.0

Best regards
Per

luc-github, 2015-10-09:
Do you use latest staging of ESP8266 ?
What branch do you use in my project? Master or test?
Also in all case seems having issue with captive portal - you may disable it at it cause some restart

luc-github, 2015-10-09:
Also seems with latest staging available memory is in critical path due to latest additions - and I seems very visible on NodeMCU 1.0 compare to ESP01
So you may need to disable CSS usage also to free some memory.

I am now using NodeMCU 1.0 also and rewriting the rendering to use template and limit memory usage - you can have a look on test branch but I will finish all porting only next week - currently only home page  and system page are done - still need to port AP configuration and STA configuration then will do the printer UI porting

lehar, 2015-10-09:
I use latest staging of ESP8266
I have tried both branches, Master and Templatetest, it's the same.
I look forward to your next update, with templates :+1:

luc-github, 2015-10-09:
template test is currently working I use it on NodeMCU 1.0 - even you do not upload the template it will show "cannot open /home.tpl" but should not restart

luc-github, 2015-10-09:
may I have more detail of your actions ?

lehar, 2015-10-09:
I have nothing more to add right now. (I press Apply on the system configuration page. it says restarting, and the connection is lost, and never come up again. Serial monitor says : F[08])þ)

Next week I have holidays, but when I'm back again, I'll try with your latest update and try to come up with more input for you.

luc-github, 2015-10-09:
Ok this is more clear you have this after submitting the form 
what setting do you change before applying ?
What speed do you set in serial monitor ?
How do you recover ? unplug / replug module ?
What OS are you using ?
Normally when starting after 10s is display M117 192.168.0.1 then module is visible on network

lehar, 2015-10-09:
I don't change anything, just press apply.
I use default baud rate 9600
I can recover, by unplug and replug the module, or upload the software again.
I am using Windows 7 Ultimate
And I don't see M117 192.168.0.1 after reboot. Only then I unplug/replug.

luc-github, 2015-10-09:
did you tried to do not just after an upload, I do not have this issue on my nodemcu but onmyESP01 the restart does not work after an upload, i need to unplug replug first

lehar, 2015-10-09:
I have found the problem, but do not know why it is a problem.
I do the following:
1. compile software and upload from Arduino IDE
2. Connect serail interface (serial monitor)
3. Connect Wireless connection.
4. Refresh browser (page load)
5. switch to System configuration
6. Press Apply (and it never start again) :-1: 

// never tryied that before, I always reload SW before.
7. Unplug/replug
8. Connect Wireless connection.
9. Refresh browser (page load)
10. switch to System configuration
11. Press Apply (and now it restart) :+1:

luc-github, 2015-10-09:
after upload if no unplug replug the module may still be in upload mode, so the restart redo upload mode waiting to be flashed, it is always better to reset after an upload

luc-github, 2015-10-09:
if ok please close issue 

and feel free feedback especially on template part, i will be the big change and will allow a better UI and easily modifiable

closed 2015-10-09 by lehar