luc-github/ESP3D #45: Does the DaVinci 1.0 have enough power ?

gamecompiler, 2015-10-16:
Hi, id would like to test your ESP8266 firmware, but do the DaVinci really have enough power on the connector ? Or should i connect the esp to the main 5V rail ? Im just asking because i dont know where the 5V line is connected, i dont want to burn the poor SAM....

Sorry for my english.

Thanks in advance.

luc-github, 2015-10-16:
I can say that on my 2.0 ir is ok with ESP01 if you use another module I cannot say
be careful ESP is 3.3v not 5V

gamecompiler, 2015-10-16:
Ok, id meassured it.
All the lines work with 3.3V, (Sorry for my first post, it was clear that connecting the esp to 5V is a bad idea).
TX/RX and VCC lines operate at 3.3V on DaVinci 1.0. But i think it would be a better idea to add a 3.3V Regulator extra.

Thanks for your great Projects, they enhance my opportunities a lot.
(Sorry for my English)

closed 2015-10-16 by gamecompiler