luc-github/ESP3D #47: Any configuration of Repetier needed ?

gamecompiler, 2015-10-16:
Hi, 

id souccessfully flashed the ESP and id got it connected with the DaVinci.
But it cant connect to the printer because the printer do not answer.
Id also checked if id miss cabled RX/RX or TX/TX, there is no communication.
The ESP sends a lot of gcode to the printer, but it dont get any answer.

Do i have to change something in the Repetier Firmware to let it work ?

Thanks in advance.

luc-github, 2015-10-16:
repetier fw need to be compiled with wifi support in configuration.h
#define ENABLE_WIFI 1, by default is not

gamecompiler, 2015-10-16:
Thanks, it works, sometimes...

On the first flashing id worked, than id deactivatet the beeper and wifi in firmware.
After that, id flashed back the wifi version with deactivatet beeper,
(there is no reason why id made it in this order)

now i cant flash any version with wifi enabled, id also grabed your current master branch repetier source and tryed it without modifiying it (just enabled wifi).

(The old completly unmodified firmware still work)

Now the printers lcd is just black, and the LED bar is flickering every 12 seconds.
That makes me headaches, i think there is no reason for that....

It also dont work if the ESP8266 isnt connected anymore.

Is the Firmware waiting for sth on startup if wifi is enabled ?

Thanks in advance.

luc-github, 2015-10-17:
OK - when you flash DaVinci FW you must have the ESP Module disconnected - flash won't work properly with ESP connected and you may need to reset with Jumper to reflash again 

If flashing do not worked first time - reset with Jumper and retry - sometimes it need 2 tries

Also for safety be sure you connect / ESP module when printer is OFF

no worry printer is not bricked - it happened to me to when I flashed and forget to remove ESP

luc-github, 2015-10-17:
question : why deisable beeper in FW ? you can do in Seetings/EEPROM

luc-github, 2015-10-19:
Did you fixed it ?

TRoager, 2015-10-19:
does it only work with repetier? and are there any specifics to know if i have to use it on Marlin for example?

luc-github, 2015-10-19:
Repetier version for Davinci support it because we add a serial selector in FW

to use on Marlin you need to do same in FW, I did not have time to take a look, sorry

Another way could be to connect rx/tx to usb connector and power 3.3v/ground from board itself so no need FW change - or you use the serial1 pins of due for communication and modify FW accordingly

luc-github, 2015-10-19:
normally as soon you can make ESP and BOARD communicate by serial it will work

TRoager, 2015-10-19:
Okay, then ill look in the davinci repository :)

Yeah i thought so too, i have to try it out again, looks like theres alot happened since last time i tried flashing my esp :)

luc-github, 2015-10-19:
In all case I will work on it as soon as I am done with SDCard / EEPROM management as I plan to use it on my RADDS/DUE which is not a DaVinci

TRoager, 2015-10-19:
Sounds good :+1:

luc-github, 2015-10-19:
Things go slowly but they find their way :wink:

gamecompiler, 2015-10-19:
I will test it again, at 19:00 clock, first i have to install my new hotend Oo!
A e3D v6 with bowden, this will be awesome !

At this time i did not fixed the error, but i will ;D
If you are interessted, i can give you my PID setup for the e3d and bed, i have the tools to meassure the settings.

luc-github, 2015-10-20:
Yes sure - 
currently FW  latest version is on ESP01 connected to my Davinci 2.0 - I sent a print and so far so good  :smile:

luc-github, 2015-10-28:
Any update ? or do we close the issue ?

gamecompiler, 2015-10-29:
Yes, id successfully installed the hotend now,
but i cant get the Wifi Firmware working.

Could you please upload your .bin file ? (If you activate verbose output 
in the IDE you see the path in the end of the log)

I think there is a issue with my IDE.

luc-github, 2015-10-30:
with staging http://arduino.esp8266.com/staging/package_esp8266com_index.json
you must use https://github.com/luc-github/ESP8266/releases/tag/v0.4-beta.1 
 and compile according your module (ESP01, ESP12E, etc...)
if you have ESP01 size of SPIFFS is 64K so you need to change data directory content by the correct one

if you have ESP12E or any module with 4M SPIFFS I suggest to select 1M SPIFFS

To add the tool to upload SPIFFS you also need to download :
http://arduino.esp8266.com/ESP8266FS-1.6.5-1105-g98d2458.zip

it will add the tools mandatory to upload SPIFFS image file. 
ESP8266FS is a tool which integrates into the Arduino IDE. It adds a menu item to Tools menu for uploading the contents of sketch data directory into ESP8266 flash file system.

Download the tool: http://arduino.esp8266.com/ESP8266FS-1.6.5-1105-g98d2458.zip
In your Arduino sketchbook directory, create tools directory if it doesn't exist yet
Unpack the tool into tools directory (the path will look like <home_dir>/Arduino/tools/ESP8266FS/tool/esp8266fs.jar)
Restart Arduino IDE
Open esp8266.ino sketch with right content in data disrectory
Make sure your board is in upload mode, and have selected the right board, right size of SPIFFS, right port, and closed Serial Monitor
Select Tools > ESP8266 Sketch Data Upload. This should start uploading the files into ESP8266 flash file system. When done, IDE status bar will display SPIFFS Image Uploaded message.
if you have MKSPIFFS failed create image, it means you took the content for more than 64K to be used in 64K.

if SPIFFS upload failed, but previous FW flash was working, verify you are in upload mode (GPIO0 to GND) and no serial terminal open, can also be a cable issue per other user experience

TRoager, 2015-11-03:
After i changed my usb to ttl converter everything seemed to work with the staging package.

Upload of firmware and files went smooth after the change. Im still waiting for some parts to test out the firmware with my printer.

luc-github, 2015-11-03:
great - soon a new staging will be released  so I will release a new beta after I will add authentification and web update of the FW if no major issue

gamecompiler, 2015-11-07:
The ESP works nice already, with webinterface and such on, very nice :+1: , but i cant get the printer firmware work well.

Id got the firmware uploaded to the printer now, but that was luck.
Im still trying to figure out why i have to try the upload a minimum of 30 times....

I will post a update, when im ready....

luc-github, 2015-11-07:
it is weird - be sure when you flash repetier FW,  ESP module must not be connected to printer - if happen you may end with black bars and need a jumper reset to recover

luc-github, 2015-11-14:
any update on this ?

luc-github, 2015-11-18:
Ok I close issue if no update

closed 2015-11-18 by luc-github