luc-github/ESP3D #50: UI enhancement/optimization

luc-github, 2015-10-20:
Basic UI is done / back end is ready - but I am not a designer anyone want to improve or have suggestion - please feel free

pcbroute, 2015-11-24:
Hi.

Some questions.

I try latest version.
Why if i changed SSID: and Password: on Access point page 
after restart is changed to the same on Stantion page too?

Why axis of printer control  by default is inversed ?
Typical on any software like Pronterface, Repetier Host, etc 
is
up + 
down -
right +
left - 

And it well be a great to add  by default 
Fan speed slider control
and Button for Motors Off command.

Thanks.

luc-github, 2015-11-24:
About Password/SSID:
Because module is currently used as AP or Client not both in same time - I save SSID/PWD in same place in  EEPROM  to save place

About UI : currently controls match the printer I have : a DAVINCI.
so directions match the printer's ones.
Also DAVINCI does not have a fan control (it is automatic) so I forget to implemented it, same for M84 command - it is automatic after a timeout.
Anyway, all these changes are easy to implement as they are in printer.tpl, may be add a generic template with generic UI with all controls like repetier or pronter face is a good idea

pcbroute, 2015-11-24:
I understand that is used not both in same time.

But it turns out we have two different tabs for the same data, why, what's the point?
I think that the printer must have its own name ( like 3d printer WIFI) not the name of my home router access point name.

Yes, Generic UI is great.

luc-github, 2015-11-24:
the 2 tabs are just ready with what it is in EEPROM, same for Static /DHCP and used only when you select apply to use these info or modify them to be in AP or Station

About name I am not sure what name you refer to - default hostname is defined by ESP and last digits of Mac address, I use same name for SSDP and mDNS, of course it is modifiable  - nothing is related to an external router

luc-github, 2016-04-08:
Will add a template directory to show different UI - one of them would be the pronterface UI pushed by @treepleks

luc-github, 2016-04-11:
I list all TODO for reference:
1 - add templates directories to show different UI and let user see the possibilities
2 - add generic UI  for printer with all commands like current one miss fan command and may be others
3 - add web customized UI allowing show hide commands
4 - add custom button eg: for light on/off and others specific commands
5 - add embedded window for wifi cam 
6 - TBD

luc-github, 2016-05-07:
this .js has small footprint but seems promizing : https://github.com/evanw/lightgl.js

luc-github, 2016-07-08:
UI with repetier style is ongoing and 
[image]

will bring some new feature like macro support 
have a smoothieware UI based on pronterface also cooking as mentioned earlier 
[image]

luc-github, 2016-07-19:
here first draft of Repetier style - still some to do , also seems rendering is different with IE vs Firefox and Chrome.
Original repetier is :
[image]

The UI style on web browser    
On chrome:
[image]

On firefox:
[image]

On IE:
[image]

I am not expert in CSS so did not found why IE display input range (sliders) in so big area

last challenge is to display the temperature cursor and should be ok

luc-github, 2016-07-20:
Ok looks IE has several limitation compare to other browser and using a relative position to put text on control is not very accurate cross browser - no mention to add a second cursor for current temperature.
So I think I will use SVG and mouse event and build my own customized  rang input instead of use css - it should have homogenous behavior cross browser as use svg and mouse event only

luc-github, 2016-07-21:
framework using div/svg and javascript is ready and rendering is identical under IE/firefox/opera/chrome and safari
one limitation is no filter are rendered on safari 5 (under window) - it is supposed Ok under 6 and 7 but I do not have a osx to check right now
so rendering is less fancy but fully working:
[image]

luc-github, 2016-07-22:
Proof of concept is now done- UI is 90% identical what ever the browser but safari 5 -  still need to test safari 6 / 7 but at least now Internet explorer is no more different:
Firefox (47.0)
[image]

Internet Explorer (11)
[image]

Chrome (51.0.2704.103 m)
[image]

Opera (38.0)
[image]

Safari (5.1.7 windows)
[image]

So next step is to integrate it into tpl file 
then work on SDCard UI part ...

luc-github, 2016-07-22:
ok on safari 9 (el capitan) it is like others browsers, so no issue so far - all good :smile: 

[image]

luc-github, 2016-07-22:
with edge present in windows 10
[image]

luc-github, 2016-07-24:
so globally the control part is done  :
[image]
which is pretty close to original repetier, but I think will separate jog from monitoring and allow to hide each one individually 
same for SD Card part and status  parts

luc-github, 2016-07-25:
Ok seems now we have 2 UI - still need one more and should be ok -  this one will be an original one as pronterface and repetier UI will be style UI as reference

luc-github, 2016-11-24:
for smoothie there is imrahil UI : https://github.com/luc-github/smoothieware-webui-for-ESP3D

I plan to add the dual SD Card support, only /sd will be used for upload but listing/print functions should be able to use both SD :  /ext and /sd using shell commands

 Add simple UI for flow/speed/fan could be nice also

luc-github, 2016-11-28:
some command can take time to be answered, like M205 which send back 4K of data, or some shell command on smoothie - displaying a loader until we get answer displayed would give better user experience

luc-github, 2017-04-22:
Complete rewrite is still ongoing :
it use bootstrap embedded, limited use of glyphicon also embeded, a modified version of smoothiecharts and it is monolitic and compressed - no external call i s necessary

luc-github, 2017-05-19:
Repetier UI is not yet implemented but smoothieware UI is and compatible with all FW  and has its own github  https://github.com/luc-github/ESP3D-WEBUI so UI is now no more part of FW itself

closed 2017-05-19 by luc-github