luc-github/ESP3D #51: ESP01 version is working please feedback

luc-github, 2015-10-20:
I use it from now but - feedback/suggestion of improvement are always welcome

luc-github, 2015-10-22:
OK some improvement are already on going but need new staging to be released so I won't push to github now but here is a preview:
Now FM display space - that was really missing especially for 64K SPIFFS to see what can be done / or not
[image]

luc-github, 2015-10-23:
please use : https://github.com/luc-github/ESP8266/releases/tag/v0.4-beta.1

luc-github, 2016-01-15:
due to ESP01 with 1M flash this is no need anymore
I close issue as now seems ok fr current release issue is more related to GIT version vs Stable version

closed 2016-01-15 by luc-github

denmorru, 2017-03-29:
Simply stated already in documentation, that ESP3D firmware needs 1M flash memory or more. https://github.com/luc-github/ESP3D/wiki