luc-github/ESP3D #53: ESP12E version is working please feedback

luc-github, 2015-10-22:
ESP12E has 4M flash, so it allow bigger code and 1M SPIFFS and cost almost same as ESP01

I just assembled one and now using it - need to do the final wiring using a PCB and socket for it to add all resistor and connector to Printer
[image]

[image]

luc-github, 2015-10-23:
please use : https://github.com/luc-github/ESP8266/releases/tag/v0.4-beta.1

pcbroute, 2015-11-11:
[image]
[image]
Hi!
I try to use two the same ESP12E WIFI module, it works, but not long and not stable, module is random crash.
I try different power supply, powered by USB, batteries,  or through module based on ams1117-3.3, or LM2596 chip, and i get the same, it crash.

I think it crash if connected to access point, if it not connect it work more stable. i try more to test it

M117 Connecting...
M117 192.168.0.85

Exception (0):
epc1=0x4010718a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffed0:  3fff51c0 3fff6a38 3fff04e4 4021da46  
3ffffee0:  3ffefe88 000000de 3ffefea5 3ffefe7c  
3ffffef0:  3fff4874 3ffefe88 3ffefe98 3ffefeae  
3fffff00:  3fff0000 4021eaaa 3fff2438 3fff4a34  
3fffff10:  3fff04d0 3fff4a38 3fff6a38 4021df6c  
3fffff20:  00000002 3fff51c0 3fff51c0 3fff3e00  
3fffff30:  00000000 3fff6a38 0000001c 3fff51c0  
3fffff40:  3fff04c2 00000000 3fff6a38 40224659  
3fffff50:  0100a8c0 00000050 00000000 0000001c  
3fffff60:  00000002 00000018 4022badd 3fff2438  
3fffff70:  3fff049c 000000f5 3ffee958 3ffee958  
3fffff80:  4022ba79 3fff2438 00000000 3fff5218  
3fffff90:  3fffdc80 00000000 3fff6a38 402250ab  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 192.168.0.85

Exception (0):
epc1=0x4010718a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffed0:  3fff51c0 3fff6a28 3ffefea0 4021da46  
3ffffee0:  00000000 3fffdc20 3ffee01c 00000030  
3ffffef0:  3fff4874 3ffef844 3ffef854 3ffef86a  
3fffff00:  3fff0000 4021eaaa 3fff2410 3fff4a34  
3fffff10:  3ffefe8c 3fff4a38 3fff6a28 4021df6c  
3fffff20:  00000002 3fff51c0 3fff51c0 3fff3e00  
3fffff30:  00000000 3fff6a28 0000001c 3fff51c0  
3fffff40:  3ffefe7e 00000000 3fff6a28 40224659  
3fffff50:  0100a8c0 00000050 00000000 0000001a  
3fffff60:  00000002 00000018 4022badd 3fff2410  
3fffff70:  3ffefe58 000000f5 3ffee9e8 3ffee9e8  
3fffff80:  4022ba79 3fff2410 00000000 3fff5230  
3fffff90:  3fffdc80 00000000 3fff6a28 402250ab  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,1)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 192.168.0.85

luc-github, 2015-11-11:
what model do you use?
do you connect them directly to power or do you add, resistor and capacitor as suggested?

pcbroute, 2015-11-12:
I tried using both modules on picture

i connect module on my board directly to 3.3v power supply.
capacitor on power line is on board.
schematic is the same, i use pullups for
reset
ch_pd
gpio0  pins

gpio15 is pulldown

TX and RX line connected directly to Arduino Due board (due is 3.3 level)

on Doit Devkit board, ESP12E module is powered by stabiliser on board.
i tried to put it on external power sypply, but it work the same

is captive portal enabled or disabled ?

if module is connected to router access point on module is disabled.
web interface is available by its IP address

luc-github, 2015-11-12:
so you do not use any resistor as suggested ? just connect directly to gnd or vcc depending of the pin

pcbroute, 2015-11-12:
yes. power line 3.3v on my board is directly connected to pin GND and VCC on module.

i tested more.
if module is not connected to router (router is power off)
AP mode is enabled automatically.

M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Connecting...
M117 Safe mode started

it works good without any crash a several hours

luc-github, 2015-11-12:
I do not have any crash in Station mode, it is suggested everywhere to add the resistors for stability reason - including on ESP github : https://github.com/esp8266/Arduino/blob/master/doc/boards.md
the wiring I suggest is based on different version to cover the maximum - I suggest to try it

pcbroute, 2015-11-12:
https://github.com/esp8266/Arduino/raw/master/doc/ESP_improved_stability.png

my schematic is the same

luc-github, 2015-11-12:
Sorry I am lost, I understood you do not use any resistor and connect directly to VCC and GND

luc-github, 2015-11-12:
Not sure what is the problem - do you have same stability issue with webserver sample ?

pcbroute, 2015-11-12:
i dont know, need to try

pcbroute, 2015-11-12:
ESP8266 can be programmed to reduce the output power of the PA to fit various application profile.
Did you know how i can to reduce TX power?

pcbroute, 2015-11-12:
I found in the datasheet that consumption is very different

Transmit 802.11b, CCK 1Mbps, POUT=+19.5dBm 215mA
Transmit 802.11b, CCK 11Mbps, POUT=+18.5dBm 197mA
Transmit 802.11g, OFDM 54Mbps, POUT=+16dBm 145mA
Transmit 802.11n, MCS7, POUT=+14dBm 135mA

It may be is the reason

luc-github, 2015-11-12:
I do not have the latest SDK in pdf, I have the 0.95 which is outdated now we are in 1.3

about PHY mode I only use G,
Transmit 802.11g, OFDM 54Mbps, POUT=+16dBm 145mA
on PC with my nodeMCU 1.0, on My Davinci printer and on my DUE/RADDS, because N was not stable, issue for ESP to connect on it,  so I set my Access Point to not use N

pcbroute, 2015-11-12:
it very strange.

i tried connect to my mobile router, and all work absolutly fine.
but on my home router i get crash.

luc-github, 2015-11-12:
As i said I had to change some settings on my main AP router to get good connection (dlink), from another hand I connect with default settings on my extenders (totolink).
If you can reproduce with default webserver sample, it could help to post the issue on ESP github as it may be a bug, but I do not know how to analyse the elf file with the dump information - moreover I cannot reproduce the error ....

pcbroute, 2015-11-12:
my home router is dlink too (D-link DIR-825/AC), i check settings later
mobile router is huawei E5221

pcbroute, 2015-11-12:
i tried to change mode from mixed to G but no result, and it get crash anyway.

but i find what is the reason.

i disable multicast over Wifi on dlink router, and now all works, ESP module is not crash.

luc-github, 2015-11-13:
I had no crash but some issue to connect when have n
[image]

[image]

on my side I do not have multicast enabled neither - good catch

luc-github, 2016-01-15:
I close issue as now seems ok fr current release issue is more related to GIT version vs Stable version

closed 2016-01-15 by luc-github