luc-github/ESP3D #54: On new ESP first start after flashing, there is  exception(28)

luc-github, 2015-10-23:
I just did 4 new ESP12 and got exception(28) after first restart after flashing the 3 firsts - none after newer flash neither restart.

I suspected EEPROM check to generate this, so I have added the Serial.begin before doing check and did some serial print to follow and did not got crash but error starting mDNS.
I cannot reproduce anymore as all my ESP are now flashed at least once. Can also due to new bootloader installation and some timing issue as adding serial message did not show the issue- but I have a lack of trial - will keep for next time I have new ESP
This not critical as after flashing it is always good to restart module - but I would prefer to have a "clean" first boot

TRoager, 2015-10-23:
i just got this issue with the newest pre-release: Error opening : /home.tpl

:/

luc-github, 2015-10-23:
this is normal you need to copy files to SPIFFS place 
the tools copy the content from data

luc-github, 2015-10-23:
I have put link in readme but may be need to highlight:

you need to get the tool from this link : http://arduino.esp8266.com/versions/1.6.5-1160-gef26c5f/doc/reference.html#file-system

luc-github, 2015-10-23:
the file must be put under windows here :
<My document directory>\Arduino\tools\ESP8266FS\tool\esp8266fs.jar

let me know if you have issues

TRoager, 2015-10-24:
I have been messing with it this morning. and it seems like im getting this error: 

Uploading 3125248 bytes from C:\Users\tr\AppData\Local\Temp\build201004763003836653.tmp/esp8266.spiffs.bin to flash at 0x00100000
......warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04

luc-github, 2015-10-24:
what esp do you use? what os ?

be sure you are in upload mode like for flashing and no arduino terminal is open

TRoager, 2015-10-24:
Im using ESP12 and ESP12E, both giving same error with freshly installed Arduino 1.6.5 and Staging release. I can upload the firmware just fine just not the SPIFF, the terminal are closed, it looks like its about to flash for a few sec, liek you can see a few dots and then the error :(

luc-github, 2015-10-24:
weird
what setting do you use to flash ?

generic 4M (1M SPIFFS)

TRoager, 2015-10-24:
[image]

This is my error :/ and looks like the upload info is correct, but please correct me if im wrong :/

luc-github, 2015-10-24:
flashing SPIFFS should work even without my FW it is not related

you have 4M flash I guess, but issue looks more communication issue, and actually if you put GPIO 0 to GND should be ok

I am outside now, so not really helpful sorry

TRoager, 2015-10-24:
Hmm i will try and look at my setup to see if somethings wrong, just wierd because i can compile and upload just fine :/

luc-github, 2015-10-24:
after flashing fw you need to unplug replug esp to be in upload mode again

TRoager, 2015-10-24:
I did

luc-github, 2015-10-24:
my understanding is: it is same tool which flash fw than flash spiffs image, just location change

and your error look like tool cannot start communication

TRoager, 2015-10-24:
like you said communication failure. i changed my RS232 cable and now it seem to work :)
i had a FTDI adabter board from ebay, it seems like its not good enough :/

luc-github, 2015-10-24:
good ^_^

closed 2015-10-26 by luc-github