luc-github/ESP3D #59: Compiling error

pcbroute, 2015-11-04:
I got this

C:\Users\Root\AppData\Local\Temp\build2354136718967723719.tmp\webinterface.cpp: In function 'void handleFileList()':
C:\Users\Root\AppData\Local\Temp\build2354136718967723719.tmp\webinterface.cpp:2400:9: error: 'class fs::FS' has no member named 'info'
  SPIFFS.info(&total,&used);

if i change this

	jsonfile+="\"status\":\"" + status + "\",";
	uint32_t total;
	uint32_t  used;
	SPIFFS.info(&total,&used);
	jsonfile+="\"total\":\"" + formatBytes(total) + "\",";
	jsonfile+="\"used\":\"" + formatBytes(used) + "\",";
	jsonfile+="\"occupation\":\"" ;
	jsonfile+= intTostr(100*used/total);
	jsonfile+="\"";

on this
	jsonfile+="\"status\":\"" + status + "\"";

compiling is ok.

it works, it sends commands by web interface to printer. server is working.

but info about printer and status (like extruder temp) in web interface is empty

i use regular repetier 0.92 firmware running on DUE board
and ESP12E module

arduino-1.6.5-r5
esp8266/1.6.5-1160-gef26c5f

luc-github, 2015-11-04:
with staging you must use the beta release, not the master branch, or you need to use the latest code from esp8266 github which is not yet released

luc-github, 2015-11-04:
about : but info about printer and status (like extruder temp) in web interface is empty
did you set same baud rate between printer and module ?

pcbroute, 2015-11-04:
Hi!

Thank you.

I use latest code for esp8266, and it was successful.
Now i see in web interface data of file system
Total/Used space and Occupation is shown

I use 115200 baud rate, and printer correct response if i send command by web interface, moving axis, reset, set temp, flow, speed, is working. m117 works too, data is shown on lcd.
But in web interface not shown current Temp, 
and field Info, Error, Status, is empty.
Status indicator is red (disconnected)
and if i click on SD card icon in web interface, it not show file list.

Printer correctly works over WiFi if i connect to it by Repetier Host. I can send any command, and i see responce from printer.

maybe it's important, I do not EEPROM 

i post log over wifi  later...

Here is my log if i conect to printer over USB

22:44:41.420 : Printer reset detected - initalizing
22:44:41.462 : start
22:44:41.462 : Info:PowerUp
22:44:41.636 : N1 M110 *2
22:44:41.636 : N2 M115 *4
22:44:41.636 : N3 M105 *4
22:44:41.636 : N4 M114 *3
22:44:41.636 : N5 M111 S6 *66
22:44:41.672 : N6 T0 *28
22:44:41.672 : N7 M20 *54
22:44:41.684 : N8 M80 *51
22:44:44.071 : Free RAM:82252
22:44:44.073 : FlowMultiply:100
22:44:44.361 : ok
22:44:44.367 : N9 M105 *14
22:44:44.459 : ok 2
22:44:44.493 : FIRMWARE_NAME:Repetier_0.92.5 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Mendel EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
22:44:44.493 : ok 3
22:44:44.493 : N10 M220 S100 *112
22:44:44.510 : T:31.28 /0 B:72.80 /0 B@:0 @:0
22:44:44.510 : ok 4
22:44:44.510 : N11 M221 S100 *112
22:44:44.511 : X:0.00 Y:0.00 Z:0.000 E:0.0000
22:44:44.511 : ok 5
22:44:44.511 : N12 M111 S6 *116
22:44:44.511 : ok 6
22:44:44.511 : FlowMultiply:100
22:44:44.511 : ok 7
22:44:44.511 : Begin file list
22:44:44.512 : N13 T0 *40
22:44:44.515 : FOOT 3044052
22:44:44.516 : holder1B.gcode 1218061
22:44:44.516 : holder1A.gcode 1715334
22:44:44.516 : peppa.gcode 3620163
22:44:44.518 : End file list
22:44:44.518 : ok 8
22:44:44.518 : ok 9
22:44:44.519 : T:31.28 /0 B:72.80 /0 B@:0 @:0
22:44:44.519 : ok 10
22:44:44.519 : SpeedMultiply:100
22:44:44.519 : ok 11
22:44:44.520 : FlowMultiply:100
22:44:44.520 : ok 12
22:44:44.520 : ok 13
22:44:44.521 : FlowMultiply:100
22:44:44.760 : N14 M105 *50

luc-github, 2015-11-04:
what do you mean I do not EEPROM ?
can you check what is the json file content for the status ?
it is http://<esp ip>/STATUS, all temperatures speed should be on it

pcbroute, 2015-11-05:
Hi!

About EEPROM, i mean, i dont have EEPROM chip connected to DUO board for store Printer settings.

Thank you.
I find what wrong.
my browser, Internet Explorer 11) , is very strange refreshing data on page, and show old data from cache, or i dont know (list of SD card data is no working on it).

I try Firefox and all works fine, temp, status message is shown, file list on SD card is shown too.
Is awesome.

luc-github, 2015-11-05:
interresting I do not have any issue with IE, may be I should add no cache header, you may also check if IE did not block the refresh due to security settings

pcbroute, 2015-11-05:
I check secutiry settings
I dont know what settings is wrong, 
i set IE to InPrivate mode, and it works, data refreshing. 

But in normal mode problem is still

luc-github, 2015-11-05:
be sure ip is recognized as local network is security and then set security to low

pcbroute, 2015-11-05:
i check, 
set security level to low, but is not work
 IE is still no refresh data. After this manipulation, InPrivate mode broke,  show temp, but no refreshing it.
may be IE need some update...

on Firefox all Ok

luc-github, 2015-11-05:
weird, chrome is also working you can try

luc-github, 2015-11-05:
so issue can be closed ? as actually no issue :smile:

pcbroute, 2015-11-05:
Yes. All good! Thank you!

closed 2015-11-05 by pcbroute

luc-github, 2015-11-05:
thanks ^_^
additionally if you can feedback how you connected the module, as you seems not using RADDS according of ESP you used would be nice
https://github.com/luc-github/ESP8266/issues/51
or 
https://github.com/luc-github/ESP8266/issues/53