luc-github/ESP3D #62: Control Printer EEPROM settings

luc-github, 2015-11-18:
EEPROM is now arround 2K which with description give a 4K of data
Having a JSON file which is an image of EEPROM on SPIFFS could be a solution to avoid to use too much memory and fast exchange, by do a first sync a start then manual sync or just focus on changes

luc-github, 2016-01-23:
another way would be to use json file describing the EEPROM mapping type / position / meaning so it could save computation and match every printer 
so when displaying, page will query setting by setting to display the EEPROM values, and so give the ability to modify any value.
this would avoid to transfer a file too big but 
need to check if there is an command to get EEPROM specific value instead of full EEPROM on 
repetier and marlin

luc-github, 2016-06-20:
Note for implementaton smoothieware does not use EEPROM but rely on configuration file on SD

luc-github, 2016-11-10:
Ok now the web command catch the answer of the command - it should be easy to grab the EEPROM content without overloading the ESP at all

luc-github, 2016-11-11:
Yes a little javascript should be able to handle grabing EEPROM content and set value
here result with M205 in pipe
[image]

luc-github, 2016-12-21:
Repetier EEPROM editor is done : https://github.com/luc-github/FW-config-editor

Smoothie configuration file should follow soon 

for marlin I need to see what can be done - seems not really using EEPROM

luc-github, 2016-12-22:
Smoothieware is now implemented so I close issue - following can be done on https://github.com/luc-github/FW-config-editor

closed 2016-12-22 by luc-github

angelo-reprap, 2017-02-17:
Perfect Work and Thank you luc.

luc-github, 2017-02-17:
Thank you 
 I will release a new version soon integrated directly in (https://github.com/luc-github/smoothieware-webui-for-ESP3D) with a lot of bug fixes and improvement - current version has issues with Internet explorer - I am adding more feature and will do a release next week I hope

luc-github, 2017-02-17:
yes - it is orginally for smoothiware but my version support smoothieware/repetier and marlin - that is why I am little bit slow - test on all FW and several browsers take time