luc-github/ESP3D #64: Problem with master branch (webinterface.cpp)

treepleks, 2016-01-09:
Dear Luc,

I have cloned your code (master branch) with the aim of using/tuning it to run on a smoothieboard variant (the AZSMZ mini) which has room reserved for an ESP01 module on its LCD board. I have bought 2 ESP01 module, they have 1MB flash, I use 128KB for the SPIFFS filesystem. 

I compiled your code (with arduino 1.6.7 and the last version of the ESP8266 arduino module), flashed it but I have a problem with the web pages. I can connect to the ESP access point, authenticate, and access all the various pages but the "Station" page. The ESP never serves it.

I inserted a long list of Serial.println() in the corresponding function *handle_web_interface_configSTA()* in webinterface.cpp and the problem occurs on the following lines:

      //revert to pure softAP
      if (revertSTA) WiFi.mode(WIFI_AP);
      KeysList.add(FPSTR(KEY_AP_SCAN_VISIBILITY));
      ValuesList.add(VALUE_ITEM_VISIBLE);

If I comment the 2 last lines... everything seems to run fine. 
But I cannot see why these 2 lines could generate a problem (except for a lack of RAM issue).

Any idea ?

PS: I see you have also a specific Repetier firmware that communicate with the ESP module. Would it be possible to know what this firmware brings compared to Repetier ? If it's significant, I may upgrade Smoothieware in the same way.

luc-github, 2016-01-09:
Well, lately I did not tested with latest IDE and latest ESP module they did a lot of changes 
my latest config was IDE 1.6.5 and 2.0.0 (stable),  I know they have several issue raised with 2.1
I remember there is a bug in scan : first entry is empty but should be fixed in Git version not staging.
So could be a bad string content that corrupt heap / overload memory

I am little bit busy right now due to moving from Asia to France and may not be able to check before end of next week, Sorry, but I keep issue open to have a look

Yes my version of repetier is for DUE board based (I have RADDS/DUE/GLCD) and Davinci Board (Due based with special pin mapping), it has custom UI (1 expert / 1 easy), more settings are modifiable from menu and from EEPROM and has some addition to better communicate with ESP8266 module, and a way to switch from serial to wifi from menu

treepleks, 2016-01-09:
Hello Luc,

> Well, lately I did not tested with latest IDE and latest ESP module
> they did a lot of changes
> my latest config was IDE 1.6.5 and 2.0.0 (stable), I know they have
> several issue raised with 2.1
> I [...]
I'm using 2.0.0 too (I didn't clone the ESP repository, just used the
Board manager of Arduino's IDE).
 
> I am little bit busy right now due to moving from Asia to France and
> may not be able to check before end of next week, Sorry, but I keep
> issue open to have a look

No problem. I'm not in a hurry. And may be I will understand what
happens in the mean time. I'm still waiting for mechanical parts to
assemble my printer and I'm taking care of the firmwares in the mean
time. I have somewhat revamped your code and gained some RAM and FLASH
(I was thinking it could be a RAM issue). Still need to flash/check. I
also started to revamp the printer.tpl web page.

BTW, I'm french, leaving in Toulouse :-)
I immediately thought you could be a French speaker (native ?) when I
saw the "oct/O" instead of "byt/B" in the code.
 
> Yes my version of repetier is for DUE board based (I have
> RADDS/DUE/GLCD) and Davinci Board (Due based with special pin
> mapping), it has custom UI (1 expert / 1 easy), more settings are
> modifiable [...]
Ok. I will dig your Repetier repository log to identify ESP related stuff.
Smoothieware listens/writes to several UARTs simultaneously and
independently. It should be easy.

Bon retour !

T.

luc-github, 2016-01-09:
Ho weird it is working on my side but I use 4M/1MSPIFFS, may I know how much memory is displaying in home page and in flash in extra setting page for the ESP 1M/128KSPIFFS 
Yes code need some serious refactoring, now framework is stable should be easier.

The HTML/Javascript in tpl was done a little bit "a l'arrache" because tried several UI to feed the need, and now it is a little messy I agree 

Yes gain ram and flash are key on ESP, need to be lower as possible - feel free to push enhancement, feel alone on this project sometimes ^_^

For repetier try to find WIFI or ESP  words, normally there is a comment to identify my code from repetier each time I add it

Haha French are everywhere, I am from Angers and have family in Toulouse ^_^

Merci

treepleks, 2016-01-09:
Le 10/01/2016 00:11, Luc a écrit :
>
> Ho weird it is working on my side but I use 4M/1MSPIFFS, may I know
> how much memory is displaying in home page and in flash in extra
> setting page for the ESP 1M/128KSPIFFS
>
Free Memory: 27040 octets

 Status: Ok  |  Total space: 110.55 KB  |  Used space: 78.19
KB  |  Occupation:  70%

> Yes code need some serious refactoring, now framework is stable should
> be easier.
>
I have done some non negligible refactoring on webinterface.cpp.
Will fork and push to GitHub when everything is in reasonably good shape.

> The HTML/Javascript in tpl was done a little bit "a l'arrache" because
> tried several UI to feed the need, and now it is a little messy I agree
>
It's Ok. Just want to make it more "sexy" with nice SVG.
>
> For repetier try to find WIFI or ESP words, normally there is a
> comment to identify my code from repetier each time I add it
>
ok.
>
> Haha French are everywhere, I am from Angers and have family in
> Toulouse ^_^
>
Funny.
I'm now from Toulouse (been there for quite a time) and had family in
Angers (they moved to...Brazil).

Going to bed.
T.

luc-github, 2016-01-10:
cool waiting for your PR
well my svg are done using notepad and I am definitly not a designer, so happy sexy design is coming.

Morning here, good night

treepleks, 2016-01-10:
Hello Luc,

Surprised by the fact that it works fine on your side, I decided to reproduce your setup from scratch. I recloned your  github repo, reinstalled Arduino 1.6.5, recompiled the firmware and the SPIFFS, reflashed both.

I have an even more disturbing result: the web site pops up. The ESP kept my previous EPROM settings so it boots in STA mode with the correct static IP. Great !

Now, both the AP and the STA pages are inaccessible:  the ESP freezes (eg. does not answer ping anymore) as soon as I click on either page link after authentification.

Any advice for debugging (beyond inserting Serial.println(F("message")) everywhere :-) ?

I will reflash with my modified (less memory intensive) firmware and see how it does. Possibly, will change the ESP8266 hardware too (since I have more than one :-)

The only explanations I can imagine are either lack of RAM, EEPROM initial values leading to the bug (but I did not change the EEPROM layout) or... hardware fault.

T.

treepleks, 2016-01-10:
Pourrais tu, si tu en as envie, me communiquer ton adresse mail personnelle (à envoyer à "treepleks à free point fr"). J'ai besoin de discuter de choses que je n'ai pas envie de voir sur GitHub (rien d'envahissant et potentiellement intéressant).

luc-github, 2016-01-10:
Indeed very strange - first reset the EEPROM to be sure it is not corrupted using serial command and use also serial command to display the settings:
here the available commands
```
-restart module [ESP888]RESTART
-Get IP (only printer see answer): [ESP111]M117
-reset EEPROM and restart: [ESP444]RESET
-display EEPROM content: [ESP444]CONFIG
-go to safe mode without restart: [ESP444]SAFEMODE
-SSID: [ESP100]<SSID>
-Password: [ESP101]<PASSWORD>
-Station mode: [ESP103]STA
-AP mode: [ESP103]AP
-IP Static: [ESP103]STATIC
-IP DHCP: [ESP103]DHCP
```
there is a debug tool in github of ESP8266 that may help but never used it.

I also suggest to disable extra feature to save space at first
comment :
```
#define MDNS_FEATURE
#define SSDP_FEATURE
#define CAPTIVE_PORTAL_FEATURE
#define AUTHENTICATION_FEATURE
```
in config.h 

No issue for email : luc a tech-hunters point com
efface ton message quand tu auras lu le mien et j'effacerai le mien aussi - il est quand meme preferable de ne pas laisser trainer ces trucs la - au pire dans 2 jours j'efface les 2

luc-github, 2016-01-10:
Just tested 2.0.0 = stable with my nodeMCU flashing as ESP Generic 4M/1M SPIFFS and works as before
Will test 1M/128SPIFFS

Questions : 
do you use resistors and capacitor to connect your ESP as suggested ?
what USB serial adapter do you use ? before people had issue and was cable / adapter issue as I remember

luc-github, 2016-01-10:
Using 1M/128SPIFFS is working as well
But Compilation is currently broken with esp8266 git version - travis compilation failed - I will open an issue as I may not have time to fix it before next week end

treepleks, 2016-01-11:
Luc

I have a reasonably good setup I think, using pull resistors (except for
GPIO2 which is wired to directly to +3.3V/0V to allow for easy
"configuration reset"). I have a 750mA 3.3V regulated power (just for
the ESP), I have added a 470µF chemical for power stabilization (close
to power) as well as a 100nF decoupling (this one soldered on the ESP
pins directly). The only "bad" thing I see is that the ESP is on a set
of 20cm long wires with female Dupont connectors for the ESP and the
breadboard side.

For USB2serial, I use a FT232RL FTDI with the ground shared with ESP (I
do not use the FTDI power output which seems too weak). Directly plugged
on the breadboard.

I reset the EEPROM (and even programmed MD5 hash based consistency
config write/reads/init to detect bad EEPROM that resets it
automatically in this case). Did not help...

I'm tired of these exceptions (3) :-(

Could you send me your two bin files (cpp.bin and spiffs.bin) compiled
for 1MB+128k SPIFFS (with flashing adresses, should be 0x0 and 0xBD000 I
think).

A tar.xz sent by mail would be fine.

I could at least see if my build setup is wrong.

Thanks a lot for your help !

T.

luc-github, 2016-01-11:
Did you try to compile without the 4 feature ? still have exeption 3 ?

luc-github, 2016-01-11:
The guys providing the fix used a 4A fuse so should be Ok

treepleks, 2016-01-11:
Not yet... this evening (working all day at office...)

luc-github, 2016-01-11:
Because if you use the stable json file you cannot have bad setup - so I do not think this is the problem

luc-github, 2016-01-11:
@treepleks May I know where you bought these new ESP01 modules ?

treepleks, 2016-01-11:
> @treepleks <https://github.com/treepleks> May I know where you bought
> these new ESP01 modules ?
>
One was bought on eBay and 2 others on AliExpress. All have the very
same look. The flash device 4014h and unknown flash vendor,
8Mbit flash and they are black below the circuit board. The upper part
is written " AI-Cloud Inside".
If you have an advice for a good place to buy ESP01, please tell me.
I would like to use an ESP01 and not higher version just because their
final destination is a board that has a dedicated location for an ESP01.

I have tried the 3 ones previously: the AT firmware works fine apparently.

I have recompiled your firmware (pure from GitHub except for initial IP
address and commented lines : no mDNS, authentication and captive
portal. I did not disable SSDP because it generates a compilation error
: error on WifiUDP, I will look into this later - during the week, I
work quite intensely during the day and I like to rest in the evening :-)

Flashed at 57600 bauds under windows using the Flash tool v2.4 (150924).

Same problem:

Home is Ok
System is Ok
The first click on Access point generates an exception (3), a stack dump
and a reboot.
Same thing with Station. same value of EPC1 and EXCVADDR as above.
A click on 3D printer just gives no reaction, but no reboot.

Both exceptions appear instantaneously. As soon as I click on the menu.
Does not feel like a WDT interrupt ?

Sniff,

T.

luc-github, 2016-01-11:
Does 1M look like this one :
http://goods.ruten.com.tw/item/show?21528084758112

Brokly, 2016-01-12:
Hello everybody ! I bought here: http://www.aliexpress.com/item/Free-shipping-5pcs-lot-ESP8266-serial-WIFI-wireless-module-wireless-transceiver/2038011896.html . No problems there, everything works fine [1M (512K SPIFFS).] But it is necessary to use very short wires and requires a powerful source of 3.3 volts (to receive the necessary power through the Arduino will not work). Sorry for my english :)

PS: Excellent project :) I added to it the ability to record to SD and work with EEPROM, but I use Marlin and RAMPS.

luc-github, 2016-01-12:
ok I will buy a couple to test them

mundsen, 2016-01-12:
Hi, great work
I have made most of the RADDS documentation at http://doku.radds.org - next I plan to add some information about how to add WiFi, using your code. One alternative is to build a module like in your description, but I have been searching for a close to "plug and play" alternative.
I did order this module from ebay: http://www.ebay.com/itm/Serial-Wifi-Module-ESP8266-ESP-12-for-Arduino/271981074883?_trksid=p5197.c100068.m2280&_trkparms=ao%3D1%26asc%3D20140211125758%26meid%3Dadf67e1e47f94194934a88878c372611%26pid%3D100068%26clkid%3D2550341474880950232&_qi=RTM2016704

Hoping that it could be connected to the serial on the RADDS EXT1 or on the serial port on the RADDS Extension board. This without having to solder resistors..

Does this module seem ok?

Best regards
Oyvind

luc-github, 2016-01-12:
Nice doc indeed
What is flash size ? Did not saw it

mundsen, 2016-01-12:
Thank you. It is an esp-12, how do I find flash size?

luc-github, 2016-01-12:
well if not mentioned there is a risk it is 512k,  it should have a nodemcu fw by default , so it should have a function to give the flash size

luc-github, 2016-01-12:
there is a thread here http://www.esp8266.com/viewtopic.php?f=13&t=2506

mundsen, 2016-01-12:
Do you know about a module like this that have more flash?

luc-github, 2016-01-12:
nope sorry but you may contact the saler - this small board looks really nice
Normally 12E have 4M of flash

mundsen, 2016-01-12:
Sent a mail to the seller, I`ll tell what flash size it has if/when I get an answer
I found this as an alternative - here you can select what ESP module to use
http://www.electrodragon.com/product/esp8266-smd-adapter-board/

mundsen, 2016-01-12:
some info about 12f:
http://www.electrodragon.com/product/esp-12f-esp8266-wifi-board/

Breakthrough design, there are new breakthroughs, based on ESP-12E.
The new four-layer board design, the new revision antenna RF performance optimization, communication distance is increased by 30% -50% compared to ESP-12E!
Semi-hole chip technology, the whole IO leads, with metal shielding shell
Passed FCC & CE & RoHS certification
Built-board PCB antenna
4M bytes Flash.
Pins and footprint are fully Compatible with ESP-12E and our relevant breakout board.

luc-github, 2016-01-13:
looks good

luc-github, 2016-01-13:
looks good - I think I will give a try to this later

one comment about this board, you will loose the recovery by jumping pin 2 or you must do some additional soldering.

luc-github, 2016-01-13:
ok just bought 2 new esp 01 in my local shop, will see if I can reproduce issue

I am also wondering if OTA will have enough space, I guess to have ota working 128kSPIFFS is the maximum

mundsen, 2016-01-13:
Can you explain "loose the recovery" and what I must do not to loose it?
Looking forward to try your code.

luc-github, 2016-01-13:
if you loose ap password, or if your EEPROM is corrupted for any reason, connecting pin 2 to ground 1s after starting module will reset EEPROM and so wifi settings to default one, just need a jumper or switch between pin 2 and Gnd

it is in Readme for more details

luc-github, 2016-01-13:
OK 
got this when connect and use reset command to get first boot :
```
AI

ERROR
AT+RST

OK

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

rl��rl��
Ai-Thinker Technology Co.,Ltd.

ready
```
so it is ESP01 new generation

luc-github, 2016-01-13:
After test module is working  properly in AP mode and STA mode so problem is likely hardware and may be as @Brokly suggested it could be a power supply issue

One note : OTA test failed - I did not dig in yet and will open issue but I suspect not enough size on flash for this

treepleks, 2016-01-13:
Le 11/01/2016 22:07, Luc a écrit :
>
> Does 1M look like this one :
> http://goods.ruten.com.tw/item/show?21528084758112
>
Yes. Precisely.

More this week end (week too busy :-)

T.

treepleks, 2016-01-13:
Le 13/01/2016 08:22, Luc a écrit :
>
> ok just bought 2 new esp 01 in my local shop, will see if I can
> reproduce issue
>
> I am also wondering if OTA will have enough space, I guess to have ota
> working 128kSPIFFS is the maximum
>
>

Yes. That's why I used 128K SPIFFS.

Thomas

treepleks, 2016-01-13:
Le 13/01/2016 11:50, Luc a écrit :
>
> After test module is working properly in AP mode and STA mode so
> problem is likely hardware and may be as @Brokly
> <https://github.com/Brokly> suggested it could be a power supply issue
>
Unlikely I think (I may be wrong of course). I use external 0.75A
regulated 3.3V power, itself powered from a stronger 2A/12V source. It's
neither the arduino power source nor the FTDI power (which would be too
weak). The max consumption of an ESP8266 is around 215mA.
http://wiki.iteadstudio.com/ESP8266_Serial_WIFI_Module

May be more at 160MHz ?

The length of the wires seems a more likely source of trouble (despite
the decoupling capacitor I soldered directly on the module).

Don't spend more time/money on this. I will dig the hardware aspect on
my side during week-ends when I find time (and contact you back when/if
the issue is solved).
 
Thanks a lot for testing.
Sorry for bothering :-)

> One note : OTA test failed - I did not dig in yet and will open issue
> but I suspect not enough size on flash for this
>

Mmm. Should work in theory :-)
The firmware is less than 350KB * 2  < 700Kb leaves > 300Kb for SPIFFS
and EEPROM according to the memory map.

T.

luc-github, 2016-01-14:
I am connect to my desktop PC with my serial adapter see the wires, the resistors are soldered and  covered according the needs :
 
[image]

About OTA yes in theory - I am just guessing as it works with ESP12E, but actually I have another guess as this ESP01 is more reactive than my other old ESP01, the response time is really shorter compare to my old module, and it failed almost immdiately, so I also suspect a timing issue, but I need to be in debug mode and may not have time before this week end.

luc-github, 2016-01-14:
OTA issue is not linked to ESP 01  module but code : https://github.com/luc-github/ESP8266/issues/67

treepleks, 2016-01-16:
Le 14/01/2016 02:46, Luc a écrit :
>
> I am connect to my desktop PC with my serial adapter see the wires,
> the resistors are soldered and covered according the needs :
>
> p_20160114_093746
> <https://cloud.githubusercontent.com/assets/8822552/12313505/f145d7de-baa2-11e5-8a17-f777c6928e6d.jpg>
>

Ok. My setup is almost the same, using a breadboard and an external
power source (much stronger then USB). Actually I have no long wires as
you have as I now directly plugged the ESP on the breadboard. I also
changed the power source and the FTDI. I tried to flash in QIO and DIO
(both Ok), I tried different the flashing baud rate (even very low and
very high: all Ok)...

But it still does not work in AP and STA pages (only). Always the same
exception.

It's hard to imagine that 3 different modules, coming from 2 different
sources have all the same problem.
I will dig again ! I feel tired now, will have a beer :-)

Where did you buy yours ?

Anyway...

I have forked your repository and committed (all the changes I had done
so far trying to reduce the memory usage (I did not commit and push the
MD5 hash based EEPROM checking but I can if you want it -- takes some
Flash and cpu time). I will send a PR. Take what you want. Everything
has not been thoroughly tested of course since my ESP does not
completely works (but most of it has been tested).

I tried to do atomic commits so that it would be easy to see what I did
at each step.

T.

treepleks, 2016-01-16:
Luc,

BTW, my fork has the surprising advantage that the "Access point" page
does now work on my (supposedly faulty) hardware. But still, the Station
page generates an Exception (3).

I cannot avoid having the feeling that the problem is not in the
hardware. Power or wire problems should not be so deterministic and
destroyable by changing the software. Will try to reflash to see if it's
the flash that went better than usual.

Which platform/OS are you using to compile ? Mac, Windows or Linux ?

I don't see why it could make a difference, but I have looked in so many
areas... I would really be interested in the two flashable.bin files
(firmware and SPIFFS) that work on your ESP01 in 1MB (128K SPIFFS) mode.

T.

luc-github, 2016-01-17:
OK will check this week and send my current bin file
I bought 2 module here in local shop in taiwan and order 2 others with complete ftdi solution also in taiwan
I compile under windows as got too many issue with latest OSX.

luc-github, 2016-01-17:
I remember before I embeded css in progmem and did some typo issues, the result was when using ie browser esp module crashed but no issue with chrome.

also I have applied a patch on my win7and win10, to speed up  display under chrome and firefox, but should not affect the esp, just rendering speed

treepleks, 2016-01-17:
Le 17/01/2016 03:46, Luc a écrit :
> I remember before I embeded css in progmem and did some typo issues,
> the result was when using ie browser esp module crashed but no issue
> with chrome.
>
> also I have applied a patch on my win7and [...]
Not sure it corresponds to what you mean but on the ESP8266 arduino
forum, the author mentions he introduced send_P variants that allow to
directly send from PROGMEM. Found several of your posts there by the way :-)

treepleks, 2016-01-17:
> I compile under windows as got too many issue with latest OSX.
>
I'm doing everything under Linux (ubuntu/wily). The cross-compilers
should not be different, but theory and practice sometimes differ.
Will install Arduino under VirtualBox/Windows. Still, would be happy to
get your "bin" files (just to kill an hypothesis definitely).

BTW, I reflashed with the version that gave a working "Access Point"
page and it stopped working. So, this is not purely software :-/
But my breadboard is now very clean. So, I can only imagine a problem in
my power sources (despite the fact that it is super stable at 3.38V) and
then my modules.

I have ordered a professional regulated power source. Should be here
next week. Will be useful for other projects anyway.
After this, I will swallow my ESP modules and order new ones (if I
survive) !

luc-github, 2016-01-17:
well using sendP is another approach, put css as file not embeded in web page, I canceled this idea because it start to be complex when restart command is sent and stop serving webserver, page  will display but may be some or full css won t be uploaded to browser, giving some timeout or ugly display

luc-github, 2016-01-17:
@treepleks  I just sent to your free account the 2 binaries

treepleks, 2016-01-17:
Surprise: it works 100% fine with your binaries :-)
What a relief !

Apparently, either I have a badly configured Arduino suite or there is a
problem with the Linux version. This is puzzling, but since I used the
"Board manager" version, I incline towards the latter. Will dig first
and then transmit to the Arduino ESP8266 team if needed.

T.

luc-github, 2016-01-17:
as you used several version of module there could be a mix in your files, as for me disabling SSDP is working when you have compilation error, so I think you need to redo ESP install
be sure everything is clean when you install with board manager, delete all others modules or even files related to ESP in the hidden directory of arduino then install again the stable module

treepleks, 2016-01-17:
Yep. I will !

For now, I'm installing a Windows based Arduino development chain.

Will shift to a clean install of Arduino 1.6.5+ESP 2.0.0+SPIFFS on Linux
just afterwards and compare. But I'm confident I had already removed
Arduino and .arduino{15} folders before.

BTW, I reflashed the ESP with your binaries, just to see if I had not
been just granted with a "lucky flash" and apparently no: it still works
fine after a reflash. Another relief :-)

T.

treepleks, 2016-01-17:
Luc,

I have now installed Arduino 1.6.5 + ESP 2.0.0 + SPIFFS manager under a
VirtualBox running Windows 7.
I have a brand new Arduino 1.6.5 installed with the same packages, exact
same versions under Linux.

I compiled and flashed from Windows and the resulting binaries run fine.
I compiled and flashed from Linux and the resulting binaries generate
exceptions, as before.

However, I have still a fundamental discrepency with you:

On a clone of your repository, under Windows, if I disable SSDP, mDNS,
authentification and Captive Portal in config.h, the firmware does NOT
compile unless I add the WifiUDP include file in webinterface.cpp as I
did in my repository. I instead get the following error:

C:\Users\tom\AppData\Local\Temp\build1734870594672797611.tmp\webinterface.cpp:
In function 'void WebUpdateUpload()':
C:\Users\tom\AppData\Local\Temp\build1734870594672797611.tmp\webinterface.cpp:2624:2:
error: 'WiFiUDP' has not been declared
  WiFiUDP::stopAll();
  ^
Error compiling.

Could you please recheck this does not happen on your side ?
Please, if you do, disable all 4 options as I did, as another than SSDP
(eg. mDNS) may still inderctly include WifiUDP on its own.

T.

luc-github, 2016-01-17:
Ok I can reproduce issue if I comment 2 features :
···
//MDNS_FEATURE: this feature allow  type the name defined
//in web browser by default : http:\\esp8266.local and connect 
//#define MDNS_FEATURE

//SSDD_FEATURE: this feature is a discovery protocol, supported on Windows out of the box
//#define SSDP_FEATURE
···

treepleks, 2016-01-17:
Good. Thanks for your help.

I posted an issue on ESP8266 for arduino as there are visible
differences between the Linux and Windows ESP packages (eg. different
versions of include files in the tools folder).

T.

luc-github, 2016-01-18:
OK - I have changed the label to SDK issue

luc-github, 2016-01-19:
I have changed title to better reflect issue

mundsen, 2016-01-22:
Hi, I do not know if you want this here, or somewhere else?
This module arrived yesterday: http://www.ebay.com/itm/271981074883?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

When I connect it to arduino, it says ESP-12E,4M so then it seems like the flash is ok on this module

luc-github, 2016-01-23:
It is not an hardware issue but seems module issue as it happen only on linux version
It may be a compilation problem as core is a zip file identical for all OS just compilation tools (binaries / libraries) are different according the OS they are used

treepleks, 2016-01-23:
Would you share your code ?
It would likely be trivial to integrate it (at least the SD recording
ability) here.

Le 12/01/2016 06:42, Brokly a écrit :
> I added to it the ability to record to SD and work with EEPROM, but I
> use Marlin and RAMPS.

[9 more replies]

closed 2016-05-10 by luc-github