luc-github/ESP3D #65: Compatibility issue with ESP8266 Git version

luc-github, 2016-01-10:
Travis failed:
```
/tmp/build5286045909151746881.tmp/wifi.cpp: In member function 'bool WIFI_CONFIG::Setup()':
/tmp/build5286045909151746881.tmp/wifi.cpp:154:25: error: 'sleep_type' was not declared in this scope
   wifi_set_sleep_type ((sleep_type)bflag);
                         ^
/tmp/build5286045909151746881.tmp/wifi.cpp:175:26: error: 'phy_mode' was not declared in this scope
       wifi_set_phy_mode((phy_mode)bflag);
                          ^
/tmp/build5286045909151746881.tmp/wifi.cpp:205:22: error: 'phy_mode' was not declared in this scope
   wifi_set_phy_mode((phy_mode)bflag);
                      ^
Error compiling.
Failed to build esp8266/esp8266.ino
```

Some enum definition have been moved to another place ?

luc-github, 2016-01-10:
seems type has been changed to 
phy_mode_t
and sleep_type_t

luc-github, 2016-01-10:
Need to wait git version is moved to stable version because doing modification now will make incompatible with current stable version or need to add some define which will not be necessary when git become stable 

So fix is on hold until these 2 enum are in stable version : 2.1 or more

luc-github, 2016-02-27:
2.1.0 is now out - it is time to try it http://arduino.esp8266.com/versions/2.1.0/package_esp8266com_index.json

luc-github, 2016-02-27:
with 2.1.0 when safe mode is starting the module crash to exeption 3

luc-github, 2016-02-27:
exception 2 also after reset EEPROM and starting as Access point

luc-github, 2016-02-27:
disabling all feature allow to start AP - will check which feature(s) generate the crash

luc-github, 2016-02-27:
crash at startup is linked to 
TCP_IP_DATA_FEATURE, if only this one is disabled, module seems fine

luc-github, 2016-02-27:
doing a simple sketch  validate the possibility of having 2 ports is still working 
so should be something in implementation code
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "NETGEAR_2GEXT_OFFICE";
const char* password = "BUREAUCHAMBRE";

ESP8266WebServer server(80);
ESP8266WebServer server2(8888);

void handleRoot() {
  server.send(200, "text/plain", "hello from esp8266!");
}

void handleRoot2() {
  server2.send(200, "text/plain", "hello from esp8266!");
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void){
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);
  server2.on("/", handleRoot2);

  server.onNotFound(handleNotFound);

  server.begin();
  server2.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
  server2.handleClient();
}
```

luc-github, 2016-02-27:
Issue looks linked to presence of 
```data_server->setNoDelay(true);```
Commenting it and module seems fine

Need long run test - additionally reaction time looks slower than with 2.0.0 but need more tests to confirm

luc-github, 2016-02-28:
seems 2.1.0 do not solve issue
https://github.com/luc-github/ESP8266/issues/80

luc-github, 2016-02-28:
travis - read me are up to date 
Still need to review all functions one by one to confirm it is Ok and no side effect, like on web update for example

luc-github, 2016-02-28:
exception(2) issue is reported : https://github.com/esp8266/Arduino/issues/1695

luc-github, 2016-03-18:
seems module is definitly slow and now new issue appeared https://github.com/luc-github/ESP8266/issues/85

luc-github, 2016-03-25:
the setNoDelay issue is fixed in git so will be part of 2.2.0

luc-github, 2016-04-08:
all fixes have been implemented - anyway 2.1.0 must not used due to critcal bugs, performances issues and so usability issue
2.2.0 should be the next stable release

closed 2016-04-08 by luc-github

mkeyno, 2016-04-10:
dear @luc-github  can you check[ this issue](https://github.com/esp8266/Arduino/issues/1889#issuecomment-207971118) ,same sketch and file have different behavior for same module, one of them quickly respond the browser request and other dumb to any request or partially respond

luc-github, 2016-04-10:
do you have same inconsistent behaviour with pure sample and none of your modifications ?

mkeyno, 2016-04-10:
yes , strange is one sketch totally work with single module and sometimes not work with another same module , in my case test it with nodemcu and esp-07 also check with different IDE, 

following is the setup and loop code , the only different is the modules have different EEPOM and may restore different parameters in setup but the other part is same
`
void setup()
{

Serial.begin(115200);

pinMode(0,OUTPUT); digitalWrite(0,HIGH);
pinMode(2,OUTPUT); digitalWrite(2,HIGH);
SPIFFS.begin();
EEPROM.begin(128);

MyMode=EEPROM_Read(0,1);

myHostname =EEPROM_Read(1,15); //add 1-14
SSID=EEPROM_Read(15,30); //add 15-29
ssidPass=EEPROM_Read(30,45);//add 30-44
AP_SSID=EEPROM_Read(45,60);//add 45-59
AP_ssidPass=EEPROM_Read(60,75);//add 60-74
IDsNum=EEPROM_Read(75,76).toInt();//add 75
reset_number=EEPROM_Read(76,77).toInt();//add 76

Serial.print("Mode="); Serial.println(MyMode);
Serial.print("Hostname="); Serial.println(myHostname);
Serial.print("SSID="); Serial.println(SSID);
Serial.print("Password ="); Serial.println(ssidPass);
Serial.print("AP SSID="); Serial.println(AP_SSID);
Serial.print("AP Password ="); Serial.println(AP_ssidPass);
Serial.print("reset_number ="); Serial.println(reset_number);
Serial.print("Server IP Address ="); Serial.println(S_IP);
if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
{
Serial.println("\t\t\t Entering the Access Mode");
WiFi.mode(WIFI_AP);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
String temp_name="Smart Node_" +String(ESP.getChipId(), HEX);
if(AP_SSID.length()<3) WiFi.softAP(temp_name.c_str(),"12345678");
else WiFi.softAP(AP_SSID.c_str(),AP_ssidPass.c_str());

delay(500); // Without delay I've seen the IP address blank
EEPROM_Write(String(0),76,77);

PerfixID=WiFi.softAPIP().toString().substring(8);
Serial.print("AP IP address: ");Serial.println(WiFi.softAPIP());
dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
dnsServer.start(DNS_PORT, "", apIP);

}
else if(reset_number<5)
{
Serial.println("\t\t\t Entering the Station Mode");
WiFi.mode(WIFI_STA);
WiFi.begin(SSID.c_str(), ssidPass.c_str());

unsigned long startTime = millis();
while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000) {Serial.write('%');delay(500); }
if(WiFi.status() != WL_CONNECTED) {
reset_number++;
EEPROM_Write(String(reset_number),76,77);
ESP.restart();
}

PerfixID=WiFi.localIP().toString().substring(8);
Serial.print("AP IP address: ");Serial.println(WiFi.localIP());
if (!MDNS.begin(myHostname.c_str())) {
Serial.println("Error setting up MDNS responder!");
while(1) {
delay(1000);
}
}
Serial.println("mDNS responder started");
}
server.on("/process", process);

server.on("/update", HTTP_POST, { Serial.println("Upload post"); / server.send(200, "text/plain", ""); / Redirect("setting"); }, handleFileUpload);
server.on("/login", handleLogin);
server.on("/check", check); //check user name pass word link
//server.on("/", { if(is_authentified()) Redirect("/"); else redirect2login(); }); //check user name pass word link
server.onNotFound(handleOther);
server.begin();
Serial.println("\t\t\tHTTP server started");

//here the list of headers to be recorded
const char * headerkeys[] = {"User-Agent","Cookie"} ;
size_t hz = sizeof(headerkeys)/sizeof(char);
//ask server to track these headers
server.collectHeaders(headerkeys, hz );

webSocket.begin();
webSocket.onEvent(webSocketEvent);
}

void loop()
{

if(MyMode[0]=='A') dnsServer.processNextRequest();
server.handleClient();

webSocket.loop();

}

`

luc-github, 2016-04-10:
this do not look like pure sample from core, I suggesr you to start with pure sample and add every modification you did one by one and test each one to see which one is causing the problem

mkeyno, 2016-04-10:
I did with library example  but when I put the some file in SPIFF , seem problem start to begin, but the problem is why the code work fine with one module and not work correctly with same another hardware  , I've just doubt about my laptop network setting which is pass the browser request to one and not to other , most the time seems my AP node cant hear anythings from browser request

luc-github, 2016-04-10:
then share the limited code not the full one check if 1K or less file
generate same behavior

quality/speed of flash memory may affect behavior also

mkeyno, 2016-04-10:
with this Serial.setDebugOutput(true); ?

luc-github, 2016-04-10:
and adding additionnal ones to better narrow

mkeyno, 2016-04-10:
out put from  not respond module 
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
.......................................endless

output from good module 

chg_A2:-40
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
 Request page : /LmacRxBlk:1
LmacRxBlk:1
 Request page : LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
chg_A2:-80
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
 Request page : /login.html	[SENT]
 Request page : /login.htmlLmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
	[SENT]
 Request page : /WIFI.htmlLmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
	[SENT]
 Request page : /func.js	[SENT]

mkeyno, 2016-04-10:
is there better way to debug the deeper level of module ?

luc-github, 2016-04-10:
need to add additionnal debug messages in your sketch to confirm but looks like your module even do not receive http request so not related to SPIFFS if not yet used for answer

I suggest you verify it to narrow down the problem

mkeyno, 2016-04-10:
dear @luc-github can you tell how can I add more specific debug messages in my sketch, for example #define DEBUG_WIFI

luc-github, 2016-04-10:
Serial.print(”my debug message”）;

mkeyno, 2016-04-10:
I have such messenger in my code specially when got any request or important function but I mean print out more information about core module , I think  Serial.setDebugOutput(true); not shows proper information as the  debug setting level define in  Arduino/tools/Generic ESP8266 board , this option is miss when I choose NodeMCU , is there way that I could enable it for my NodeMCU?

mkeyno, 2016-04-10:
output of not respond module
`
wifi evt: 7
chg_A2:-40
wifi evt: 7
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
wifi evt: 5
:urn 43
:urch 43, 50
:urch 93, 35
:urch 128, 48
:urch 176, 51
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
...................................................endless
`

mkeyno, 2016-04-10:
dear @luc-github  I find the problem , in  setup function I restore couple of parameter from EEPROM and when I use the new hardware the restore parameters is empty , the hardware with preset EEPROM  work fine but the new hardware not , I disable this restore function and module now accept browser request , but somethings is still strange , the restore parameters has got nothings to do with any other function and wifi setting but brutally affect the module performance , can you kindly check the following code and let me know your comments 

`
void setup()
{

Serial.begin(115200);
SPIFFS.begin();
EEPROM.begin(128);

                                                       MyMode=EEPROM_Read(0,1);
                                                        myHostname =EEPROM_Read(1,15); //add 1-14
                                                      SSID=EEPROM_Read(15,30); //add 15-29
                                                      ssidPass=EEPROM_Read(30,45);//add 30-44
                                                      AP_SSID=EEPROM_Read(45,60);//add 45-59
                                                      AP_ssidPass=EEPROM_Read(60,75);//add 60-74

if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
{
Serial.println("\t\t\t Entering the Access Mode");
WiFi.mode(WIFI_AP);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
String temp_name="Smart Node_" +String(ESP.getChipId(), HEX);
if(AP_SSID.length()<3) WiFi.softAP(temp_name.c_str(),"12345678");
else WiFi.softAP(AP_SSID.c_str(),AP_ssidPass.c_str());
delay(500); // Without delay I've seen the IP address blank
EEPROM_Write(String(0),76,77);
PerfixID=WiFi.softAPIP().toString().substring(8);
Serial.print("AP IP address: ");Serial.println(WiFi.softAPIP());
dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
dnsServer.start(DNS_PORT, "", apIP);

}
else if(reset_number<5)
{
Serial.println("\t\t\t Entering the Station Mode");
WiFi.mode(WIFI_STA);
WiFi.begin(SSID.c_str(), ssidPass.c_str());
unsigned long startTime = millis();
while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000) {Serial.write('%');delay(500); }
if(WiFi.status() != WL_CONNECTED) {
reset_number++;
EEPROM_Write(String(reset_number),76,77);
ESP.restart();
}

      PerfixID=WiFi.localIP().toString().substring(8);
      Serial.print("AP IP address: ");Serial.println(WiFi.localIP());
      if (!MDNS.begin(myHostname.c_str())) {
      Serial.println("Error setting up MDNS responder!");
      while(1) {
                delay(1000);
                          }

}
Serial.println("mDNS responder started");
}

server.on("/process", process);
server.on("/login", handleLogin);
server.on("/check", check); //check user name pass word link
server.onNotFound(handleOther);
server.begin();
Serial.println("\t\t\tHTTP server started");
//here the list of headers to be recorded
const char * headerkeys[] = {"User-Agent","Cookie"} ;
size_t hz = sizeof(headerkeys)/sizeof(char);
//ask server to track these headers
server.collectHeaders(headerkeys, hz );
webSocket.begin();
webSocket.onEvent(webSocketEvent);
}

void loop()
{

if(MyMode[0]=='A') dnsServer.processNextRequest();
server.handleClient();
webSocket.loop();

}

`

luc-github, 2016-04-10:
I do not know ```EEPROM_Read``` but I guess you did not have any check to avoid buffer overflow and ensure all your strings have zero terminal - these things bring unpredicted behaviors

mkeyno, 2016-04-10:
this is functions
String EEPROM_Read(byte from, byte to)
{
  String s="";
  for (byte i = from; i < to; i++)
                  {
                  byte c=EEPROM.read(i);
                  if(c>31 && c<128)
                  s += char(c);
                  }
  return s;
}

void EEPROM_Write(String s,byte from, byte to)
{
for (byte i = from; i < to; i++) EEPROM.write(i, 0); //earase
for (byte i = from; i < from+s.length() && i < to; i++) EEPROM.write(i, s[i-from]);
EEPROM.commit();
}

mkeyno, 2016-04-10:
but how this blind the module to read the browser request , all restored parameters not used in wifi setting

luc-github, 2016-04-10:
for me your read logic is not correct, it do not ensure the data is correct
and even do not handle the read is failing or do not read garbage

just my 2cents

mkeyno, 2016-04-10:
my idea was to read just readable scope of ascii code but you may right and I don't know  what would be best way to read form EEPROM , I'm gonna pre fill them before program any new hardware

luc-github, 2016-04-10:
the usage is to have a magic bit to detect the content is already initialized, if magic bit is not correct then write default values first then read - if not read values as they are

luc-github, 2016-04-10:
on my project I check the value of baud rate if value is not one of the supported ones then EEPROM is not correct and I write default values

mkeyno, 2016-04-10:
dear @luc-github  can you kindly send me link to those code , I want to use your example

luc-github, 2016-04-10:
https://github.com/luc-github/ESP8266/blob/master/esp8266/esp8266.ino#L73-L108

mkeyno, 2016-04-10:
thanks @luc-github and so sorry for this trouble , I'm gonna use your approach but believe me buddy  code of guys like you is so scary for novice like me

luc-github, 2016-04-11:
I have cleared thread to keep focus
After 3 days running non stop on monitoring printer page, not even 1 crash on nodeMCU, so now it is solid rock as 2.0.0 with current git version, tested all features and looks ok so far
will start long test run on ESP12 on my printer today to confirm status

luc-github, 2016-04-12:
after 24h monitoring printer on ESP12E and even monitoring a 20min print - still no crash so I can confirm it fine until new test on final 2.2.0