luc-github/ESP3D #78: Allow to disable or authenticate data port

luc-github, 2016-02-01:
add a compilation directive to add authentication or disable fully the data commands
use same login password as web

luc-github, 2016-02-01:
for safety purpose

luc-github, 2016-02-02:
Disable feature is now possible https://github.com/luc-github/ESP8266/commit/e4af1950eb5c1efbd31f6e83b6f809e66f23df8f
For authentication - implementation problem com from host - repetier do not have authentication protocol - just raw data 
One solution is to open a telnet session using same port and if authentication succeed IP is granted until last packet from this is delayed of certain amount of time - but not really smooth

luc-github, 2016-02-02:
Need to find a proper approach - but now I am dry on it

mkeyno, 2016-04-27:
hi @luc-github , I know you have busy time but I have question about the `bool ESP8266WebServer::authenticate(const char * username, const char * password)` and really appreciated for your help 
actually I intend to overload authenticate function to return decoded base64 input username and password so I could check it against my user database(all accepted user-pass data) , I've check [lib64/cdecode.c](https://github.com/esp8266/Arduino/blob/633e48f3aec5f1c3c11d4498fc90d378d49e6e9f/cores/esp8266/libb64/cdecode.c) header file but I can't figure out how to decode base64 user-pass data
can I have your idea about how to do that?
BR

luc-github, 2016-04-27:
the usage is not to decode what it is in header but to encode what you accept as user:password and to check if they match like done here : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp#L108

that is why there is no decode function in https://github.com/esp8266/Arduino/blob/633e48f3aec5f1c3c11d4498fc90d378d49e6e9f/cores/esp8266/libb64/cencode.h

mkeyno, 2016-04-27:
so what should I do? how can use ESP8266WebServer::authenticate function than only accept one user-pass , ? can I convert allowed user's-pass and then compared with encoded header ? any help really appreciated

luc-github, 2016-04-27:
do a loop testing all couple user:password you granted - no ?

mkeyno, 2016-04-27:
how about this temple function 
retrieve encoded user-pass form browser header
`
String ESP8266WebServer::encoded_user_pass(void){
  if(hasHeader(AUTHORIZATION_HEADER)){
										String authReq = header(AUTHORIZATION_HEADER);
										if(authReq.startsWith("Basic")){
																	  authReq = authReq.substring(6);
																	  authReq.trim();
																	  
																	}
										authReq = String();
										return authReq;
									  }
  return "";
}
`

convert user-pass string to base64 encoded string
`
String   convert_user_pass2encode(String _username, String _password)
{
	String _encode;
/*I dont know which function can convert*/
	return _encode;
}
`
compare two encoded string from either preset user-pass or store data in flash
`
bool check_user_pass(String _encoded_user_pass)
{
  if( convert_user_pass2encode(preSetUser,preSetPass).equals(_encoded_user_pass) )  return true;
  File f = SPIFFS.open("/data/user.dat", "r");
   if(!f) return false;
   while(f.available()) {    
						  String line = f.readStringUntil('\n'); 
						  if(line.length()>2){
												int commaIndex = line.indexOf(',');
												int secondCommaIndex = line.indexOf(',', commaIndex+1);
												String U = line.substring(0, commaIndex); U.trim();
												String P = line.substring(  commaIndex+1, secondCommaIndex); P.trim();											
												if(if( convert_user_pass2encode(U,P).equals(_encoded_user_pass) )) {  f.close();  Serial.println("\t\t[PASS OK]"); return true; }
											   }
						  } 
     f.close(); 
   return false;
}
`

luc-github, 2016-04-27:
1 - won't recommend to store username:password on SPIFFS unless you add some protection to the file to be accessible from browser, to store in EEPROM is more simple and little bit safer IMHO
2 - yes you got the point
3 - can you use gitter (mine or ESP8266) to ask questions unrelated to ticket itself ? It is hard to keep good tracking of issues if something unrelated is discussed in same thread. ESP8266 Gitter is very active mine less but open anyway  
I do not want to be rude but  this will avoid me to delete posts to keep thread clean 
Thanks

mkeyno, 2016-04-27:
its ok buddy , I'm so sorry for any inconvenience , I'll do call you in gitter

luc-github, 2016-07-04:
as there is no software using the authenticate feature for data port - I guess we are good and no need 
Authentication of data port is not necessary but may go with websocket 
so as feature can be disabled - lets close issue

closed 2016-07-04 by luc-github