luc-github/ESP3D #79: bool WIFI_CONFIG::Setup()': 'sleep_type'& 'phy_mode' was not declared in this scope

mkeyno, 2016-02-19:
hi buddy , I cant find this 2 function in wifi.cpp and face the following error , could you help me out 
I use the Arduino 1.6.4 with last staging version of ESP8266
 

\wifi.cpp: In member function 'bool WIFI_CONFIG::Setup()':
\wifi.cpp:161:27: error: 'sleep_type' was not declared in this scope
     wifi_set_sleep_type ((sleep_type)bflag);
                           ^
\wifi.cpp:181:28: error: 'phy_mode' was not declared in this scope
         wifi_set_phy_mode((phy_mode)bflag);
                            ^
\wifi.cpp:217:28: error: 'phy_mode' was not declared in this scope
         wifi_set_phy_mode((phy_mode)bflag);
                            ^
Error compiling.

and one more this , are you aware of awesome work that some cool guys doing this
https://github.com/openhardwarecoza/LaserWeb

treepleks, 2016-02-19:
Don't use the last staging version but version 2.0  (the name of the
types have changed in 2.1). Alternatively shift to the new type names
(something like sleep_t and phy_t ???? just check in the header files).

T Reepleks

mkeyno, 2016-02-19:
ok , tnx , but have you check the this https://github.com/openhardwarecoza/LaserWeb
they used the serial interface to connect the printer , I think they should join you to create wifi transparent link

luc-github, 2016-02-19:
Staging version is not stable yet do not use it for managing printer, there are several issues that generate crash and generate debug output that could ruins your print. So please use 2.0.0

I am waiting the staging get more stability to port the current code to 2.1 or upper.

Thanks for the link, I plan to build a milling cnc and a laser one, so I will check for sure,

luc-github, 2016-02-19:
I close this issue as duplicate of https://github.com/luc-github/ESP8266/issues/65

closed 2016-02-19 by luc-github