luc-github/ESP3D #80: stack fail when try to access on station or access point menu

giutrec, 2016-02-27:
Hi, i've looks a problem when i try to open station or access point menu the system crash and reboot,
on serial print this
```
Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024100f depc=0x00000000

ctx: cont 
sp: 3fff1190 end: 3fff15b0 offset: 01a0

>>>stack>>>
3fff1330:  3ffe87c6 40240940 3fff1350 402167b4  
3fff1340:  3ffe87c6 40240940 3fff1394 4020d5ee  
3fff1350:  3fff5510 0000001f 00000014 00000004  
3fff1360:  3ffe87c6 3ffeffe7 00000000 4020b972  
3fff1370:  ffffffff ffffffff 3ffe8c80 00000020  
3fff1380:  3fff4ac8 3fff54d8 00000000 00000000  
3fff1390:  40102600 ffffffff ffffffff 3ffe8c80  
3fff13a0:  00000021 3fff4ab0 3fff54f0 00000000  
3fff13b0:  00000000 3ffedf00 402410e1 402410d9  
3fff13c0:  402410dd 402410e1 00000001 00000002  
3fff13d0:  00000003 ffffffff 3fff5b10 0000001f  
3fff13e0:  00000011 3fff5af0 0000000f 00000001  
3fff13f0:  3fff4a38 0000000f 0000000b 3fff4a18  
3fff1400:  0000000f 0000000d 3fff49f8 0000000f  
3fff1410:  0000000b 3fff3fd0 0000000f 0000000b  
3fff1420:  3fff4ff0 0000001f 00000010 3fff4fc0  
3fff1430:  0000001f 00000011 3fff3e98 0000000f  
3fff1440:  00000000 3fff5100 0000006f 00000011  
3fff1450:  00ffffff 0100a8c0 0100a8c0 40210102  
3fff1460:  00000000 00000004 00000003 4021685a  
3fff1470:  3fff3f60 0000000f 3fff14b0 00000001  
3fff1480:  00000001 00000001 3fff3760 40215f22  
3fff1490:  00000006 3ffefef8 3fff3760 402123e6  
3fff14a0:  402123f0 3fff3760 3fff14d0 40212422  
3fff14b0:  3fff3d80 0000000f 0000000a 402169b0  
3fff14c0:  402123f0 3fff3760 3fff2bc0 40212539  
3fff14d0:  3fff2b80 0000000f 0000000a 40210507  
3fff14e0:  3fff2bfc 00000000 3fff1520 00000001  
3fff14f0:  00000000 00000006 000003e8 00000030  
3fff1500:  00000000 3fff4978 40214bd4 3fff1550  
3fff1510:  3fffdc20 000003e9 3fff2bc0 40212634  
3fff1520:  3ffe8ed8 00000000 000003e8 000289db  
3fff1530:  3fff2be0 3fff4978 00000000 3fffdc20  
3fff1540:  4020ee98 3ffefeb0 3fff0574 4020eea9  
3fff1550:  3fff3eb8 0000000f 00000000 3fff3e98  
3fff1560:  0000000f 00000008 3ffe8883 00002580  
3fff1570:  00000000 00000000 00000016 40101941  
3fff1580:  40214bf5 00000000 00000000 3fff057c  
3fff1590:  3fffdc20 00000000 3fff0574 40214c1d  
3fff15a0:  00000000 00000000 3fff0590 40100114  
<<<stack<<<

```
My esp is an esp12 (esp8266mod), and the memory is 

`python2 esptool.py -b 115200 -p /dev/ttyUSB0 flash_id                              :(
Connecting...
Manufacturer: e0
Device: 4016
`
Probably it is an BergMicro BG25Q32 (4MB)

I've try another esp12 and 201 with winbond memory up to 1MB but the result is same.

Best Regards

luc-github, 2016-02-27:
I have nodemcu 1.0, ESP12E, and ESP01 1M and cannot reproduce issue, code is quite stable for a while and no change recently

what ESP core version, Arduino IDE version, OS are you using ?
it has been reported that 2.0.0 is bugged when compiled/flashed under linux when no issue under linux

giutrec, 2016-02-27:
I have arduino ide 1.6.5 with 2.0.0 relased of core on my ArchLinux (4.2.5-1-ARCH #1 SMP PREEMPT Tue Oct 27 08:13:28 CET 2015 x86_64 GNU/Linux)

I've tnow tried also nodemcu 1.0 and the issue persist :(

luc-github, 2016-02-27:
so looks like the linux issue :
https://github.com/luc-github/ESP8266/issues/64
reported on espcore : github.com/esp8266/Arduino/issues/1459

giutrec, 2016-02-27:
Ok, thanks!
I try the esp core version 2.1.0 (when its return downloadable)

luc-github, 2016-02-27:
I did not do the necessary modifications to support 2.1.0 as not yet available
Some types changed at least : https://github.com/luc-github/ESP8266/issues/65

luc-github, 2016-02-27:
seems more than just type to change to be compatible 2.1.0 - I have crash that do not have before

treepleks, 2016-02-27:
The Linux 2.0.0 version is bugged.
I'm sorry to tell that I had to  shift to Windows :-(

I will test 2.1.0 to see if it's better and tell you.

luc-github, 2016-02-28:
just update code to support 2.1.0
https://github.com/luc-github/ESP8266/commit/9885d2cab944d876b3823130caf1f3ebc92eede3

you may want to try

giutrec, 2016-02-28:
Unfortunately the issue persist also 2.1.0 :(
On windows works correctly

luc-github, 2016-02-28:
I have reported your feedback on #64 thanks a lot
I close issue as duplicate of #64

closed 2016-02-28 by luc-github