luc-github/ESP3D #82: Add camera support

luc-github, 2016-03-10:
several project / camera already exist:
http://www.roboremo.com/jpeg-camera--esp8266.html
http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/
http://www.ladyada.net/wiki/tutorials/products/camera/index.html
http://hackaday.com/2016/01/24/truly-versatile-esp8266-wifi-webcam-platform/

luc-github, 2016-03-10:
Arducam seems the cheapest camera ( around 28US$) versus others (around 54US$)

geoanimator, 2016-03-12:
Great idea. Are you thinking of keeping it on a separate / dedicated ESP?

luc-github, 2016-03-12:
I need to do test but my first idea is no dedicated ESP, it will depend of MCU load actually, if can handle monitoring and streaming in same time or not

luc-github, 2016-03-21:
ok camera received and working with demo sketch - can capture images :smile: 
Next step is to do live streaming :thought_balloon:

luc-github, 2016-03-21:
hmm not sure it worth the time to develop something actually 
can have complete wifi cam for little cheaper : http://www.ebay.com/itm/1x-Mini-HD-Wifi-Wireless-IP-Camera-Security-Network-IR-Night-Vision-Webcam-EU-TR-/152012558200?hash=item2364a78f78:g:3lcAAOSwwpdW4541

geoanimator, 2016-03-22:
Nice find. I think you might be right. 
Other than the challenge of getting it to work it might not be practical. I do remember having problems with timecapture on my Rasp B with internal camera in combination with octoprint. I found it did affect the gcode streaming after a few hours the PI's response would be sluggish and I would miss steps. So I have since kept the two functions separate. I am sure if I research the issue upgrade etc. I might have found a way around it but didn't seem worth the complication. I just turn it off and on via octoprint menu as needed to view how a print is going. I don't use it for time capture at this moment.

luc-github, 2016-03-22:
Yes looking a code sample I am afraid video streaming keep MCU too busy to do anything else so unless there  is a cam that cam be used for ESP for less than 10$ I do not think it is necessary to work on it 
I close topic and reopen if find opportunity

luc-github, 2016-04-11:
I have ordered this one to check
http://www.ebay.fr/itm/Wireless-WIFI-P2P-Mini-Remote-Surveillance-Camera-Security-FOR-Android-IOS-PC-JK-/262238005881?
if OK definitly the one to use in printer for 14,89 EUR

geoanimator, 2016-04-11:
Yes that looks the way to go.

Hopefully you are able to use http and stream the video.
ie so you can view from a browser plus capture video or stills for time
lapse vids..

Not sure if the mobile phone versions use a dedicated protocol?

You might be able to capture the packets and see?

It will be interesting to see what you can get out of it.

great price.

*Regards Gary*
*email:** gary@gazpl.com <gary@gazpl.com>*
*mobile**: 0408923004*

luc-github, 2016-04-11:
it says it support Android / IOS / PC so I guess it use H264 or mjpeg that can be streamed in web browser like common ip camera, will test and confirm when got it ^_^

mkeyno, 2016-04-12:
hi @luc-github I made 1 meter long  POV with teensy which is show pre save image in sd card , I was thinking to ESP to replace it to teensy , do you think is it goods option  to upload the image while  POV running or just use the http upload before start the loop? do you think ESP at all good option for lighting project while it has too internally  serve the wifi function ?

[image]

luc-github, 2016-04-12:
all depend what else your esp is doing, what size is your image and what frequency you need to change this image.
ESP is able to do vnc and also to stream video with correct fps but this should occupy fully the MCU 
So if you only use ESP to upload image I guess it should work but it just a guess, I am not familiar with these features so it is just a guess - the best is to try

mkeyno, 2016-04-12:
for 25 fps I need 400 to 600 RPM and in each revolution I should read each line of image and put in 144 pixel in strip and wait for 3-5 ms , do you think this delay suitable for  other process such as wifi functions or even stream the new image ?  is it possible during this delay upload the next image ( each image  have size of 400*144*3  = 168kb)

luc-github, 2016-04-12:
This is unknown area for me, 
I think refresh image of 168Kb by wifi every 5ms is too much for ESP - but if all images are on SD like for teensy and wifi is only used to uploaded all at once - should be OK - but again just a guess

mkeyno, 2016-04-12:
no no buddy , esp need 144x3 every 5ms and 168kb per revolution

mkeyno, 2016-04-14:
hi @luc-github , can ESP stream the video frame in size of (60x60x3=10.6KB) in each 40ms (25fps) and then make UDP packet and push it to Ethernet connected to? (from browser to ESP and from ESP to Ethernet LAN)

luc-github, 2016-04-14:
sorry as mentioned earlier video streaming it is not my area - I have no idea

luc-github, 2016-06-09:
even it is nice and cheap - and have it's own battery - I did not find a way to use it easily and need specific application to display - developer did not answered to my emails so - it is not qualified for the purpose

luc-github, 2016-06-09:
I ordered : http://www.aliexpress.com/snapshot/7769617635.html?orderId=75923081605482
video I saw on the net show the use of IE so I hope to be able to display in web page - TBC
price is cheap - size / holder fit location in printer - will give feedback later

luc-github, 2016-06-09:
leave open for better following

geoanimator, 2016-06-10:
I think P2P camera's are going to give problems? Do you still have the link showing it used in IE?
A lot of manufacturers are going down this path as it makes it easy to setup for use outside the local lan ie no DDNS required etc but not good for people who like to tinker. ie they don't have a web interface for a browser generally from what I have seen? ie can't configure network IP settings etc.

More info here
http://www.hkvstar.com/technology-news/how-does-p2p-ip-camera-work.html

I must admit I am not a fan of them. Don't like the security risk.
It will be interesting to see how it goes.

luc-github, 2016-06-10:
yes I do not like p2p neither but this one use browser and app, about IE support, I cannot find the video in my historique but you have everything in description:
 
check PC requirememt : IE / Safari / google Chrome, standard browsers.....
so looks promising also have several screenshots

login use IE : 
http://g02.a.alicdn.com/kf/UT8ZH14XvlaXXagOFbXY.jpg
[image]

IE need pluggin but not sure for chrome
[image]

also seems there is  a not P2P fonction like others IP cams I use at home :
[image]

so will see ^_^

mkeyno, 2016-06-10:
@luc-github  do you thinks is it possible to install ESP-Arducam inside the PIR sensor so it send the image back to server as soon as it got motion

luc-github, 2016-06-10:
@mkeyno  if your PIR use ESP yes I do not see the problem - sensor trigger the command to take picture - but to be honest even cheap cam have this feature and no need IR sensor for that

luc-github, 2016-06-23:
Ok got the cam - after a setup using a phone and audio configuration (yes audio ^_^ as no cable to connect on this cam) I make it work :
[image]
The cam is twice bigger as a logitech C170  so it is acceptable it works under chrome without addon I think as I did not add for it, and it need a plugin for IE.
I support 2 resolutions, configuration is neat as others IP cam I use to have and even better - there is some delay (2~3 seconds) due to buffering but for printer is fine.

The only confusing part is first setup - as wps was not working I did audio - here a video I found that helped me a lot to understand.
1 - wait 1 minute the cam start 
2 - download the app on your phone
3 - connect your phone to local wifi
4 - setup the wifi / password in audio configuration page
5 - when blue led of cam is fixed - push configuration button with phone close to the face of cam, location need to be quiet as it will configure the cam using sound - It remember when radio was broadcasted game and we need to record on tape ^_^
6 - when blue led is blinking scan the code on cam using the app to add the cam - type default password and if it works  you cam is online :fireworks: 

then you can use the browser to finish the configuration. for less than 22$ it is definitly good

luc-github, 2016-06-23:
so it can be embedded in any html page of UI - just need to save the stream path with or without password

luc-github, 2016-07-16:
I forget to add :
I get only the video stream and not full UI
on IE:
http://<ip>/preview.html
[image]

on chrome and firefox :
http://<ip>/flashview.html
[image]

on safari:
sorry not tested

the request of login/password is done on the fly

geoanimator, 2016-07-17:
Glad you got it working Luc. Still not a fan of the P2P webcams but it will suit this purpose and at a great price. Now you can play with timelapse :-) Is it able to do snapshots?

luc-github, 2016-07-17:
well I do not use as p2p but direct access, the authentication is windows basic authentication, so actually you can hard code the login/ password in url and not use the p2p system
you can disable the manufacturer ddns and change port so I guess this not a true p2p camera
[image]

about snapshot it use flash on chrome (instead of usual MJPEG)  and a plugin on IE and only this plugin allow to take snap but seems issue in path
[image]

Yes next step will be to change the holder to fix cam and add in ESP3D page a window for cam

luc-github, 2016-11-26:
I close this topic as no FW related

closed 2016-11-26 by luc-github

Zefram88, 2018-10-20:
Is it possible to implement RTSP protocol on camera view?

luc-github, 2018-10-20:
From what I read here you need an external program https://stackoverflow.com/questions/2245040/how-can-i-display-an-rtsp-video-stream-in-a-web-page

luc-github, 2018-10-20:
A plugin or a player installed

wilbur4321, 2019-09-04:
Have you considered support for running esp3d on an esp32-cam board and using it's camera?

luc-github, 2019-09-04:
Yes a POC has been done using a limited UI by someone - but it use a custom webserver example code - so it will need some tuning to make both having a good cohabitation but not in same time - will need to switch between ESP3D/WEB UI or Camera UI

Zefram88, 2019-09-05:
There's a method to run code on 2nd core of the ESP32 through Arduino IDE, maybe it can be used exclusively to run camera code.

luc-github, 2019-09-05:
Core 0 is for WiFi
Core 1 is for application
This allows to get smooth behavior, Stream camera video use a lot of both, 
here the work done by @eben80 https://github.com/eben80/ESP32-CAM-Print-Monitor

ETE-Design, 2019-10-16:
Will you implement it to ESP3D?

luc-github, 2019-10-16:
https://github.com/luc-github/ESP3D/issues/239#issuecomment-536529510

ngoalongcang, 2020-04-17:
how is esp 3.0