luc-github/ESP3D #83: compile error

mkeyno, 2016-03-18:
hi @luc-github , thanks for sharing the code , I put your codes in Arduino 1.6.7  and face the following error , also why do you use ptl format instead the html,  
br

`
sketch\wifi.cpp: In member function 'bool WIFI_CONFIG::Setup()':

wifi.cpp:161: error: 'sleep_type' was not declared in this scope

     wifi_set_sleep_type ((sleep_type)bflag);

                           ^

wifi.cpp:181: error: 'phy_mode' was not declared in this scope

         wifi_set_phy_mode((phy_mode)bflag);

                            ^

wifi.cpp:217: error: 'phy_mode' was not declared in this scope

         wifi_set_phy_mode((phy_mode)bflag);

                            ^

exit status 1
'sleep_type' was not declared in this scope

`

luc-github, 2016-03-18:
there are several difference between ESP core 2.0.0 and 2.1.0
master is for 2.1.0 when 0.5.1 is for 2.0.0
It is mentioned in readme
```
Development version:
Arduino ide 1.6.5 with stable 2.1.0 from ESP8266 : Build Status
Stable version:
Arduino ide 1.6.5 with stable 2.0.0 from ESP8266, please use https://github.com/luc-github/ESP8266/releases/tag/v0.5.1
```

I use .tpl extension for dynamic content (referring to template) check keywords in https://github.com/luc-github/ESP8266/blob/master/keywords.txt
This is just a personal choice, html files and others files with static content are also supported

mkeyno, 2016-03-18:
thanks @luc-github for quick reply , I have same problem with 1.64 & 1.67 with last stable version , do you have plan to update for new version , also why don't make it as arduino library so we could extend its examples , your code as standalone example is very complicated for novice user like me 
by the way  have you check the new [Markus code on serial bridge ](https://github.com/Links2004/arduinoWebSockets/issues/61#issuecomment-198110316)
I really appreciated to have your comments 
BR

luc-github, 2016-03-18:
you are using latest ESP core = 2.1.0 you must use the master of my git - this is not what you are doing , you use older version - this has been fixed https://github.com/luc-github/ESP8266/commit/9885d2cab944d876b3823130caf1f3ebc92eede3

you must sync your repository to get latest code

mkeyno, 2016-03-18:
hi @luc-github ,I've upload the new master code when I upload the code and data file on my nodemcu , nothings happened , any sign of station or AP mode , also the serial output is like stock in boot mode

luc-github, 2016-03-18:
did you uploaded the data directory ?

mkeyno, 2016-03-18:
yes I did buddy

luc-github, 2016-03-18:
what serial output show ? default is 9600 bauds

luc-github, 2016-03-18:
you can discuss with module using serial also 
like typing : [ESP444]CONFIG

mkeyno, 2016-03-18:
I put some Serial.print in the setup function , but nothings come out

luc-github, 2016-03-18:
when connect Nodemcu nothing show ? even garbage ?

mkeyno, 2016-03-18:
little garbage such as when it goes to boot mode , How ever I compiled the code with arduino 1.64 and many compile error come out , but with 1.6.7 no error

luc-github, 2016-03-18:
Normally with nodemcu when you connect:
1: some garbage coming from nodemcu device
2: during 10s nothing display it is period to allow hard reset
3: IP
[image]

luc-github, 2016-03-18:
here clean start then type [ESP444]CONFIG on my node mcu

[image]

luc-github, 2016-03-18:
additionally - I only use 1.6.5 ide so not sure the behaviour with 1.6.7

mkeyno, 2016-03-18:
[image]

mkeyno, 2016-03-18:
nothing else

luc-github, 2016-03-18:
add return carriage to be able to key commands in serial monitor, need \n

mkeyno, 2016-03-18:
now its works , tnx buddy for your patient and time but when I connected to AP and browse to 192.168.0.1  nothings happened and downloaded
[image]

luc-github, 2016-03-18:
do not forget to close issue if ok - thanks

mkeyno, 2016-03-18:
sure , but still no web loaded on any 80 or 8888 port ,   don't you think it would be better if use the SPIFF Dir to check the files or any of feature such as Server.OnfileUpload to update the file

luc-github, 2016-03-18:
I can reproduce the issue - I was in station mode before 
seems new issue due to 2.1.0 as it was ok with 2.0.0 - will check

mkeyno, 2016-03-18:
thanks any way Luc , I'll be following you

closed 2016-03-18 by mkeyno

luc-github, 2016-03-18:
I create issue https://github.com/luc-github/ESP8266/issues/85
seems problem with captive portal -