luc-github/ESP3D #85: in AP mode 192168.0.1 is not responding with 2.1.0

luc-github, 2016-03-18:
was ok in 2.0.0

luc-github, 2016-03-18:
Seems related to captive portal - disable it and it works

luc-github, 2016-03-18:
CaptivePortal.ino sample is working so library is working - need to dig in more

mkeyno, 2016-03-18:
I disable CAPTIVE_PORTAL_FEATURE and its work buddy ,however I can't find default admin password

luc-github, 2016-03-18:
https://github.com/luc-github/ESP8266#default-configuration

luc-github, 2016-03-18:
dhcp looks not correct - ip is set to 192.168.0.1 but it works on another range:
```
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
```

mkeyno, 2016-03-18:
thanks dear Luc, now it seems is everything in order ,but do you have any idea if we intend to upload the g-code file and send it line by line just like the repeater host or weblaser of @openhardwarecoza
 I'm gonna to change 3d printer setting to somehow a laser machine but I cant find the @j0hnlittle tools as you speak of , does he has any tutorial  for that

luc-github, 2016-03-18:
I only print from SDCard this is the most reliable and not subject to communication issues
to print directly using host you need to use the data port of ESP (xxx.xxx.xxx.xxx:8888)
and the tool is there https://github.com/luc-github/ESP8266/tree/master/tools

mkeyno, 2016-03-18:
so you use websocket only for port 8888?why don't you upload the g-code file in SPIFF and then read it line by line because when we use python its means using  third party and this is not what we expected from your code

luc-github, 2016-03-18:
I do not use websocket just TCP communication
GCODE file on SPIFF ? man print files are around 3M to 64M, there is no space on ESP for this
I use Wifi SD Card so copy files is fast
Not sure what you mean about python - why you need python for GCODE ?

mkeyno, 2016-03-18:
the tools you suggest is the python script , hence I need python as third party to communicate with webserver
also as I remember my laser file not exceed than 250 kb however I wasn't successful to connect the SD card to ESP and when SPIFF come out I abandon it , but  if we could send the file to SD card through  TCP communication, why not use it

luc-github, 2016-03-18:
please read again carefully the readme:
```
Modifying and Testing tpl files

To help to visualize tpl modifications a local tool has been created by j0hnlittle to avoid to upload everytime your tpl files just to see the results of your modifications
It is a python script (2.7+) located in tools directory, launch it: python server.py, then open browser: http://localhost:8080
It will display the web ui and allow some navigation
```
the tool is just to display tpl file instead of uploading them on ESP to save time when developping html pages 

About SD Card reader it on my RADDS a shield for due - not on ESP

I do not know the constraint for laser cnc - did not build one yet - but if file is small and you have space on SPIFF - yes I guess it is not hard to open it and send it line by line

mkeyno, 2016-03-18:
sorry for my careless buddy and really appreciated for your patient , yes you are right , its work but I don't know where the  directory location  that script lunch from , so I could put my modified file on it 
Any way I've designed  new lesser  specific due shield for my CNC and if you probably have time , I've really appreciated  to look at it

luc-github, 2016-03-18:
Sure - I clean this thread as lot of things not related to issue

luc-github, 2016-03-18:
CaptivePortal.ino is pretty fast compare to the current code - there is may be some overload that generate some timeout

will recreate a minimum code to see what is the difference between sample and current code

mkeyno, 2016-03-18:
I thought you use websocket combined with ajax too enhanced speed , it is also my problem that every webserver  I built is not reliable and slow ,maybe we should wait for ESP32

luc-github, 2016-03-18:
the speed issue come with 2.1.0 - with 2.0.0 it is ok

luc-github, 2016-03-20:
Speed issue looks related to CPU speed command changing to 160
Issue for display on AP mode looks on send data part - page send data but nothing go
No issue if use small page

luc-github, 2016-03-20:
seems display page is slower and because in AP mode all requests go to 192.168.0.1 webserver cannot send all data and is overload and result is nothing is displayed
moving home page to 192.168.0.1/HOME and doing a redirection on / to /HOME should allow fast answer when / is query and display all pages

luc-github, 2016-03-20:
Trick is not really helping and display is definitely slower on 2.10 compere to 2.0.0 where pages are very responsive.
As in 2.1.0 webserver has been rewritten there is may be new settings that I missed
I have open one issue on ESP8266 github to get some explanation as I do not see anything in code orz

luc-github, 2016-03-22:
trying git version give same behavior

luc-github, 2016-03-23:
after investigation slow performances are due to a memory check to control heap corruption, this check will be disabled by default and activated only in debug mode
AP not responding is due to current initialization sequence not supported in SDK 1.5, currently AP is setup then static IP is defined, seems static IP must be defined before setup AP

In addition now having in AP mode MDNS / SSDP and Captive portal is give some random failure when connecting, it may make sense to use captive portal only in AP mode, first to avoid to overload, second as no need discovery is all request go to AP

mkeyno, 2016-03-23:
hi @luc-github,thanks for your effort to catch and fix bugs,if I was how could I find the memory allocation malfunction 
any way  where your file modification will be , in your branch or let the Iggor to do in new release

luc-github, 2016-03-23:
it is part of the core I cannot do anything in my code - that is why I gave instruction in previous post
until 2.2.0 is released as mentioned in github.com/esp8266/Arduino/issues/1800 or at least in github

luc-github, 2016-03-25:
the fix for memory performance is already in git : https://github.com/esp8266/Arduino/commit/3d3c928e2eec353637bd19574745dbfafde42221

luc-github, 2016-04-08:
everything looks fine with 2016-04-08 git version and current master that include several fixes
I consider the esp core 2.1.0 not suitable for this fw so I keep stable for 2.0.0
2.2.0 should be the good one for next stable - do not use 2.1.0
 
I close issue

closed 2016-04-08 by luc-github