luc-github/ESP3D #86: best Authentication method

mkeyno, 2016-04-01:
hi @luc-github ,I really appreciated if I have your comment for this Authentication method
  I want to define couple of pages with `server.on(page, method, handler);` but I want each client authenticate before access to each page , I know both HttpBasicAuth & SimpleAuthentification example but I don't know which way is the best 
your help really appreciated

luc-github, 2016-04-02:
I do not use HttpBasicAuth  because it give login/password in base64 in each header
I use another method with a token like in SimpleAuthentification but with small data base of IP and timeout on ESP so password is sent once -it is  just a personnal choice,  none solution is full proof anyway

mkeyno, 2016-04-02:
thanks dear @luc-github for quick reply , regard to your suggestion can I use such code , `server.on(every_pages, method, check_Authentification_handler);
if(check_Authentification_handler) redirect_to(requested_pages)
else                                                redirect_to(login_pages)
`
also as you mentioned, does HttpBasicAuth method can automatically check session validation for each requested page so in this case not necessary to use  more code ?
BR

luc-github, 2016-04-02:
sorry never use this function neither read sample code as recent in 2.1, and did not have usage for it as I already implemented another authentication method.

mkeyno, 2016-04-02:
can you share me your authentication  code

luc-github, 2016-04-02:
everything is on this github project, most code is in webinterface.cpp, feel free to have a look

luc-github, 2016-04-07:
If no update I close it

closed 2016-04-07 by luc-github

mkeyno, 2016-04-07:
thanks any way Luc