luc-github/ESP3D #90: After upload - instructions

openhardwarecoza, 2016-05-10:
Hi

I am using an ESP07  - just bench testing 

: Set to 1MB flash, 512K SPIFS
: Uploaded the .ino
: Uploaded the SPIFS

Restarted with GPIO0 in RUN mode

Nothing popping up in serial monitor, not any AP to connect to?

luc-github, 2016-05-10:
Are you sure your ESP 07 has 1M flash ?
What is your IDE version / ESP Core version / Operating System ?
default baud rate is 9600 and need 10 s before it display ```M117 192.168.0.1```
Set your terminal to CR+LF and type 10 sec after starting
```
[ESP444]CONFIG
```
If flash is ok it should display EEPROM config

openhardwarecoza, 2016-05-10:
Spot on!
Sorry!   It only has 512K - not 1M as advertised (thanks china lol)

closed 2016-05-10 by openhardwarecoza

luc-github, 2016-05-10:
Well for 512K current FW won't fit - you need to use the limited one which do not support latest core
I dropped the support as ESP01 has now 1M flash

openhardwarecoza, 2016-05-10:
I set it to 1M/128 SPIFS:

Uploaded without issue and up and running 

[image]

Havent done any more testing.  But noted will order some 1M ESPs then (:

luc-github, 2016-05-10:
good then - Web FW update should work with 1M/128K but not with 1M/256k
I suggest you to check 12E or 12F models they usually have 4M flash and almost same price as ESP01 - and they are more confortable for working with

luc-github, 2016-05-10:
Do not forget to flash with mcu at 160Mhz to speed up ui reaction

openhardwarecoza, 2016-05-10:
Thanks.  I'll contibute a few Wiki pages to the project (:

I still have 60+ ESP07s here so glad it sort of works. 

Last question: Whats the admin user's default password?

openhardwarecoza, 2016-05-10:
Yes i am on 160mhz

luc-github, 2016-05-10:
admin / admin -  yes saw it so removed my comment :smile:

openhardwarecoza, 2016-05-10:
(: Sorry the email notification came through still (:

Pretty responsive I must say.  Now to get back home and hook it to a printer

luc-github, 2016-05-10:
Thanks a lot for your feedback - appreciate - I am not good in wiki :blush:

openhardwarecoza, 2016-05-10:
First two wiki entries up for your review:

https://github.com/luc-github/ESP8266/wiki/Flash-Size

https://github.com/luc-github/ESP8266/wiki/Install-Instructions

luc-github, 2016-05-10:
Wow nice - thanks a lot -I noticed list was not displaying properly so I corrected it,  markdown is sometimes painful with line feed - need to add 3 spaces at the end of the line to get proper return carriage

Thanks a lot