luc-github/ESP3D #91: Proprosal: Ren

openhardwarecoza, 2016-05-10:


luc-github, 2016-05-10:
well I will work on smoothiware support soon and it is not really reprap - I agree ESP8266 is too light but github description is  there - and because now many links point to this name , to change the name means to create a new github with new name and create a redirection on current one to new one.
I do not sure it worth 
about arduino project file and directory itself  inside GitHub- I think it is ok, I was also thinking about new ino file like : ESP_for_3Dprinter but looking for shorter one

TRoager, 2016-06-01:
ESP3D?

luc-github, 2016-06-01:
Yes could be a good candidate :smile:

openhardwarecoza, 2016-06-01:
I like esp3d!

luc-github, 2016-06-07:
seems forward feature is available : https://github.com/blog/1508-repository-redirects-are-here
so I will rename repository as well after some tests

luc-github, 2016-06-07:
I have validated that renaming a repository keep forwarding previous github address to new one - so far so good

If no one comment against this I will change this week from ESP8266 to ESP3D as I did not found another similar project with same name

TRoager, 2016-06-07:
Yay i had an impact on the name :D

luc-github, 2016-06-08:
what about a logo if name is defined ?
[image]

TRoager, 2016-06-08:
I agree there should be a logo but not sure if thats the logo for the project but im no designer so can't help there :p

luc-github, 2016-06-08:
me neither but I am sure you saw it :smile:

paint.net my limit

luc-github, 2016-06-08:
Ok, code is modified / links are updated  / readme is updated - I will change name tomorrow and push changes to match new name just after.

luc-github, 2016-06-09:
Name modification is done - thank you guys

closed 2016-06-09 by luc-github