luc-github/ESP3D #92: smoothieware support

luc-github, 2016-05-11:
got smoothieware compatible board - will check compatibility

luc-github, 2016-05-13:
Got the card AZSMZ that support ESP01 connection natively :smile: so no delay to connect 
* Temperatures catch/set are ok if T / T1 / B are used for dual extruders +bed, but need to add T0 / T1 / B customization support like in config file for safety
* position catch is OK 
* Speed/Flow need to be adjusted
* info/warning/error messages catch need to be adjusted

currently I plan to add a TARGET_FW define (REPETIER/MARLIN/SMOOTHY) to sort these changes at compilation time,once done I will see if runtime change does not impact performances to give more flexibility

raykholo, 2016-05-13:
Hi luc,
I designed a Smoothie variant that has an ESP8266 onboard (it is based off of the Smoothiebrainz board that Peter designed @openhardwarecoza) and want to be able to control it directly from Chilipeppr (or Laserweb) without need for the Serial Port JSON Server or any other utility to bind the tcp to a virtual serial port - just direct browser connection.

What I like about your code is that the web server gui is available as a fall back if someone does not like the web-based model of CP/ LW or does not have internet access.  

These are the things that I would like to address, it seems they have been brought up before:

1. Websocket support.  We can't have a raw tcp socket in the browser so it seems like websockets are a must in order to be able to send gcode from javascript.  
2. Ability to upload a full gcode file job externally (api, not from within the web server page)  HTTP POST could work for this, but we run into A LOT of CORS issues so making it part of the protocol over websocket would be better.  This is for preloading a job so that a computer does not have to be on constantly feeding lines to the machine.  Also makes it easier to coordinate multiple jobs on different machines. 
3.  Lastly, the obvious question of Smoothie support - I am converting multiple of my machines to my new boards and the Smoothie community is super helpful, so just ask if you need anything.  

Let me know if any of this has been done already, or what it would take to make this happen. 
Thanks,
Ray

luc-github, 2016-05-13:
Hi Roy,
Thanks for the feedback, yes websockets are in TODO list but as currently I do not have a clearly defined application/ user case from A to Z, at least in 3D printing to valided the implementation, so I did not hurry until now.

about your points:
1 - well I am not sure what you mean, sorry, my current HTML UI use javascript and send GCODE commands - actually that what mainly do the UI - just send GCODE commands - so I do not get why you say websocket is a must for that.

2 - do you mean to upload file to smoothie SDCARD but not using M28 / M29 commands that take sooo much time ? At least for 3D print file which can be about several Mb, I do not know for laser files - I am still looking as smoothie FW sources - but not sure it is possible without Smoothie FW modification - still need to dig in

3 - Yes will do - this is my first "real" contact with smoothie world and it is interesting,  I plan to do a CNC or Laser encraver but need to build the frame and find the  components

openhardwarecoza, 2016-05-13:
We are using our own web interface,  therefore the 'want'   for a websocket
implementation.

With the esp12e we can talk directly to the SD card over SPI (and you can
wire a multi master SPi bus (;

luc-github, 2016-05-13:
Hi peter yes I remember the discussion we had that is why it is in my todo, but as I mentioned the A to Z is not clear to me yet as never play with laserweb, did not build a laser machine to understand the constraints, etc...

About SPI and talking to SD I know, but this a more complex situation than connect Esp01 to AZSMZ lcd or Azteeg 5X mini
using Esp12E connected to SDCard is possible and very easy when you do the hw, I was talking using ESP as serial bridge which is different.
Everything is possible when you own the full specs :smile:

mkeyno, 2016-05-13:
hi Luce,I used you ESP code and I thought you used websocket to read and send commands to cnc controller, I'm supersized you are not implemented yet  , however I also test Markus transparent links and it so fast and clean ,  why don't you use his prototype

luc-github, 2016-05-13:
@mkeyno pls read again my post :smile: 
yes Markus's library is definitly the choice I plan to use, but you are talking about websocket as transparent bridge to serial, Roy is talking about websocket only, Peter is talking about websocket /serial too (may be?) but also about websocket /ESP -SPI. 
So as said specs are not clear for me - especially as I currently do not know nothing about laserweb.
I am not rejecting - just explaining specs are important to code something and reach the (correct) target. 

Currently I am porting my fw working with repetier/marlin FW to work with smoothie FW but I am more than open to implement additionnal features if they are well defined. :smiley: there are always many ways  to reach the target

luc-github, 2016-05-14:
@raykholo I think this is what you are looking for https://github.com/nailbuster/esp8266FTPServer
I followed some discussions on gitter when he was developing and it seems working.
I work with SPIFFS so should be OK with SD also but may need some adjustment.
It is in my todo list to add it but was not urgent as I already have a webuploader for the SPIFFS but that was in my plan to upload big gcode files to SDCARD but my first approach was not like @openhardwarecoza suggested hard wire the sdcard to ESP but using the serial pipe so I delay this part.
In addition to explain my delay I currently use wifi SDCard (flashair) on all my 3D printers so uploading files was not a problem - but it is in my list to avoid to handle 2 devices

raykholo, 2016-05-14:
@luc-github Thank you for the responses.  I have not had time yet to run your code on an ESP so I apologize if I have said something incorrect. 

To clarify a few points:  I was talking about websocket in general: I wanted to be able to stream gcode from the control app, like the others are suggesting, but I was also hoping to add other commands for the ESP, not the smoothie - such as "save this entire job", "run job".  But for those I wanted to save the file directly on the ESP, not use M28/29.  I am not sure if we would be able to fit a several mb file (or larger) on the ESP due to memory constraints.  

@openhardwarecoza would it be possible for the ESP and Smoothie to share a single SD card?  I am not as optimistic about having to put 2 microsd sockets on my board.  

I will have to get better acquainted with your code Luc.  There are different levels of success here.  I would be very happy with just implementing the websocket serial bridge (this seems to be what the others are suggesting) into here so that we can use that with CP/ LW but then your GUI is available as a fallback option if someone does not have internet access.
But a future step might be something like @chilipeppr Serial Port JSON Server that can take other commands besides just passing gcode to the controller, which is how I saw the "save this job, run this job" commands working. 

@cinezaster please see this thread.

luc-github, 2016-05-14:
@raykholo yes everything is possible - I already use some specific commands for ESP [ESP<command>]<parameters>

About serial/Websocket most of the code is already available, Markus provided it "as it is" so may need some adjustment : https://github.com/Links2004/arduinoWebSockets/issues/61#issuecomment-198231422 

And about futur that is the beauty of coding, the limit is your imagination - so everything is possible :smile:

luc-github, 2016-05-14:
Anyone has a basic html page or App to test the websocket /serial bridge ? If not I will build the HTML/ javascript page for reference

raykholo, 2016-05-14:
https://github.com/chilipeppr/widget-spjs

This is how CP communicates with Serial Port JSON Server.  I think the protocols are quite specific to that but you may be able to borrow some code or use it as a starting point.

luc-github, 2016-05-14:
thanks - seems Markus provided it : https://github.com/Links2004/arduinoWebSockets/blob/master/tests/webSocket.html

so should be Ok to check

luc-github, 2016-06-20:
Smoothieware is now ok, all topics discussed here have it's own ticket (https://github.com/luc-github/ESP3D/issues/94,https://github.com/luc-github/ESP3D/issues/99,https://github.com/luc-github/ESP3D/issues/93) so I close issue

closed 2016-06-20 by luc-github