luc-github/ESP3D #93: Direct SD Card support

luc-github, 2016-05-16:
As suggested by @openhardwarecoza : add direct wires to SD Card reader for file upload

SD Library examples suggest SD card attached to SPI bus as follows:
 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4

luc-github, 2016-05-16:
currently I use wifi SDCard with my ESP,  would be great to not handle 2 devices.
Issue would be to avoid concurent access to card, so need a flag in board fw to let esp know it can use sd card and same for esp to release SD card to board

luc-github, 2016-05-16:
On AZSMZ board with Smoothieware connected to PC the SDCard is automaticaly visible under windows as a Removable Disk - using Repetier and so M28/M29 to upload file does not work - 
```
11:31:27.257 : No firmware response detected for sd action. Assuming success and continue.
```

luc-github, 2016-05-16:
After restarting board the file is present on SD Card - so concurrent access seems working

luc-github, 2016-05-16:
Seems depending of how fast is repetier to load files before windows mount the removable disk.

From ESP the M20 command never work when SDCard is mounted on window when in repetier is seems working but actually it do not - I guess it cache the information instead of really getting them and so giving wrong files list

openhardwarecoza, 2016-05-16:
Going to tag in @arthurwolf -  arthur,  this is some very beta craziness,
but we are attempting to do the crazyness of accessing the sd on a multi
master spi bus (; off the esp8266.   Or something like that.  I honestly
missed a lot of the conversation between Ray and Luc too (;

arthurwolf, 2016-05-16:
I didn't know about this project, it's very very nice.

Liam Jackson is actually working on something that looks like this, essentially teaching an ESP to do the following : 
1. Allow users to connect to the board via serial at a fixed baudrate ( with a telnet port )
2. Get it's Wifi config from the board via serial ( "cat" command http://smoothieware.org/console-commands )
3. Essentially "act" exactly like a Smoothieboard, so anything coded for Smothieboard's Ethernet also works with it, which includes :
a) When doing an http get for "/", answer with a static webpage
b) When doing an http get for "/sd/something", use the "cat" command to get the file and answer it's content
c) When doing an http post for "/command" with the command as a post arg, call that command over serial, and answer it's answer over http

Because the Smoothieboard is supposed to come soon with a web interface on it's SD card, implementing a pass-through like that to that interface, would probably be a good idea.

openhardwarecoza, 2016-05-16:
Now thats what I was looking for!

luc-github, 2016-05-16:
well current version support the console-commands from TCP/IP terminal like putty and from web interface  - but on web interface I did not implemented the full output yet - currently only info/error/warning are displayed but I planned to add full output on an additional list

[image]

[image]

arthurwolf, 2016-05-16:
Wow, really really nice.

luc-github, 2016-05-16:
the web UI has few hook to retrieve specific information in json format (sd files/temperatures/positions/etc..) because it is easier, but it is mostly a pass-through so any serial command will work on web ui - and on any TCP/IP port defined in setting page

but where web ui has authentication - TCP/IP port currently do not have authentication like telnet

luc-github, 2016-05-16:
but back to original topic - all SD access go through serial pipe instead of direct SD reader connection with ESP. which is not really fast for file upload

luc-github, 2016-08-05:
Web upload to SD Card using ESP give me  95000byte/s which is a lot faster than serial transfer
but seems I got instability if SDCard reader is powered by ESP and if ESP is using also SPIFFS - need to investigate - if some of the issue are due to too much pwer drained by card and/or if there are some conflict in core as SD and SPIFFS share same API naming, SPIFFS has a namespace but not SD (I use #define FS_NO_GLOBALS) I see some update in git to limit the objects - I will test it

openhardwarecoza, 2016-08-05:
I have been playing with this too,  and can confirm that its not power
related.  I get lockups and hangs with sd when using the arduino sd card
library along with webserver. (even without spiffs)  -  but man,  that
upload speeds are sooo nice,  its worth spending time to get this working

luc-github, 2016-08-05:
Actually I got 0 issue when doing my test with SDWebserver I was able to upload 9Mb file, several times when I was not able to go over 1.5Mb with my FW - may be was luck

openhardwarecoza, 2016-08-05:
Try downloading the file you just uploaded

luc-github, 2016-08-05:
ok thank you will test that, I did only one way actually

openhardwarecoza, 2016-08-05:
I only played for an hour or so earlier in the week.  Didnt get a
resounding confident feel.  But hey,  i am watching your work very
closely.   Sd proxy to smoothie was smoothesp's goal (but no progress in a
few months)  ,  and i bet we can raise you a few donations to get this
feature working!  Shout where i can help!

luc-github, 2016-08-05:
I did several tests not during 1h but several minutes :
Upload 7Mb file - 3 times - then download it 3 times, then compare downloaded file with original one using meld which said it is identical. and watch the serial output when during test so far so good

I got few : LmacRxBlk:1, which means too many data are coming on wifi, but this is because I am in debug mode and seems no data corruption as final result is ok

Can you share what was your circuit and what size of file did you used ? I would like to reproduce the issue you meet 

Here my system using 2 usb adapters one for ESP and one for SD Card reader:
[image]

 with https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/SDWebServer

openhardwarecoza, 2016-08-05:
> ESP8266WebServer/examples/SDWebServer

Ahh yes,  lmacrxblk was what i was getting too.  Several,  eventually
responses botg serial and wifi suspends till reset.

My test setup is a esp12-e,  ad over spi.  I tried running a http <->
serial loop while uploading (genuine use case: i want machine feedback
while uploading.)

luc-github, 2016-08-05:
I see - webserver is not what you must use if you want to get feedback while uploading as it is a SyncServer 
I suggest you the Asyncwebser from me-no-dev - it is in my long todo list to integrate it
https://github.com/luc-github/ESP3D/issues/110

openhardwarecoza, 2016-08-05:
Cool!

luc-github, 2016-08-06:
Ok now ESP3D is working well (so far) using SD Card - still need to work on several feature like automount and File manager for SD card and blocking other requests when doing file upload to avoid to overload wifi but on good way finally

I also found an issue and reported it : github.com/esp8266/Arduino/issues/2368 but I can workaround easily for the moment

Aleks-MO, 2016-10-15:
How will be SD united with  ESP12F (https://www.aliexpress.com/item-img/2015-New-version-1PCS-ESP-12F-ESP-12E-upgrade-ESP8266-remote-serial-Port-WIFI-wireless -module / 32643052680.html #

luc-github, 2016-10-15:
what do you mean ? wires follow normal SD reader recommendation for ESP as mentioned in first thread:
[image]
after it depend of your sd reader

Aleks-MO, 2016-10-15:
[https://encrypted-tbn2.gstatic.com/images?q=tbn:ANd9GcQ_7XWPAaz2KyTufywp8cESzfFGl6kvdRAW2Suf3lQJKOnoYyTwlw](url)

luc-github, 2016-10-15:
what is that?  it show 404 page not found

Aleks-MO, 2016-10-15:
http://www.mikrocontroller.net/attachment/286085/ESP8266-12F_Tronixlabs_Australia.pdf

luc-github, 2016-10-15:
you mixed pin number and GPIO number check page 7
6 IO12 GPIO12; HSPI_MISO
7 IO13 GPIO13; HSPI_MOSI; UART0_CTS

Aleks-MO, 2016-10-15:
thanks, understood the SPI bus.
** CLK - 5 IO14 GPIO14; HSPI_CLK
** CS - 16 IO15 GPIO15; MTDO; HSPICS; UART0_RTS
Here too that at you. 
http://esp8266-arduinoide.ru/sd/

luc-github, 2016-11-24:
Support is ready and will be available later so I close issue

closed 2016-11-24 by luc-github

griffin-117, 2018-07-25:
Hi guys. I have designed a small attachable device that has an ESP-12E with a micro sd card reader directly on the back of it. I have attached the pins as noted above, but it does not seem to see the sd attached to the back. It does see the one on the mainboard through its serial connection, but as file upload, that way is incredibly slow, I want to upload through the direct connection over SPI. Is there something that I need to change from the current version of the ESP3D compilation? I just flashed the most current source code.

The board I am using is Smoothieware based.

[image]

Really appreciate the help guys!

DouglasPearless, 2018-07-26:
Well, I have been down this path (months working on this problem) and the answer is probably not one that you will like, summary:

(1) The serial port on the Smoothie was only intended as a console and does not implement resource locking so you get corruptions at high baud rates when multiple things are happening as Smoothie is a co-operative multi-tasking kernel (from MBED) and not a true multi-tasking kernel (the Cortex chips lack the hardware to implement this).  While i have implemented serial port resource locking, it is tied up in a bunch of updates that I have not pushed to the Smoothie repo.

(2) The libraries for the ESP-12E do not reliably allow a high through put - I have worked with the Espressif author of said libraries and collectively we were unable to get anywhere near 921600 baud with any degree of reliability using a telnet<->serial bridge mode.

So, this means that until someone figures out how to get the ESP8266 to run really fast, we are stuck; I do note that the ESP8266 has been used to stream video, which is not the same as sending data, getting an acknowledge and sending more data; we have even tried sending entire buffers full of data to minimise the handshaking going one, but it was still too slow :-(

Cheers
Douglas

griffin-117, 2018-07-26:
Hmm, this was exactly my fear, Douglas. I appreciate the detailed write-up nonetheless! Is direct SD access for Smoothie still out of the picture for general users in the short term? ESP3D has quickly become my goto for a quick and easy web UI for my boards, but can't stand the slow upload over serial.

DouglasPearless, 2018-07-26:
I occasionally use direct SD access (the MSD capability from MBED), but it can cause issues when the host computer does things that interfere with Smoothie’s use at the same time (no resource locking on the SPI bus either).

I have looked at adding locking to SPI but so far no client wants to pay for it to be done as it is quite a bit of work and I do not have time to do it in my (haha) spare time.

The slow serial is definitely a problem with the ESP, rather than the serial as I have run the serial at 921600, which is fast enough for most 3D printing :-)

griffin-117, 2018-07-26:
I know that Roy from Panucatt has a modified version of ESP3D that has direct SD access for his X5 Mini Wifi. It really interests me to have a cheap onboard wifi module especially with the ease of use that ESP3D has.

As far as I can tell there is an intermediate buffer that interrupts the SPI bus from the LPC in some way. Makes sense but dunno how exactly it works. What I do know is that the upload speed is quite excellent (Nice work Luc).