luc-github/ESP3D #99: FW compatibility table

luc-github, 2016-06-20:
build compatibility table for FW:
Repetier, Repetier Davinci, Marlin, Smoothieware, Reprap

may be list necessary GCODE and supported feature should be ok

luc-github, 2016-08-01:
Table will look like this for each information needed by UI

**References**

Board |  GCODE 
------------ | -------------  
Repetier | https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L39-L144 
Marlin | http://www.marlinfw.org/articles/gcode/overview.html 
Smoothieware | http://smoothieware.org/supported-g-codes 
Reprap | http://reprap.org/wiki/G-code 

<HR>
**Temperature query**

Board |  GCODE | answer | Note
------------ | ------------- | ------------- | ------------- 
Repetier |  M105 |    | 
Marlin | M105 | |
Smoothieware | M105 | 

<HR>
**Position query**

Board |  GCODE | answer | Note
------------ | ------------- | ------------- | ------------- 
Repetier |  M114|    | 
Marlin | M114| |
Smoothieware | M114| 

<HR>

luc-github, 2016-08-01:
Wiki page is under creation for this: https://github.com/luc-github/ESP3D/wiki/Firmware--support

denmorru, 2017-03-28:
It would be nice to show firmware name and version in UI,
possibly add compatibility information for user.

P.S. links for both Marlins are broken in https://github.com/luc-github/ESP3D/wiki/Firmware--support

luc-github, 2017-03-29:
Yes it is done in my devt version but not yet available, I will sync in coming weeks in theory.

thanks for pointhing this out, I have updated wiki

luc-github, 2018-06-03:
I close this topic as wiki is mostly up to date for this

closed 2018-06-03 by luc-github