luc-github/ESP3D #103: Voltage devider in wiring section

WyvernDex, 2016-06-27:
Hi,
the voltage devider (1k and 2k in the rx line) caused connection problems in my setup.
Why do you inserted them?
In case of the Davinci, the UART works with 3.3V, so no voltage adaption is needed.

luc-github, 2016-06-27:
Hi electronic is not my area - so it may be not optimized I agree - I used the wiring long time ago from here : http://www.martyncurrey.com/arduino-to-esp8266-serial-commincation/

It works on my davinci 2.0 and my RADDS/DUE as well as my FTDI (3.3), before i use it I had random exception and no more after - so I concluded it was Ok.

WyvernDex, 2016-06-27:
At the page you posted are the both resistors correct, because the arduino is a 5V device and could destroy the 3.3V esp module.

In this case leaving them out would improve stbility.

luc-github, 2016-06-27:
but on picture is connected to 3.3v, as said I never got issue so never think about this since

luc-github, 2016-06-27:
May I know what kind of issue you meet ? your board is which model ?

luc-github, 2016-07-03:
If no update I close issue

closed 2016-07-03 by luc-github