luc-github/ESP3D #109: Add Netbios support feature

luc-github, 2016-08-05:
as now it is available in esp core git : https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266NetBIOS

luc-github, 2016-09-01:
But need git version of ESP Core

Pablo2048, 2016-09-15:
So until the next version of Core is released You can use my original library from http://www.xpablo.cz/?p=751 ...

luc-github, 2016-09-17:
Thank you  - do you plan to do an english version of your pages ? would be easier to spread it

Pablo2048, 2016-09-17:
Unfortunately i'm not good in english so probably not :-( . But at least i can translate my comments inside the library into english ones...

luc-github, 2016-09-17:
May be not all articles but - you will gain a lot at least for your library and other any code you do - just my 2 cents

great job anyway - thanks for sharing you work

Pablo2048, 2016-09-17:
Thank You. Maybe after a lot of learning I can try... :-)

luc-github, 2017-05-19:
This present in core 2.4 and git version so I can close issue

closed 2017-05-19 by luc-github