luc-github/ESP3D #110: Implement AsyncWebServer instead of SyncWebServer

luc-github, 2016-08-05:
Using the great library of me-no-dev https://github.com/me-no-dev/ESPAsyncWebServer -  this would allow to have better experience instead of using one html file to split css and javascript files and decrease template file interpretation workload parsing only part of html

mkeyno, 2016-08-06:
good luck @luc-github   this would be best feature of ESP3D

luc-github, 2016-09-28:
created a branch for this purpose : https://github.com/luc-github/ESP3D/tree/asyncwebserver
will take opportunity to rewrite webinterface.cpp to use `addHandler` method instead of `on` method to split file in several classes/files for clarity

luc-github, 2016-09-29:
Easier way is to do https://github.com/luc-github/ESP3D/issues/114 first to simplify coding

luc-github, 2016-10-02:
Will use embedded authentication method - this should simplify the code also   
As I can see  page handles need to be rewrited because use `AsyncWebServerRequest` class

luc-github, 2016-10-17:
Ok after testing (a lot) and rewriting/porting part of ESP3D code for async I will put on hold for a while.
Looks like library is not yet fully stable and because maintainer is busy with ESP32, so no idea when fixes would come, and async concept is not clear from code point of view for me, so I won't dig in

What I have found/observed so far :
 1 - Multiple files upload is not working (issue open)
mitigation would be to allow only 1 file upload
2 - Embedded authentication is not working under window 7 / IE 11 - but seems ok on chrome/firefox/windows safari, I did not tested android (issue open)
mitigation would be to use my authentication method which do not rely on browser but IP
3 - Webserver is definitly faster but  I have also randomly no display page / no answer, I cannot define reproductible steps to report issue properly, and looking at others open issues it is not a priority.
4 - SSDP seems not working - but I did not dig in yet, Netbios looks fine.

So Async library is no mature yet for ESP3D but but very promising -  it also gave me opportunity to redo some code with another approach that I will use in sync server, waiting for update on async

luc-github, 2016-10-18:
Update 
1 - Multiple files upload is not working (issue open)
2 - Embedded authentication is not working under window 7 / IE 11  -  issue fixed by @me-no-dev - thank you
3 - Webserver is definitly faster but I have also randomly no display page / no answer, - this may be due to 5 connections limits and  the number of files linked in html pages (2 js and one css) - seems better to combine them in one gzip files to limite the issue 
4 - SSDP seems not working - This one is on me (:shame:) I forgot to enable the /description.xml with new request method

me-no-dev, 2016-10-18:
if you have static files you might consider gzipping and putting them in progmem. That will speed up the interface immensely :) I convert gzipped file to progmem using https://gist.github.com/me-no-dev/f137a950ce6dedb641d427d8db6355d2

me-no-dev, 2016-10-18:
some browsers seem to make more than one request at a time sometimes which grows connections to more than actually needed. Could also be the case.

me-no-dev, 2016-10-18:
[this method](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L354-L511) deals with uploads and detecting when one ends and another starts. In some cases with new lines at the end of the file you might confuse the method at some point and not detect the end of the file.
If you find a bug there please report as you seem to have a good test case

luc-github, 2016-10-18:
@me-no-dev thanks - actually the goal is to let user to customize the html pages without ding in FW, I have anther request to be able to serve smoothie files (https://github.com/luc-github/ESP3D/issues/112) so in anyway I may not control the files update / content but I can set some limitations

Thanks a lot for pointing me this function, I will dig in and report to you - I should confess I have started to looks at async code but I am <s>little bit</s> fully lost even read me is very good.

Thanks a lot for taking time to come here, I know you are very busy with ESP32 ^_^ ( i cannot wait to play with it)

me-no-dev, 2016-10-18:
btw I have Flashforge Dreamer 3D printer that I would like to test with your project :) Some commands are different but I have them somewhere here. It has WiFi but it sucks... will be nice to get a working one

luc-github, 2016-10-18:
Yes Flashforge dreamer has wifi, and it sucks ? I though was good printer, I like enclosure and removable bed,  I remember it is based on makerbot but should be easy to adapt (S3d can handle it so should be easy)- I have a Beaver II (also a makerbot clone) but I put a wifi SD on this one 😉

me-no-dev, 2016-10-18:
the printer itself is amazing, but some combination of the softwares is not and the printer drops the connection at some point and you can no longer connect to it if WiFi. I have spent many weeks communicating with flashforge. They even sent me a different WiFi... still the same. But if I am in control, I can do it ;) I have written libusb things for it and could control it but never got to do wifi with interface

luc-github, 2016-11-24:
Working on multifile upload is still in my TODO list but was busy on other topic will go back on it ASAP

luc-github, 2016-11-30:
Ok issue of multiple upload is now fixed - the second issue I meet is: randomly the test page display...nothing
I suspect the 5 connections limit as the page has one html, 2 js (bootstrap and Jquery) one css (bootstrap) and do one call to list all files on SPIFFS and there some hidden call like favicon, etc...

As recently I have discovered GULP, thanks to smoothie imrahil project, that I have ported to ESP3D (https://github.com/luc-github/smoothieware-webui-for-ESP3D) to concat files , minify file and gzip file at once, so I think keep the monolitic pages as for static web server but using original js to avoid waste time in optimizationis the way to go and was suggested several times by the dev guru @me-no-dev

the next step is to find an easy mechanism to select which file need authentication and which do not : may be add a config file which list all protected files and which level of authentication: guest, user, admin

luc-github, 2017-05-19:
New FW and New UI are very fast with sync web server so implementation will be postponed until the need grow

luc-github, 2017-11-05:
Status as of today about Async port : 
on ESP8266 : 
- UI is loaded :smile: 
- xxx.xxx.xxx.xxx:forcefallback=yes to force embeded page is working :smile: 
- SPIFFS file management is working :smile: 
- web update is working :smile: 
- SDDP is working :smile: 
- serial upload on SD is working on repetier :smile: but not all boards and got issue on marlin :cry: I suspect timing issue as delay usage is prohibited I may fine tune timing after M28 command before upload
- web command is working partially ( ESP queries seems ok) on Serial small queries look fine - but M205 on repetier which dump EEPROM (4K) randomly fire Soft WDT did not found why :cry:  :exclamation:   :exclamation:  :exclamation:  this is critcal part.    
Adding WDT functions did not helped:
```
Soft WDT reset

ctx: sys 
sp: 3ffffb80 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffffd30:  3a525045 34322033 2e302036 20303030  
3ffffd40:  72747845 6120312e 00000000 4021a7d9  
3ffffd50:  3d305b20 5d66666f 3fff59ac 4020cf89  
3ffffd60:  00000000 00000000 00000000 00000000  
3ffffd70:  00000000 00000000 3fff50a4 0000003f  
3ffffd80:  00000028 3fff4fe4 0000007f 00000000  
3ffffd90:  3fff463c 0000000f 00000004 4010020c  
3ffffda0:  3fff4eb4 00020fc1 3ffffd60 0000002a  
3ffffdb0:  3ffed2fa 3ffffdd0 3fff4e8c 00000002  
3ffffdc0:  3ffed2fa 3fff59ac 3fff59ac 4021d304  
3ffffdd0:  00000000 00000000 00000000 40214c04  
3ffffde0:  3fff59bc 3fff59ac 3fff3dec 40214f60  
3ffffdf0:  00000001 3fff59ac 3fff59d0 00000002  
3ffffe00:  3ffed2fa 00000002 3fff59ac 402129e4  
3ffffe10:  3ffed2fa 00000002 3fff59d0 4021b818  
3ffffe20:  3ffed2fa 3ffee3b0 3fff5c30 00000002  
3ffffe30:  3ffed2fa 00000002 3fff59ac 40212c42  
3ffffe40:  3ffeb1dc 00000000 00000000 4010068c  
3ffffe50:  00007fff 0801adbb 401021ee 00000100  
3ffffe60:  3fff59d0 00000000 3ffeb1dc 00000001  
3ffffe70:  00000001 00004a88 0000046b 3fff5c08  
3ffffe80:  00000001 3fff4624 3fff5bb4 40212c8d  
3ffffe90:  3fffc278 40101f88 3fff5bb4 4021779d  
3ffffea0:  3fff53bc 3fff365c 3fff53bc 4023cdbd  
3ffffeb0:  3fff2160 3fff216c 00000006 3fff20bc  
3ffffec0:  3fff53bc 3fff20c4 3fff20c0 402177e8  
3ffffed0:  3fff53bc 3fff20c4 3fff20c0 40242588  
3ffffee0:  0000015a 2601a8c0 00000018 0000015a  
3ffffef0:  00000000 00000018 3fff0a7c 40106858  
3fffff00:  00000000 3ffebef4 3ffebf10 3fff2164  
3fffff10:  3ffed17a 3fff216c 3fff4624 402409a5  
3fffff20:  3fff1fe8 3fff288c 3fff288c 3ffef538  
3fffff30:  00000000 3fff4624 0000001c 3fff288c  
3fffff40:  3ffed16c 00000000 3fff4624 4023fda5  
3fffff50:  1601a8c0 000001b8 00000000 00000024  
3fffff60:  00000002 0000001a 40235c9f 3ffee3b0  
3fffff70:  3ffed144 3fffdcc0 3ffea988 3ffea988  
3fffff80:  40235c12 3ffee3b0 00000000 3fff3b24  
3fffff90:  3fffdc80 00000000 3fff4624 402439ef  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

```

I need to test the config.txt on smoothieware which is around 20K but I expect same issue  as repetier EEPROM dump and SD File listing...
- Authentication - not yet ported
- wifi scan is not working but  sample code is available on readme so need to be ported also

-on ESP32 : 
- nothing tested / done yet (finger crossed it works after ESP8266 is ported)

luc-github, 2017-11-08:
Update : WDT issue on ESP8266 is now fixed (use ESP.wdtFeed(); helped and use `beginChunkedResponse` instead of `beginResponseStream` is mandatory for long flow

luc-github, 2017-11-11:
The branch https://github.com/luc-github/ESP3D/tree/asyncwebserver is updated

luc-github, 2017-11-13:
I close issue as now code is available and async is no more a blocking feature thanks @me-no-dev :smile:

closed 2017-11-13 by luc-github

me-no-dev, 2017-11-13:
will be in a couple of days in CZ, will give it a go :)