luc-github/ESP3D #112: Smoothie support

arthurwolf, 2016-08-05:
Hey there.

This is really great. I have an idea that would make it even nicer ( at least for Smoothie users ).
Essentially with very little code, you could make this a fully transparent drop-in for Smoothie's ethernet interface. This means users would be able to use the web interfaces the Smoothie serves over ethernet, as-is. 

To do this all you need to do is : 
* When users access http://ip_of_esp3D:80/sd/some_file_name, serve that file by reading it on the smoothieboard ( "cat" command over serial )
* When users access http//ip_of_esp3D:80/command and pass some command as a POST parameter, send that command to the Smoothieboard, and answer whatever it answers back to the client.

Does that make sense ?
Doing this would essentially make it a fully transparent bridge, using which users can use Smoothie's existing interfaces, which is something a lot of Smoothie users want to have.

Cheers :)

luc-github, 2016-08-05:
Hi - yes part 1 will be easy for sure
the second part is already half  implemented, this is how ESP3D communicate with smoothie or other boards, but display raw answer part is still missing as ESP3D is polling board to get temperatures / positions / flow rate / speed rate / and filter specific answers to display informations porperly, so just need to add a continuous output which is already in TODO list

Actually the easy solution for point 2  is to use the dataport 8888 in any terminal and it works already, this on my Azsmz board having esp8266 1Mb flash:

[image]

[image]

arthurwolf, 2016-08-06:
What do you think about this : 
* I send you a free Smoothieboard
* In exchange you implement this, and add a page to the smoothie wiki explaining smoothie users how to go from a "fresh" esp8266, to one that can do this.

Deal ? :)

openhardwarecoza, 2016-08-06:
please please take Arthurs offer (: - if you get that working I'll also run you a crowd donation campaign on the LaserWeb project (;

arthurwolf, 2016-08-06:
@openhardwarecoza well then *you* also need to make sure laserweb can be run stand-alone in the browser ( no need for a serial bridge or a server of any kind, like fabrica ). then users just get a "raw" smoothieboard and an esp8266 and they can get it to serve laserweb over the network without a raspi playing server. that would be so awesome. there's no major block to this right ?

openhardwarecoza, 2016-08-06:
none at all:  The serial bridge is just for serial.  If i can replace the serial bridge with an ESP (serial queue code is already javascript under nodejs, so thats easy enough to move to frontend too) thats a go.  I am anyway doing that for the ethernet support.

arthurwolf, 2016-08-06:
ideally you also need to do what fabrica does and package everything into
one single big html file ( so there is no problem with the browser
requesting several files at once )

luc-github, 2016-08-06:
@arthurwolf about this point :
>When users access http//ip_of_esp3D:80/command and pass some command as a POST parameter, send that command to the Smoothieboard, and answer whatever it answers back to the client.

Need a page for entering the command, right ? what it should looks like ?
or current way is ok use existing send command, and I "just" need to put answer in info box
  ?
[image]

arthurwolf, 2016-08-06:
No no, this isn't something users do. This is something your server allows,
and then once it allows that, *existing* smoothie web interfaces, that use
this specific adress/system, can be compatible out of the box.

openhardwarecoza, 2016-08-06:
I guess what we actually need is a fresh project,  just a sketch that does
these two things.  Nothing else (:

Fetching ip/sd/index.html  shouldnt come off spiffs,  it should already
come from 1) above

luc-github, 2016-08-06:
@arthurwolf ok that clear now so even easier

luc-github, 2016-08-06:
sorry what I should install on computer to have this *existing* smoofhie web interface to test the ursl on esp?
I am not familiar of smoothieware

arthurwolf, 2016-08-06:
http://smoothieware.org/install-web-interface

luc-github, 2016-08-06:
yes but this interface is on sd card, i know it, you mean this interface will be hosted on spiffs of esp ? or is there a way to connect esp to smoothie sd card reader?

arthurwolf, 2016-08-06:
Yes, you can read anything on Smoothieboard's sd card using the "cat"
command.

cat /sd/webif/index.html will return the content of that file, which you
then just have to send back to the client over http

luc-github, 2016-08-06:
Ok I got it now when you initialy say files I was only thinking about gcode files,  ok more clear

luc-github, 2016-08-06:
Looking at the html page, the commands should work using the "pipe"
Loading a file should work even for   ```<script type="text/javascript" src="functions.js"></script>``` in theory

But this simple bridge won't work for Upload file part as need to use the M28/M29 command to transfert files over serial unlike ethernet that have access to SDCard -  I do not see a console command for that in http://smoothieware.org/console-commands

openhardwarecoza, 2016-08-06:
Remember we can always get @Wolfmanjm to add features to smoothie to ease
upload

arthurwolf, 2016-08-06:
why a command ? why not use the gcodes normally used for file upload over
serial ?

luc-github, 2016-08-06:
@openhardwarecoza my understanding is no change on smoothieware part when have already an existing solution ( or not)

BTW @arthurwolf I followed your suggestion for Nextion .  I limited my interaction with smoothie general code to kernel to hook my SerialConsole module and module handle everything - good advice - thank you, - 
 
I will work on the "cat pipe" next week and then on the "cmd pipe"  as I have everything to test on hand - upload file is special case that just need a third pipe using M28/M29 for file upload, it is already available but need to adjust for Smoothie uri

Will update when I do progress

openhardwarecoza, 2016-08-06:
Awesomeness!  Shout if you need anything!  (ESPs,  lasers lol anything)

openhardwarecoza, 2016-09-01:
Hi Luc

Just checking in if you made an progress on this? (:  Even if its just a seperate sketch

luc-github, 2016-09-01:
No I am late, busy on others projects and real life - I hope start implement some command before next week - Already thinked on some algorithm as need to catch the anwser but nit transmit  Ok of ack command, and small others details

DarklyLabs, 2016-09-05:
Happy to support this implementation. Let us know if you need anything that will help.

luc-github, 2016-09-05:
I am still at cat file level, it take more time than expected because cannot use stream feature of serial, there is an `ok\n\r` at the end of the file which give size mismatch error, also cannot use normal chunked with unknown size because it works only for text file and not for binary (yes need to handle image also) ,  so need to get the file size, and of course there is no easy way to get the file size but list the directory and catch the file matching the name and get size in listing.

So now I need to write a function to get file size before continuing -  i will keep updated

DarklyLabs, 2016-09-06:
Thanks for the update. We appreciate the work you are putting into this.

luc-github, 2016-09-07:
i have created a branch for test : https://github.com/luc-github/ESP3D/tree/smoothie-test
only display page / navigation is implemented for the moment no command will work

http://192.168.1.53/sd/ will check if index.html it is there and display it
http://192.168.1.53/sd/webif/ will check if index.html it is there and display it but seems not - debug works but production don't - need to dig in
http://192.168.1.53/sd/webif/index.html wil display html page

I did not test yet with images or binaries but serial communication is very slow - so classic seems OK because page is 100k but bigger  go in time out 

I will test more and fix issues before starting command management part

luc-github, 2016-09-07:
Seems cat command is mostly for text not binary so content looks corrupted somehow - not sure need to dig in smoothie code

openhardwarecoza, 2016-09-07:
@wolfmanjm @arthurwolf

wolfmanjm, 2016-09-07:
Yes cat is for text only. Why would you even need to cat a binary file?

arthurwolf, 2016-09-07:
@wolfmanjm : They use cat to "get" a file over serial and serve it from
their webserver : client asks for a file, they ask for it via cat, get the
file back, serve it back to the client.

openhardwarecoza, 2016-09-07:
For sending image files etc for the web interfaces we intend hosting on the
smoothie sd,  but serving over esp. . We may need to extend smoothieware.

luc-github, 2016-09-08:
I used chrome extension loadtime to get loading time using serial stream

page | Time
----- | -------
http://192.168.1.53/sd/webif/index.html | 21.4 s
http://192.168.1.53/sd/webif/classic/index.html |  12.3 s
http://192.168.1.53/sd/webif/imrahil-ui/index.html | 15s and not working (should be due to lack of command support )
http://192.168.1.53/sd/webif/cnc-ui/index.html | 4.87 s
http://192.168.1.53/sd/webif/fabrica/index.html | 17 s and not working  (should be due to lack of command support )
http://192.168.1.53/sd/webif/documentation/web/html/index.html | 3 s and not working  (should be due to lack of command support )
http://192.168.1.53/sd/webif/file-explorer/index.html | not working  (should be due to lack of command support )
http://192.168.1.53/PRINTER | 3.62 s

the pages on SPIFFS are responsive vs serial ones (PRINTER is 3.62s because of SD content it is the heaviest one) 
you can  check others here https://github.com/luc-github/ESP3D/issues/104#issuecomment-230229267

Looking at the fact the serial is very slow (and it is normal), that  images / fonts / cannot be loaded properly because they are binary, big limitation.

I am not sure this is a good end user experience for smoothie users, I do not think this way should be proposed.
What do you think ?

arthurwolf, 2016-09-08:
> you can check others here #104 (comment)

What serial baud rate are you using ?
Smoothie can support extremely high rates.

luc-github, 2016-09-08:
115200 but this is not the issue I think , the issue is the serial buffer of ESP is 128byte

the file transfert from web to SD are also very slow - that is why I worked on direct SD access  :
https://github.com/luc-github/ESP3D/issues/93

arthurwolf, 2016-09-08:
Would you mind trying doubling this anyway, just to check ?

luc-github, 2016-09-08:
at 230400
http://192.168.1.53/sd/webif/index.html take 15.6 s vs 21.4 s at 115200
still very slow IMHO

arthurwolf, 2016-09-08:
Yeah it likely needs more cache and some optimisations ...
no DMA available ?

luc-github, 2016-09-08:
Anything could be done of course, but it means not using the standard ESP core.
I know Duet team use their own core - but this need more code management and need provide an easy way to build / compile to match the esp target according flash size/SPIFFS size, ESP model, etc...

Which is not my target currently

openhardwarecoza, 2016-09-11:
@arthurwolf any objection against hosting the files on the SPIFS?  (apart from the size limit)

arthurwolf, 2016-09-11:
What is the size limit ?

openhardwarecoza, 2016-09-11:
http://arduino.esp8266.com/versions/1.6.5-1044-g170995a/doc/reference.html#file-system

arthurwolf, 2016-09-11:
> doc/reference.html#file-system

Yep that's not going to cut it for anything useful, unfortunately.
Then, this shouldn't concentrate on serving files at all, and just on
having the "/command" path working correctly.
Then we can use files hosted on the computer or on the web, and have those
talk to the /command path directly, and it'll work.

openhardwarecoza, 2016-10-06:
Hosting the app off-world (ie local webserver,  or cloud,  or pi,  or something else)  we only need firmware for for the esp to a)  upload data to smoothieware sd card. . @raykholo and @arthurwolf i bet we can send @luc-github some testbed hardware? 
And b)  single gcode commands -  for jog etc

openhardwarecoza, 2016-10-10:
@luc-github,  swee https://github.com/Links2004/arduinoWebSockets/issues/61#issuecomment-198231422

openhardwarecoza, 2016-10-10:
Anyone want to get paid to get this done for @darklylabs https://plus.google.com/+PetervanderWalt/posts/BkzJmtaanGc

luc-github, 2016-10-10:
Hi Peter websocket bridge already exists : https://github.com/hallard/WebSocketToSerial
or links code in thread you mention.
My current code do not work well with several files calls that is why it use singles file page including css and js.
I am rewriting the whole project using https://github.com/me-no-dev/ESPAsyncWebServer to have better multi files support and library has websocket support also (readme: Async WebSocket plugin offering different locations without extra servers or ports)

openhardwarecoza, 2016-10-10:
> Hi Peter websocket bridge already exists : https://github.com/hallard/
> WebSocketToSerial
> or links code in thread you mention.
> My current code do not work well with several files calls that is why it
> use singles file page including css and js.
> I am rewriting the whole project using https://github.com/me-no-dev/
> ESPAsyncWebServer to have better multi files support and library has
> websocket support also (readme: Async WebSocket plugin offering different
> locations without extra servers or ports)

Thanks Luc! I will investigate https://github.com/hallard/WebSocketToSerial
tomorrow (:

luc-github, 2016-11-10:
Ok now command feed back is now implemented and working for imrahil-ui
[image]

classic UI seems using different way to send commands, when imrahil-ui send command in post plain parameter to url /command like "plain:ls -s", classic seems sending directly the command as parameter name like "ls -s:"  so parameter is see empty and not catched by ESP - I do not know if it is normal or not

For the upload imrahil-ui and classic UI seem not using standard upload process, using input type=file  but put flie content in parameter as binary, this is not supported by ESP - so the solution are to recode ESP core  to support smoothie specs or to modify html/js to use standard API
I have no plan to change standard ESP behaviour  - so I will look into the html/js changes

luc-github, 2016-11-23:
After reading all code and thanks to imrahil clarification I am now able to change properly the code of upload - I have also improved the minified version as the code is barely ureadable after being minimified so I take advantage of esp webserver feature of  reading gzip file and concatened alls files and gzip them to get unique index.hml.gz file for only 11.38 KB and no need for ESP to server external files.

Will update code when all open issues will be closed

luc-github, 2016-11-24:
ESP3D version of imrahil web ui is here : https://github.com/luc-github/smoothieware-webui-for-ESP3D

Imrahil version is now one file and has several cosmetic fixes, the only difference with master version is the upload protocol, as ESP3D web server use standard one but Smoothie is specific.

I think other UI need to be ported as well in same way

luc-github, 2016-11-24:
I think this topic can be closed based on :
1 - cat cannot handle image/binary but text only
2 - serail transfert take ages
3 - workaround to host file on SPIFFS need to rewrite file upload - imrahil is done - other can follow if requested 

@arthurwolf @openhardwarecoza  what do you think ?

openhardwarecoza, 2016-11-25:
Excellent news! I'll have to find time to play with this and see how it is (: Havent tested yet

After the comments in https://plus.google.com/+PetervanderWalt/posts/BMkq52ME4DW?sfc=true I am a little sad about SD support in Smoothie overall, so I am throwing all my support behind your Dual SD plans too!

luc-github, 2016-11-25:
Well Dual SD support is not really related to ESP3D actually  - it is just UI files modification using js and html but specific to smoothie.
I will work on it as soon I finish open items related to ESP3D FW itself

luc-github, 2016-11-26:
@openhardwarecoza dual SD listing is done https://github.com/luc-github/smoothieware-webui-for-ESP3D
Seems there is a bug or a limitation for deleting file on /ext - so I have open an issue -https://github.com/Smoothieware/Smoothieware/issues/1067 -  wait feedback to disable delete function on /ext if it is a limitation which I hope it is not

luc-github, 2016-11-28:
If no feedback on this issue I close it

closed 2016-11-28 by luc-github