luc-github/ESP3D #113: Garbage from ESP on power up.

JamesNewton, 2016-08-08:
How are you managing the "garbage" that ESP-8266's send when they first power up?
http://bbs.espressif.com/viewtopic.php?t=2212

Is it just that the controller happens to ignore that message because it's at the wrong baud rate, or because it's also powering up at the same time, or.... ???

Shouldn't there be a buffer in front of the TX line that stops it sending anything unless the unit is in bootloader mode or until after the code you have previously bootloaded is in control?

I'm mentioning this because I'm guessing at some point in the future, this is going to cause weird, random, strange errors. E.g. the unit is restarted and the device it's connected to jumps for no apparent reason.

luc-github, 2016-08-08:
You mean you have some garbage when boot up ?

Currently there is only a swap operation when a restart happen to put all output on serial1
and serial is enabled only after few second when boot if recovery is enabled
and if no recovery pin enabled the printer serial usually is enabled after esp serial so no garbage go to printer.
the only garbage I am not able to handle currently, is a crash dump as it is in bootloader. but I do not have so - it is fine

luc-github, 2016-08-08:
As you can see I have no garbage at boot or at restart

[image]

JamesNewton, 2016-08-09:
I'm guessing that's because it's being sent at an unusual baud rate (see the link I provided above) and the device is ignoring it. If you can easily attach a scope, I would suggest watching it as it powers up... If it's not transmitting anything at all, then I guess they fixed that in the model you are using. But my ESP-01's and ESP-12's and 7's  all transmit a burst of data on power up.

luc-github, 2016-08-09:
well if swap or delay at boot do not work for you , the ultimate solution is to modify the boot loader - but I did not see the need so far

JamesNewton, 2016-08-09:
Now that I'm thinking about it, the swap from TX to Serial1 makes sense... And you still bootload new code to TX? When you jumper the IO pin into bootloader mode. I'll have to look up the code that swaps the serial output... Thanks!

luc-github, 2016-08-10:
I did need to swap at boot just at restart as none of my ESP 12E / 01 show any garbage at boot - only my nodemcu do, but it is devt board, so not for printer usage

luc-github, 2016-08-16:
Any update on this ? or we close issue ?

JamesNewton, 2016-08-16:
Well, my ESP 01 and ESP 07 and ESP 12 all DO show garbage on the TX pin at boot, but switching to the Serial 1 pin seems to be a solution, so I guess it's a non-issue.

closed 2016-08-16 by JamesNewton