luc-github/ESP3D #116: Davinci Duo 2.0a

DocMcC, 2016-08-18:
Hi,

After much effort I have got the ESP8266  flashed and am able to access the webserver etc.

Worked out by trial and error the wiring to the firmware and connected it all up.  

Issues:

1) sometimes the printer connects, some times it doesn't (seems most reliable at 250000) - when it works it shows the ip address - no idea why sometimes it doesn't

2) I can send commands to the printer but get no information back - not sure why - thought it could be a wiring issue but don't know how to test - any help would be great

3) if I hit the force stop button on the web interface, the printer crashes and destroys the eeprom.bin file on the SD card (thought it had bricked the printer but eventually worked out the problem was the card.  When in this state I immediately get 2 black bars on the screen and the led flashes.  To recover I have to delete the EEprom.bin file.

I think my problem might be due to a problem on the Rx connection but then I would have thought that if the connection was bad it would not work at all!!

I added the #define FIRMWARE_TARGET REPETIER4DV to the configuration.h file - this was not there - is this the correct place to add this,?  Do I need to add anything else?

Any ideas would be much appreciated.

W.

luc-github, 2016-08-18:
Hi 
how did you wire the ESP to the davinci ? someone reported me he had some connection issue due to the 1k/2k resistors on RX.

you do not need to add #define FIRMWARE_TARGET REPETIER4DV to the configuration.h file - I guess you are talking to the printer FW

I guess you enabled wifi on printer FW ? which version do you use ?

The emergency stop just send M50 command so if EEPROM.bin is destroyed - it means there is a problem with SDCard reader

be sure baud rate are identical on ESP and printer for communication me I use 230400 on my Duo

DocMcC, 2016-08-18:
Hi,

For wiring I followed your schematics – I was just thinking that it might be something to do with the wiring.  Found the following: http://esp8266.github.io/Arduino/versions/2.3.0/doc/boards.html  should I remove the resistors and connect the Rx directly to the printer?

I added #define FIRMWARE_TARGET REPETIER4DV to the config.h file for the ESP8266 as per the installation instructions:

To flash the module :
You must define the targeted FW, can be REPETIER (Original Repetier)/ REPETIER4DV (Repetier for Davinci) / MARLIN (Marlin)/ SMOOTHIEWARE (Smoothieware) in config.h and enable/disable any others features

Have enabled the wifi and set the baud rate same on both – I believe it has to do with the resistors so will remove – what exactly are they there for?

Much appreciated,

W

luc-github, 2016-08-18:
Yes try remove them and let me know    
about FIRMWARE_TARGET you can see it is present here already : https://github.com/luc-github/ESP3D/blob/master/esp3d/config.h#L57

DocMcC, 2016-08-19:
Hi,

Tried it with no resisters and just the 1k (removed the 2k) no luck – did notice that the Tx pin on the motherboard was giving essentially a 0v reading (1.4mv)  - Checked it with a multimeter – not sure if this means anything.

When I attach via PUTTY, I can send commands but get nothing back, not even gibberish.

Do you know if I attached the UART directly to a laptop using a TTY to serial could I test it?

Also, Would you know how the other person resolved their issue?

BTW, the config.h I was using was the one from the 0.6.2 zip file – I just added it to it.

The version of Repetier I am using is 0.92

Much appreciated,

W.

luc-github, 2016-08-19:
hi 1k and 2k must be removed, only keep 10k, keep 1k is not correct.  
here the topic https://github.com/luc-github/ESP3D/issues/103   
can you share picture of your connection to davinci?
ok the 6.2 do not need it so no issue

DocMcC, 2016-08-20:
Hi,

I have removed the resistors to no avail.

I believe I might have damaged the connection on the motherboard when I was soldering on the wires ☹  would you know what pin on the processor the TX would be connected to so I can do a continuity check?

I have updated to your newer version of the ESP firmware as per your previous mail. FW: V0.7.81

Repetier version is (0.92) 1.15-10-30.1.3a  (I assume this is an October 2015 build – is there a newer version you would recommend?)

At this point I was thinking of maybe attaching a osciliscope to the Tx to see if there is any “life” or would you know how to connect the printer to a pc using the UART?

Any help is appreciated

W.

luc-github, 2016-08-20:
For Davinci board: here the sheet of MCU : http://www.atmel.com/Images/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf you will find the pin for the UART programming I think

For Repetier: Devt branch is the latest one based on 0.92.10, master is stable one based on 0.92.3

ESP3D git version: master branch is latest one for normal use

my understanding on Davinci board (at least on my 2.0, 2.0A TBC by you) only connector is missing for UART so connect the 4  wires of USB cable should do the trick

DocMcC, 2016-08-20:
Hi,

Thanks for your help, just got it working, traced the pins back to the processor (found the spec on line) – I had managed to wire the UTxd to earth ☹

All fixed thanks for your help.

Next job is to work out why it is not giving me a list of files when I click the SD button – other functions seem to work ok

W.

luc-github, 2016-08-20:
Hi good news :smile: 
If you can share a picture of the location on board where you connect ESP would be great, I could complete the documentation.

About SD send M20 command in command input box and after check the result of 
http://[your ESP IP]/SDFILES
this will tell if issue is on html page or on FW side

I am currently rewriting this part of SD file list should be ready early next week

luc-github, 2016-08-20:
you can also use putty to connect to your ESP using port 8888 and send M20 and see result

DocMcC, 2016-08-20:
Hi,

Pins are (from Left to right) 

UTxd, 
+3.3v
Gnd
URxd

photo attached - is clearly marked on the MB.  Would not do it the way I did it if I was doing it again - would put in a socket.

Anyway, sometimes it comes back with the file names, sometimes not.  M20 returns thefile names and the JSON has them if I download it.

Also, sometimes when I got the list it would not print - not sure what is going on.

Anyway looking forward to testing the new version :)

As regards github - is there an easy way to download a fork that does not have the download option?  Can be quite frustrating :)

Your help is much appreciated,

W.

[image]

luc-github, 2016-08-20:
Thanks for sharing -  yes I think socket is better 

Looks like communication issue - M20 should always send back list - you mentioned you use 2500000 for baud rate, now you fixed tx problem did you try to decrease to 115200 to check ?

Do you still have same issue with M112 emergency command ?

DocMcC, 2016-08-20:
Hi,

Will try lower baud rates.  Have noticed that if I enable wifi it works but when I reboot it can’t read the eprom.bin file – seems to be corrupting it.

Have you seen this before?

W.

luc-github, 2016-08-21:
no I use heavily emergency button on web ui to test connection and never saw the issue

do you have same issue using emergency stop keys on printer ?

DocMcC, 2016-08-21:
When I switch on WIFI on the printer, it kills the EEPROM.BIN file - probably something for the Repetier issues list.

As regards GitHub, if I go into one of the forks there is no clone or download options, I have to open each file separately as raw and the copy/paste the code into notepad then save it down.  Checked again today and they are there - maybe I am just going blind.

Thanks for your help.

W.

luc-github, 2016-08-21:
which fork are you using ?
there is no download only if you are at commit level 

For repetier FW yes please open issue on repetier FW with repro step as I cannot reproduce it

DocMcC, 2016-08-24:
Hi, 
apart from a few niggles think all is sorted,

Many thanks for your help and if you need me to do testing let me know.

W.

closed 2016-08-24 by DocMcC