luc-github/ESP3D #118: ESP32 support

luc-github, 2016-09-11:
ESP32 is out need to support it

luc-github, 2016-11-12:
Ok got my nano32 and play with https://github.com/espressif/arduino-esp32
so far still need to wait for EEPROM and Webserver library - a lot of debug message are on output

Keep watching until official first release

luc-github, 2016-11-26:
SDK is not yet ready : EEPROM, Sync WebServer, SD libraries are still under development or planned

luc-github, 2017-05-19:
Based on current ESP32  status - the FW will need some rewrite so better to have dedicated github when time will be (soon :wink: )

luc-github, 2017-09-17:
Reopening as now most of necessary library are available

Porting already started will create a branch for it soon

Current main issue is web server - embeded page is working but streamfile function is not
first draft is not really clean as I have `#ifdef ARDUINO_ARCH_ESP8266 ` and `#ifdef ARDUINO_ARCH_ESP32` everywhere so need to do some code refactoring using global macro once all needed functions will be identified

luc-github, 2017-09-17:
Ok problem solved using proper library for web server : https://github.com/bbx10/WebServer_tng

nostahl, 2017-09-19:
keep up the good work! I've been keeping eye on this project for long time now.

luc-github, 2017-09-20:
Thank you 
for the moment everything seems now fixed and working but : 
- NetBios due to lack of available library
- SSDP due to lack of available library
- Silent boot so will move output from Serial to Serial1 and avoid the bootloader message and missing swap function
- baudrate function -> need to do a PR
- setRxBufferSize -> need to do a PR or ignore it as default is 256

so very promising  😸

nostahl, 2017-09-20:
Is there possibility for arducam or some offering that's not a separate up cam too?

nostahl, 2017-09-20:
Ip cam

luc-github, 2017-09-21:
Well I already studied that https://github.com/luc-github/ESP3D/issues/82
Even ESP32 is definitly better for this purpose as more pins and more powerful the arducam module is not cheap and cannot compare with cheap IP cam.
I am not expert in video stream neither electronic so would need a lot of time for me to find a very cheap compatible video module if exists.

if there is some code and really cheap compatible camera I am open to give a try

luc-github, 2017-09-21:
baudRate PR (https://github.com/espressif/arduino-esp32/pull/651) done but UART Driver may have issue, so need an update:  https://github.com/espressif/arduino-esp32/issues/649

me-no-dev, 2017-09-21:
I love everything about this project but one thing ;) it's not Async :P

luc-github, 2017-09-21:
Thank you 
well this in pipe (https://github.com/luc-github/ESP3D/issues/110), soon I will release a big update for SDCard as serial upload is slow like hell, but I cannot release the code publicly yet , also I am worry to do async upload(s) to SD but it will come eventually...
Using Async as it is , it is easy as you did all the crazy job ^_^, (I saw when debugging the upload issue) - but adding more code on async scare me a little haha

me-no-dev, 2017-09-21:
what are you scared of :D async does not bite :P

luc-github, 2017-09-21:
Porting update:
* Better to use Serial 2 as default output  because Serial 1 use same pins (9/10) as flash D2/D3 and QIO is default mode instead of DIO, so if user do pay attention when compiling, ESP will go to endless restart

* Need to check ESP32 SPIFFS upload speed vs ESP8266: it seems really slower but need to benchmark to confirm

* Need to confirm no OTA with 2Mb boards according https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/minimal.csv

me-no-dev, 2017-09-21:
> Serial 1 use same pins (9/10) as flash D2/D3

you can use any pins (almost)

me-no-dev, 2017-09-21:
> Need to check ESP32 SPIFFS upload speed

hold on for a day or two ;) there is a bug in spiffs and fix is coming

luc-github, 2017-09-21:
@me-no-dev thanks a lot 😸

luc-github, 2017-10-04:
Porting update:
 - Need to redo the travis scripts to implement ESP32
 - NetBios/SSDP libraries are still missing
 - Serial instead of Serial2 can be used if GPIO15 is set to GND (thanks Me-No-Dev) 
 - Need to review SPIFFS upload performance on ESP32 vs ESP8266
 - Need heavy testing

So I will push it to devt branch this week

luc-github, 2017-10-07:
I just did some bench and SPIFFS itself is not slow - it is the upload which is slower : 
for same file size it take 2.6 times more on ESP32 vs ESP8266

Data | ESP8255 | ESP32
------------ | ------------- | -------------
total Upload | 14284ms | 37664ms
 Write time | 13248ms | 1911ms
 File size | 381216B | 381216B

me-no-dev, 2017-10-08:
what do you mean by upload? is the flash frequency the same? SPIFFS clean? all of those can result in different speeds

luc-github, 2017-10-08:
Hi @me-no-dev  welcome back ^_^ no Jet lag ?
I mean file upload to SPIFFS using browser :

me-no-dev, 2017-10-08:
Oh lots of jetlag... i am not sure if I am awake or sleeping :D check flash freq ;) default might be 40MHz and you can try 80 (also make sure FS is clean)

luc-github, 2017-10-08:
The SPIFFS is not the issue as I can see as write time is really low - it is total upload time which is 2.6 bigger
the total Upload is the time between start and end upload, write time is cumuled SPIFFS time used by write

me-no-dev, 2017-10-08:
so where is the time lost?

luc-github, 2017-10-08:
looks like the time between each packet to be write is bigger:
```
/SPIFFS files uploader handle
void SPIFFSFileupload()
{
#ifdef DEBUG_PERFORMANCE
    static uint32_t startupload;
    static uint32_t write_time;
    static size_t filesize;
#endif
    //get authentication status
    level_authenticate_type auth_level= web_interface->is_authenticated();
    //Guest cannot upload
    if (auth_level == LEVEL_GUEST) {
        web_interface->_upload_status=UPLOAD_STATUS_CANCELLED;
        ESP_SERIAL_OUT.println("M117 Error ESP upload");
#ifdef ARDUINO_ARCH_ESP8266
        web_interface->web_server.client().stopAll();
#else 
		web_interface->web_server.client().stop();
#endif
        return;
    }

    static String filename;    
    //get current file ID
    HTTPUpload& upload = (web_interface->web_server).upload();
    //Upload start
    //**************
    if(upload.status == UPLOAD_FILE_START) {
#ifdef DEBUG_PERFORMANCE
            startupload = millis();
            write_time = 0;
            filesize = 0;
#endif
        //according User or Admin the root is different as user is isolate to /user when admin has full access
        if(auth_level == LEVEL_ADMIN) {
            filename = upload.filename;
        } else {
            filename = "/user" + upload.filename;
        }
        ESP_SERIAL_OUT.println("M117 Start ESP upload");
        //create file
		web_interface->fsUploadFile = SPIFFS.open(filename, SPIFFS_FILE_WRITE);
        //check If creation succeed
        if (web_interface->fsUploadFile) {
            //if yes upload is started
            web_interface->_upload_status= UPLOAD_STATUS_ONGOING;
        } else {
            //if no set cancel flag
            web_interface->_upload_status=UPLOAD_STATUS_CANCELLED;
            ESP_SERIAL_OUT.println("M117 Error ESP create");
#ifdef ARDUINO_ARCH_ESP8266
			web_interface->web_server.client().stopAll();
#else 
			web_interface->web_server.client().stop();
#endif
        }
        //Upload write
        //**************
    } else if(upload.status == UPLOAD_FILE_WRITE) {
        //check if file is available and no error
        if(web_interface->fsUploadFile && web_interface->_upload_status == UPLOAD_STATUS_ONGOING) {
#ifdef DEBUG_PERFORMANCE
            filesize+=upload.currentSize;
            uint32_t startwrite = millis();
#endif
            //no error so write post date
            web_interface->fsUploadFile.write(upload.buf, upload.currentSize);
#ifdef DEBUG_PERFORMANCE
            write_time += (millis()-startwrite);
#endif
        } else {
            //we have a problem set flag UPLOAD_STATUS_CANCELLED
            web_interface->_upload_status=UPLOAD_STATUS_CANCELLED;
#ifdef ARDUINO_ARCH_ESP8266
			web_interface->web_server.client().stopAll();
#else 
			web_interface->web_server.client().stop();
#endif
            ESP_SERIAL_OUT.println("M117 Error ESP write");
        }
        //Upload end
        //**************
    } else if(upload.status == UPLOAD_FILE_END) {
#ifdef DEBUG_PERFORMANCE
        uint32_t endupload = millis();
        DEBUG_PERF_VARIABLE.add(String(endupload-startupload).c_str());
        DEBUG_PERF_VARIABLE.add(String(write_time).c_str());
        DEBUG_PERF_VARIABLE.add(String(filesize).c_str());
#endif
        ESP_SERIAL_OUT.println("M117 End ESP upload");
        //check if file is still open
        if(web_interface->fsUploadFile) {
            //close it
            web_interface->fsUploadFile.close();
            web_interface->_upload_status=UPLOAD_STATUS_SUCCESSFUL;
        } else {
            //we have a problem set flag UPLOAD_STATUS_CANCELLED
            web_interface->_upload_status=UPLOAD_STATUS_CANCELLED;
#ifdef ARDUINO_ARCH_ESP8266
			web_interface->web_server.client().stopAll();
#else 
			web_interface->web_server.client().stop();
#endif
            SPIFFS.remove(filename);
            ESP_SERIAL_OUT.println("M117 Error ESP close");
        }
        //Upload cancelled
        //**************
    } else {
			ESP_SERIAL_OUT.println("M117 Error ESP close");
			return;
        web_interface->_upload_status=UPLOAD_STATUS_CANCELLED;
        SPIFFS.remove(filename);
        ESP_SERIAL_OUT.println("M117 Error ESP upload");
    }
    delay(0);
}
```
I did not check further - neither did a minimal sketch to report issue - My jet lag just gone ^_^ I just did easy stuff

me-no-dev, 2017-10-08:
Since LwIP goes through a few threads instead of direct writing, maybe that introduces some time difference.

luc-github, 2017-10-08:
May be - that is just the feeling the upload was slower was not just a feeling - now if it is normal - I am fine with it ^_^
have rest and drink coffee - Jet lag will be gone soon :wink:

luc-github, 2017-10-08:
Ok ESP32 is now part of devt branch - still need to update travis but code and readme is there now : 
https://github.com/luc-github/ESP3D/tree/devt

me-no-dev, 2017-10-08:
Here is a Speed Test I did today on ESP32 and ESP8266 without going through FS. Upload used the same 61912518 B file (same size for Download too):
- ESP32 Upload: 3.99 Mbps
- ESP32 Download: 8.32 Mbps
- ESP8266 Upload: 9.68 Mbps
- ESP8266 Download: 8.46 Mbps

Following handlers used:
```cpp

    server.on("/upload", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send(200, "text/html", "<form method='POST' action='/upload' enctype='multipart/form-data'><input type='file' name='test'><input type='submit' value='Update'></form>");
    });

    server.on("/download", HTTP_GET, [](AsyncWebServerRequest *request){
        AsyncWebServerResponse *response = request->beginResponse("text/plain", 61912518, [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
          if(!index){
              Serial.printf("DownloadStart: (%lu)\n", millis());
          }
          memset(buffer, '0', maxLen);
          if((index + maxLen) == 61912518){
              Serial.printf("DownloadEnd: (%lu)\n", millis());
          }
          return maxLen;
        });
        response->addHeader("Content-Disposition", "attachment; filename=\"test.txt\"");
        request->send(response);
    });

```

me-no-dev, 2017-10-09:
Did a new test with my PC connected to Ethernet instead of WiFi. Here are the results ;)
- ESP32 STA Upload: 8.66 Mbps
- ESP32 STA Download: 9.02 Mbps
- ESP32 AP Upload: 7.94 Mbps
- ESP32 AP Download: 9.40 Mbps
- ESP32 ETH Upload: 15.53 Mbps
- ESP32 ETH Download: 23.12 Mbps
- ESP8266 STA Upload: 9.65 Mbps
- ESP8266 STA Download: 11.81 Mbps
- ESP8266 AP Upload: 7.45 Mbps
- ESP8266 AP Download: 9.27 Mbps

luc-github, 2017-10-09:
I see what you want ^_^ - you want me to use AsyncWebServer instead of Sync one - no worry I plan to do it :wink: but better to have stable sync version to compare and move to Asunc
your first data are interesting are they match mine (I did not tested download as looked fine forme) : 
* ESP32 Upload: 3.99 Mbps
* ESP8266 Upload: 9.68 Mbps 
ESP32 is 2.4 slower than ESP8266 in upload on my side was 2.6 on smaller sized  file which increase unaccuracy 

about your second part using PC to Ethernet instead of Wifi - (this is what I used to do my tests also) : 
* ESP32 ETH Upload: 15.53 Mbps (1.6 slower than ESP8266)
* ESP32 STA Upload: 8.66 Mbps (1.1 slower than ESP8266)
* ESP8266 STA Upload: 9.65 Mbps 
 ESP8266 looks identical but ESP32 is now almost as fast as ESP8266 so I am confuse -  for ESP32 ETH I do not have such device so I cannot test

me-no-dev, 2017-10-09:
how is 15 Mbps slower than ESP8266 :D

luc-github, 2017-10-09:
Arf morning sorry .... I correct

luc-github, 2017-10-10:
Another difference between ESP8266 and ESP32 is format SPIFFS : 
On ESP8266 the output is clean but on ESP32 the WD generate message:
```
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: ipc0
CPU 1: loopTask
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: ipc0
CPU 1: loopTask
```
Need a yield() or something during the loop of erasing sector ? the original SPIFFS library do not have but this does not trigger ESP8266 which has a bigger SPIFFS (3M vs 1.31M)

me-no-dev, 2017-10-10:
I'll make a few assumptions here ;)
- you are using Arduino as IDF component
- You have enabled WDT on idle task of Core 1

Disable idle task WDT in menuconfig as it's done in Arduino and you will be fine.

luc-github, 2017-10-10:
Hi @me-no-dev  no I use arduino-esp32 
but I checked sdkconfig as you suggest and seems WDT is not enable on idle task for CPU1
```
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
CONFIG_TASK_WDT_PANIC=
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=
```

me-no-dev, 2017-10-10:
yeah... but I have no issues formatting SPIFFS, so I am not sure where and how you are getting this if WDT is not enabled

luc-github, 2017-10-10:
Formating is working fine actually - just getting output but any output is a possible issue if go to 3D printer

I will try use aduino as component, check the menu config to see if it make difference - may be the sdkconfig present in sdk directory is not the real one but an old one

luc-github, 2017-10-11:
Hi @me-no-dev 
After compiling arduino as idf component using sdk sdconfig I get same result 

So using make menuconfig  I have disabled WDT on idle task on CPU0 ( I do not see CPU1 in UI but it is present in original sdk sdconfig)
[image]

no more output during formating - so I guess setting is not used
when I  check at the new sdconfig generated
no more 
`CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=`
but instead
` CONFIG_TASK_WDT_CHECK_IDLE_TASK=`
 So I guess the issue is default sdconfig used to esp-arduino is no more valid and need to be generated again

me-no-dev, 2017-10-11:
strange... will be done in next lib rebuild

luc-github, 2017-10-11:
Do you want me to create issue in tracker ?

me-no-dev, 2017-10-11:
nope ;) i need to do it anyway

luc-github, 2017-10-11:
Ok Boss -  thank you ^_^

kafftass, 2017-10-19:
Hi ESP3D Folks , what a great work.
I think this project will have a great future, because with ESP32 the platform is more powerful.
I tested the Dev on a ESP32, what a great jump to the 8266 version i have tested before.
The UI is yery good , on PC and smartphone, and fast.
AND, this project is a very good toolkit to do more/other with the ESP32
Thx a lot.

I must changed some code in wificonfig.cpp. I could not get a connection in STA mode to my WLAN.
I put the setup PHY_MODE section before i set the Station Mode.  And disable WiFi.enableAP(false)  Now it works for me
Line 340
        //WiFi.enableAP(false);
        delay(100);
         //setup PHY_MODE
        if (!CONFIG::read_byte(EP_STA_PHY_MODE, &bflag )) {
            return false;
        }
#ifdef ARDUINO_ARCH_ESP32
        esp_wifi_set_protocol(ESP_IF_WIFI_STA, bflag);
#else
        WiFi.setPhyMode((WiFiPhyMode_t)bflag);
#endif
        //setup station mode
        WiFi.mode(WIFI_STA);
        delay(100);

luc-github, 2017-10-20:
Thank you 
I have 2 different ESP32 and not such issue 
the line you commented is not supposed to block STA but to be sure it use STA only so I feel strange it is a problem
and putting PHY mode after make the feature not to be used so I suspect more the PHY value is not supported for your network
when you say it is not working : the ESP crash or cannot connect to AP ? what is error message for not connecting ?

kafftass, 2017-10-28:
Today i got ready built modules with WROOM32 Chips and have the same issue with the original code.
First i thought it was because i build the first module by myself with a barebone ESP32 Chip.
But that seems not to bee the solution. Because it is working fine with this changes described before it is no great issue for me. 
I use WPA2 WLAN security in conjunction with a Fritz-Box 6490, with the standard code he tries to connect and went into AP mode after a while. With the changed code it is connected after 3-5 tries ( connecting ...)
really nice project

luc-github, 2017-10-28:
Ok I can reproduce your issue on one fresh module - but same code is working on other ESP32 
The difference is old ESP32 was connected initially using ESP sample not full ESP3D code   
I suspect some default setting not being initialized  - originaly this sequence was done to avoid crash to fresh ESP8266 - so It seems now the oposite happen on ESP but instead of  crashing it does not connect

May I ask you to confirm now your module is able to connect with your modification to revert and flash original code to see if issue is still there ?

I will dig on my fresh module

luc-github, 2017-10-28:
Well even after able to connect - reverting code still have issue on fresh board but not on old one - 

Could be a silicon issue ? Anyway applying the move of phy mode seems solving issue and also working on old board - so all boards are able to connect now

good catch @kafftass I pushed the fix : https://github.com/luc-github/ESP3D/commit/df9ec26a1e84b66b702d7e65abbcec1ade960bd0
Thank you

luc-github, 2017-11-13:
Ok now ESP32 is supported with today core limitation (SSDP/NetBIOS not yet available) I think it is Ok to close issue

Thank you

closed 2017-11-13 by luc-github

luc-github, 2018-03-20:
I have ported SSDP/NetBIOS libraries  https://github.com/luc-github/ESP32SSDP https://github.com/luc-github/ESP32NETBIOS

So now definitly 100% compatible -  Update will be in 2.X