luc-github/ESP3D #122: server.py wrong for python 3.5.2

kisoft, 2016-10-02:
The server.py file doesn't work at python v3.5.2.
I have made changes to this file and now it works.
I badly know Python, it is possible to eat some mistakes.
Excuse for my English, it is machine translation.

File attached (server.txt)

[server.txt](https://github.com/luc-github/ESP3D/files/504974/server.txt)

Good luck!
Thanks for esp3d!

luc-github, 2016-10-02:
Hi thanks for the code - I still use 2.7 so did not spent time on 3.X - will have a look to your code
did you tried 2to3 tools to check your changes ? https://docs.python.org/2/library/2to3.html

kisoft, 2016-10-02:
Yes, "2to3" converts correctly, server.py begins to work. In that case there is no sense in my file.

closed 2016-10-02 by kisoft