luc-github/ESP3D #126: Connection not possible to SSID with space character

szepnorbee, 2016-11-01:
Hi! I cant connect to my SSID with "HUN Wlan" network name. I selected this from the list and i typed the password also. I get always "No SSID found" error message. Why? Space char in the SSID not allowed?

luc-github, 2016-11-01:
I did because of this : https://github.com/esp8266/Arduino/issues/1342

you can remove this part to see if you experience issue https://github.com/luc-github/ESP3D/blob/master/esp3d/config.cpp#L73-L75 - I did not checked as issue is still open

szepnorbee, 2016-11-02:
Thanks! It works after removed this part of code. (Core 2.3.0.0 ;Arduino 1.6.8)

luc-github, 2016-11-02:
Thanks for the feedback - so I guess I can remove the limitation now - allowing space for SSID should not be a problem for ESP commands but will check

luc-github, 2016-11-24:
I have removed the check - thank you
https://github.com/luc-github/ESP3D/commit/ec90a27e7e666d3dfe3129783c5fe4b21d45ed56

closed 2016-11-24 by luc-github