luc-github/ESP3D #127: RADDS with NodeMcu Lua ESP8266 ESP-12E

angelo-reprap, 2016-11-03:
Hi luc-github,

Perfect job and thank you.
Smart Solution and work fine for me.

I have a small "TINY 3D-Printer" and no Space for Raspi Odroid C2 etc..
So for this the perfect Solution:
http://www.protoworx.de/

I will try to add EEPROM-Settings in the HTML Menue Bar (for Editing the values).
Where I can add the cach Size value (127 for Mega and 63 for the Arduino DUE) ?

[image]

Discription in German:
http://forums.reprap.org/read.php?336,719437

Install guide :
https://github.com/luc-github/ESP3D/wiki/Install-Instructions

Have Fun
Angelo

luc-github, 2016-11-03:
Hi which cache are you talking about ? serial ? command ? repetier host ?

Yes would be nice to update EEPROM - what will be your approach to get the EEPROM value(s) ? update it self is easy but I was still thinking how to grab EEPROM content without overloading ESP memory / or wear out SPIFFS

angelo-reprap, 2016-11-03:
Receive cache size
https://www.repetier.com/documentation/repetier-host/rh-installation-and-configuration/

The Idea is:
M205 - Output EEPROM settings (all Versions)
http://reprap.org/wiki/G-code#M205:_Advanced_settings
Only for Repetier-Firmware easy with:
M360: Report firmware configuration

And put the vaues in an Editable  HTML Formular.

M206 - Set EEPROM value
https://github.com/repetier/Repetier-Host/wiki/EEPROM-settings

If maybe ESP cache can  works with a local.cfg file, more easy to read with M501 and write the values in the config. file. The HTML Page list then the values in a formular, editing the values and write it in the cfg File and so we can Push the config values via M500 in the eeprom.

I have no expieriance with the ESP8266 and there possibilities. I have to try it at first in the G-Code commands with the HTML GUI, than I try to add it .

Nice, If we can make it generally for RepRap-Firmware + Smothie +Marlin and for repetier-Firmware compatible :-). Repetier- Firmware seems easy but I do not know it exactly.

Thank you 
Angelo  
M500: Store parameters in EEPROM

With this M206 read the modify Value and put it in the eeprom.

closed 2016-11-03 by angelo-reprap

luc-github, 2016-11-03:
There is already this open : https://github.com/luc-github/ESP3D/issues/62 - feel free to comment