luc-github/ESP3D #128: Unstable connection

penzuk, 2016-11-08:
Hi! I use firmware repetier 0.92.8 on mega2560 + ramps1.4 + nodemcu v0.1. Nodemcu is connected to RAMPS through level converter mh (3.3/5 volts) к AUX-1 (D0,D1). While connecting to repetier host some mistakes in the log appear. Warning: Missed line detected - correcting bufer usage. When I start  printing, the printer fulfills some commands, then it stops (approximately for a minute), after a while it continues printing but then it stops again and loses the connection to repetier host. Some mistakes appear in the log: Warning: Seems like we missed a ok, got a wait - continue sending. Resend. Wrong checksum.I have tried different releases of esp3d but the problems remains unsolved. Tell me please what the problem is. May be I need any firmware repetier tunnings?

luc-github, 2016-11-08:
I guess you verified that with repetier host manual control everything is fine and connection is correct, command are well send and well received ? be sure CPU is 160Mz ?

I would say to try to decrease the buffer to limit the overflow

penzuk, 2016-11-09:
Yes, the commands are manualy sent and received well. CPU is surely 160MHz.  Where can I find the buffer to limit the overflow? If you mean Receive cache size in Repetier host connection settings, I tried to change it but nothing happenned. I  use Delta printer and high printing speed (about 60-80mm/sec).When printing using a usb cable, the "buff" on the screen of the printer is always 16. When printing using wifi , the "buff" goes from 0 to 16. When it is about 0 the printer lags. If I decrease the printing speed, the situation becomes better and the "buff" comes to 16. How can I print with high speed without any problems?

luc-github, 2016-11-09:
Well ESP serial buffer size is fixed to 128 bytes so it can be a bottleneck depending of the speed of your host send data - also you must be sure you do not have data corruption due to serial communication that bring more traffic on serial than expected due to errors and retry bringing more bottleneck for the data flow, so you may need to  add some capacitor and/or resistor to make serial stable (if it is the problem) - but I cannot help for this as electronic is not really in my skills

About cache I did you try value under 63 ? 

Out of topic, IMHO printing by USB / wifi is not the safer way to print as it is subject to several possible problems - printing from SD is safer

penzuk, 2016-11-09:
I don't think that the problem is in the serial connect. I have checked oscillograms rx/tx lines, they seem to be ok. I have tried value 63 and it doesn't help. 

When I sent gcode to sd cart of the printer, it takes a long time, and it is not always successful. It easier to take the cart away from the printer and copy  gcode using card reader, then return card to the printer and start printing. What are the advantages of printing using wi-fi then?

luc-github, 2016-11-09:
Yes using M29/M30 to tranfer file is a nightmare, that is why I work on this : https://github.com/luc-github/ESP3D/issues/93

for what I can see same kind of issue happen on ESP with another FW: https://github.com/Links2004/arduinoWebSockets/issues/61#issuecomment-259341639

Currently I use wifi SD card, transfert is very fast and I use ESP to launch / monitor print

luc-github, 2016-11-14:
If no update I close topic

closed 2016-11-14 by luc-github