luc-github/ESP3D #133: add wiring guide for other esp8266 boards like wemos d1 mini and also ramps 1.4

nostahl, 2016-11-17:
i have a wemos d1 mini, how would one of these be wired up to a ramps 1.4 board

luc-github, 2016-11-17:
do you have the pin out of your board ? I do not know this one

nostahl, 2016-11-17:
https://www.wemos.cc/product/d1-mini.html
https://www.wemos.cc/sites/default/files/2016-09/mini_new_V2.pdf

luc-github, 2016-11-17:
I guess 5V->5V, GND->GND, Tx/RX ->Serial 2 pins on Aux 4 D17/D16 of Ramps
No ?

nostahl, 2016-11-17:
something about voltage divider needed?

luc-github, 2016-11-17:
Use same wiring as mentioned in readme but use 5v as RAMPS only has 5V- the readme wiring has already a voltage divider on RX

nostahl, 2016-11-17:
ok.

so all of the other wires/resistors in your example are the extra components required for bare esp8266.  these boards like wemos d1 mini have their own components and regulators etc.

luc-github, 2016-11-17:
Yes for wemo do not need the capacitor and no need 10K resistors as already on board - just the RX voltage divider

luc-github, 2016-11-17:
after a quick google seems several people use ESP8286 on Aux 1 D0/D1 which is Serial

luc-github, 2016-11-24:
If no update I close issue

closed 2016-11-24 by luc-github