luc-github/ESP3D #134: SD upload using M28/29 seems broken

luc-github, 2016-11-23:
I guess it is linked to latest update as now SD upload failed - the file is created but it is empty

luc-github, 2016-11-23:
Seems actually ok on Smoothie and failed on repetier

luc-github, 2016-11-23:
orz... I have tried to upload file which is not GCODE file so of course it failed when doing the content check....
M28/M29 check content unlike direct file upload

closed 2016-11-23 by luc-github

denmorru, 2017-03-27:
There is issue with file delete M30 command (from BIN button): it does not seem to work at all in web interface of current version. It makes confirmation popup dialog, and nothing else (the same about "Print" button).
Tested Marlin/RAMPS. In case I send from "Command" form proper GCODE, it does deletion well.

luc-github, 2017-03-27:
@denmorru this is not related to this closed issue which you can see it is invalid actually 

please open a new issue describing :
1 - which FW version you are using - 
2 - which ESP coreversion you are using 
3 - which UI you are using  
4 - which arduino ide version you are using 

this will help to keep one ticket for one issue

Thank you