luc-github/ESP3D #138: Add support Relay in Sonoff

khseal, 2016-12-02:
That device https://www.itead.cc/wiki/Sonoff
You can add a relay control via a web interface? 
Use the relay can be switched on and off remotely 3D printer.
Relay connected GPIO12.

luc-github, 2016-12-02:
git version has already direct pin access :
```
*Get/Set pin value
[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES]
if no V<value> get P<pin> value
if V<value> 0/1 set INPUT_PULLUP value, but for GPIO16 INPUT_PULLDOWN_16
GPIO1 and GPIO3 cannot be used as they are used for serial
if PULLUP=YES set input pull up, if not set input
if RAW=YES do not set pinmode just read value
```
about modifying UI it is up to you according which one you are using,
if you use my sample UI there is a Javascript command for this that you can embedded in button or image:
`Sendcommand('[ESP201]P12 V0');` and `Sendcommand('[ESP201]P12 V1');`
you may also add a read command to handle a toggle state

I hope it answer your question

khseal, 2016-12-02:
Nice, thanks for information!

luc-github, 2016-12-02:
I just ordered this module to have a look - seems already have a fw on it - how did you flahed it ?

khseal, 2016-12-02:
Quite simply, that's pinout
[image]
To flash the need to hold the button when turned on.
There is a detailed description of
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/wiki/Compile-and-Upload

luc-github, 2016-12-02:
I see, thanks for sharing

khseal, 2016-12-02:
Only I have one problem. When connected to a 3D printer  Ramps 1.4 + Marlin 1.1.0-RC7. Arduino is not always start when connected to ESP. Disconnects Rx input Arduino. Arduino runs without problems.

luc-github, 2016-12-02:
what esp board do you use ? what setting do you have ? be sure debug is disabled

khseal, 2016-12-02:
Debugging disabled. I use this device Sonoff.
Almost all settings by default. Wi-fi in client mode.
I chose #define FIRMWARE_TARGET MARLIN
I think that the problem is in the order of start microcontrollers. I'll try to  individually supply power the ESP board.

luc-github, 2016-12-02:
My esp are connected to board due with repetier and smoothieboards AZMSZ / Azteeg 5 with smoothieware and no issue, I did not try marlin / mega ramps since a while.
is you arduino/ramps powered by PC or by external power when you test ?

Issue can be too much current drawn a boot or garbage coming from esp boot loader that do some glitch on start in that case just comment the serial.begin(xxxx); in marlin FW for test should confirm the problem

khseal, 2016-12-02:
One power supply for Arduino and ESP. I tried to connect to a PC and take power from him. Nothing changed. The first start everything is OK, after second boot the Arduino freezes at startup.

I'll try to use the power source of ESP. 
I did buttons for relay control. Everything is working
[image]

khseal, 2016-12-02:
Everything is fine. Remote power control work. 
Sonoff powered from its  internal power source, Arduino is now always run. 
Simultaneous power on Arduino and ESP resulted to a problem startup Arduino.
If ESP is enabled before Arduino, then all is well.
I recommend Sonoff for 3d printer =)

luc-github, 2016-12-02:
good news ^_^

thanks for sharing

luc-github, 2016-12-02:
I see there is a connector for DTH11 do you use it ?

khseal, 2016-12-02:
There is a free gpio on the board. Depending on revision of the PCB.
I do not connect DTH11. I have another device with the measurement environment parameters.

luc-github, 2016-12-02:
so actually you just connect gnd / rx /tx from ESP to Arduino and it is ok - right ?
vcc is not necessary as self powered as you say

khseal, 2016-12-02:
Yes that's right. I used a resistor divider  for input Rx ESP. Arduino has a logic voltage is 5V.

luc-github, 2016-12-02:
Thanks - I think would be nice to have a wiki page (https://github.com/luc-github/ESP3D/wiki) for this setup - feel free to contribute 
I will see when I get the module

luc-github, 2016-12-21:
Got some module but no idea when will have time to work on them

luc-github, 2016-12-31:
I have created the page in wiki which very basic information but it is a start -  can be completed later 
https://github.com/luc-github/ESP3D/wiki/Sonoff

Thank you

closed 2016-12-31 by luc-github