luc-github/ESP3D #139: Status and Information empty.

khseal, 2016-12-03:
This is normal and that the status information is constantly empty?

luc-github, 2016-12-03:
This UI is originaly done for my version of Repetier FW which has a lot of flag. I think Marlin do not have info: flag in serial messages so no information

khseal, 2016-12-03:
I try to find information. Error information is displayed in the logs.

luc-github, 2016-12-03:
the flag are processed here : https://github.com/luc-github/ESP3D/blob/master/esp3d/command.cpp#L524-L556

May be marlin ones have different case or changed since I checked

khseal, 2016-12-03:
Information message have flag "echo:".  It seems there are no status messages in Marlin.

luc-github, 2016-12-03:
Ok will add this flag then for marlin

luc-github, 2016-12-21:
should be done by https://github.com/luc-github/ESP3D/commit/6cbb11ddf0f34105d585c51b0d36d17ad87f99f4

closed 2016-12-21 by luc-github